openthread/openthread #5457: Question - How to use network time synchronization

d1ee, 2020-08-25:
I wanted to experiment with network time synchronization, but I don't know how to use it. When I compile with `TIME_SYNC=1` and type `networktime` in CLI, then I receive an unsynchronized time.
```
> networktime
Network Time:     2507380311us (unsynchronized)
Time Sync Period: 30s
XTAL Threshold:   300ppm
Done
```

Using the api function `otNetworkTimeGet` in `example/main.c` shows more or less the same result.

Compiling again with additional `SNTP_CLIENT=1`, the CLI command `sntp query` was successful after installing ntp on border router. 
```
> sntp query fdde:ad00:beef::ff:fe00:fc00
> SNTP response - Unix time: 1598396643 (era: 0)
```

(In case you wonder why I didn't use the google's default sntp server, currently the devices doesn't send any packets to border router for global ip addresses as well as NAT64 addresses, and I have no idea why)

Maybe it's a stupid question, but does `network time synchronization` synchronize unix time? Or does it have nothing to do with SNTP/NTP?

And what do I have to do if I want to synchronize time, so that the CLI command `networktime` shows `(synchronized)`?

Thanks!

jwhui, 2020-08-28:
The time sync feature allows devices in the same Thread network to synchronize to a single time base that is rooted at the leader. The time sync service is independent of SNTP.

The time sync service relies on sending/receiving MLE Advertisements messages to synchronize time. As a result, you may have to wait up to 30 seconds (or more if messages are getting dropped) to receive network time.

Note that the time sync feature is experimental and not part of the Thread Specification.

d1ee, 2020-08-31:
@jwhui Oh understand, thanks for your detailed answer :)

closed 2020-08-31 by d1ee