openthread/openthread #5460: [continuous-integration] fuzz failure in `Ip6::InsertMplOption

Irving-cl, 2020-08-26:
**Describe the bug** A clear and concise description of what the bug is.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

abtink, 2020-08-26:
I think using `memmove` in `Write()` in this case should help address this. Will submit a PR for this soon.

I think the overlap write was an issue with original implementation as well and was being masked by the fact `CopyTo` would read content into a separate 16 bytes buffer from source message and the write them in destination. If there was an overlap it could overwrite the same data.

abtink, 2020-08-26:
Unfortunately using `memmove` instead of `memcpy` won't work in this case.
We can write into a next chunk in destination message before we get the chance to read it.

abtink, 2020-08-26:
Thanks @Irving-cl  for submitting this and finding the issue.

I submitted https://github.com/openthread/openthread/pull/5462 to unblock.

As I mentioned in the PR commit message, the `CopyTo()` (even with `memmove()`) cannot be used to copy bytes forward within same message (we don't seem to use it anywhere in the code to forward copy within same message and I added an assert for now to block such a use).

The bytes in a message are stored in bunch of `Buffer` chunks. When a forward copy spans over multiple `Buffer` chunks, the `CopyTo()` implementation can potentially write stuff to a next `Buffer` chunk before getting the chance to read it. One way I can think of to handle this, is to go through the `Buffer` chunks in backward order (though not easy to realize as the `Buffer`s are in singly linked list).

jwhui, 2020-08-26:
Resolved by #5462.

closed 2020-08-26 by jwhui