openthread/openthread #5476: Question about the final function to call to send out IPv6 unicast message

oubotong, 2020-08-30:
I tried to implement MQTT-SN protocol with OpenThread and everything works normally within OpenThread. When I tried to build the same program in Zephyr, my board doesn't send out the `CONNECT` unicast message to the broker(I built up a sniffer, and it can't detect the `CONNECT` message).

While I tried to debug it, I couldn't find out the final function to send out the radio. So I want to seek advice on what could be the possible reason for my issue.

Platform: NRF52840-DK
Test successfully by running: `make -f examples/Makefile-nrf52840 MQTTSN=1 JOINER=1`
Test failure in zephyr with the following configuration file `prj.conf` and `overlay-ot.conf`:
**prj.conf**:
```
# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_NET_CONNECTION_MANAGER=y

# Kernel options
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=80
CONFIG_NET_BUF_TX_COUNT=80
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10
```
**overlay-ot.conf**:
```
CONFIG_NEWLIB_LIBC=y

CONFIG_OPENTHREAD_COAP=y
CONFIG_OPENTHREAD_MQTTSN=y

# Disable TCP and IPv4 (TCP disabled to avoid heavy traffic)
CONFIG_NET_TCP=n
CONFIG_NET_IPV4=n

CONFIG_NET_IPV6_NBR_CACHE=n
CONFIG_NET_IPV6_MLD=n
CONFIG_NET_CONFIG_NEED_IPV4=n

CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL_STACK_SIZE=3072

CONFIG_NET_L2_OPENTHREAD=y

# mbedTLS tweaks
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768

# A sample configuration to enable Thread Joiner, uncomment if needed
CONFIG_OPENTHREAD_JOINER=y
CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
```
Thanks for the help in advance!

oubotong, 2020-08-30:
Here is my source code:
```

#include <string.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdbool.h>

#include "openthread/cli.h"
#include "openthread/instance.h"
#include "openthread/thread.h"
#include "openthread/tasklet.h"
#include "openthread/ip6.h"
#include "openthread/mqttsn.h"
#include "openthread/dataset.h"
#include "openthread/link.h"

#define NETWORK_NAME "OpenThread"
#define PANID 0xFACE

#define GATEWAY_PORT 47193
#define GATEWAY_ADDRESS "fd11:22::1637:7871:d626:bf3e"

#define CLIENT_ID "NRF52840"
#define CLIENT_PORT 47193

static void HandleConnected(otMqttsnReturnCode aCode, void* aContext)
{
    OT_UNUSED_VARIABLE(aCode);
    OT_UNUSED_VARIABLE(aContext);
    printf("Connected\n");
    otInstance *instance = (otInstance *)aContext;
    const char *clientStateString;
    otMqttsnClientStateToString(otMqttsnGetState(instance), &clientStateString);
    printf("MQTT state:%s\n",clientStateString);

}

static void HandleDisconnected(otMqttsnDisconnectType aType, void* aContext)
{
    OT_UNUSED_VARIABLE(aType);
    OT_UNUSED_VARIABLE(aContext);
    printf("Disconnected\n");
}

static void StateChanged(otChangedFlags aFlags, void *aContext)
{
    otInstance *instance = (otInstance *)aContext;
    if ((aFlags & OT_CHANGED_THREAD_ROLE) != 0)
   {
       otDeviceRole changedRole = otThreadGetDeviceRole(aContext);

       switch (changedRole)
       {
       case OT_DEVICE_ROLE_LEADER:
           printf("Leader\n");
           break;

       case OT_DEVICE_ROLE_ROUTER:
           {
           printf("Router\n");
         if(otMqttsnGetState(instance) == kStateDisconnected){
            printf("begin to connect\n");
            otIp6Address address;
            uint16_t port=47193;
            otIp6AddressFromString(GATEWAY_ADDRESS, &address);
            otMqttsnSetConnectedHandler(instance, HandleConnected, (void *)instance);
            otMqttsnSetDisconnectedHandler(instance, HandleDisconnected, (void *)instance);
            otMqttsnConnectDefault(instance,&address,(uint16_t)port);
	 }  
           }
           break;
       
       case OT_DEVICE_ROLE_CHILD:
           {
           printf("Child\n");
           }
           break;
       
       case OT_DEVICE_ROLE_DETACHED:
           printf("Detached\n");
	   break;
       case OT_DEVICE_ROLE_DISABLED:
           printf("Disable\n");
           break;
        }
    }
}

int main(int aArgc, char *aArgv[])
{
    otError error = OT_ERROR_NONE;
    otInstance *instance;

    otSysInit(aArgc, aArgv);
    instance = otInstanceInitSingle();

    // Set default network settings
    // Set network name
    error = otThreadSetNetworkName(instance, NETWORK_NAME);
    // Set PANID
    error = otLinkSetPanId(instance, PANID);
    // Set channel
    error = otLinkSetChannel(instance, DEFAULT_CHANNEL);

    // Register notifier callback to receive thread role changed events
    error = otSetStateChangedCallback(instance, StateChanged, instance);

    // Start thread network
    error = otIp6SetEnabled(instance, true);
    error = otThreadSetEnabled(instance, true);
    error = otMqttsnStart(instance, CLIENT_PORT);

    while (true)
    {
        otTaskletsProcess(instance);
        otSysProcessDrivers(instance);
    }
    return error;
}
```

LuDuda, 2020-09-02:
Hello @oubotong,

As this is more related to the Zephyr / MQTTSN library usage than to OpenThread itself, can you please post this issue [here](https://github.com/zephyrproject-rtos/zephyr/issues), and close this one?

MarekPorwisz, 2020-09-03:
Hi,

Provided code will only work on bare metal OpenThread. Zephyr is an OS and will take care of network support. You don't need to initialize OT instance as it will be initialized by zephyr. also tasklets and drivers processing is handled by Zephyr.
Here is a simplest possible sample of OpenThread in nRF Connect SDK https://github.com/nrfconnect/sdk-nrf/tree/master/samples/openthread/cli :) As you can see main function is completely empty.

It is advised to use zephyr APIs instead of directly using OT API however I'm not sure if there is zephyr API that supports MQTT-SN. Here is another sample that shows how to use OpenThread api in Zephyr (again pointing to NCS): https://github.com/nrfconnect/sdk-nrf/blob/83764f01f04f58c80bb1ae1d3e0a640496f79b49/samples/openthread/coap_server/src/ot_coap_utils.c#L187

Please also note that Thread network setup in zephyr starts even before your main and it is possible that OT is already a router when you call otSetStateChangedCallback. You may need to delay the OpenThread start if it is not desired by specifying CONFIG_OPENTHREAD_MANUAL_START=y and starting it manually after you are set to go (https://github.com/nrfconnect/sdk-nrf/blob/83764f01f04f58c80bb1ae1d3e0a640496f79b49/samples/openthread/coap_client/src/coap_client_utils.c#L268)

If you plan to stick to the nRF devices I would suggest switching from using clean Zephyr to nRF Connect SDK (https://www.nordicsemi.com/Software-and-tools/Software/nRF-Connect-SDK). It uses Zephyr OS as well but extends its functionalities.

closed 2020-09-05 by oubotong