openthread/openthread #5477: Question - Guide for using '#include "filename"' or '#include <filename>"

Irving-cl, 2020-08-30:
Currently I cannot find related information about when to use `#include "..."` and when to use `#include <...>`. Do we have any rules for it? Though it makes no big difference, I think it would be better to unify the usage.

I noticed both usages are in our code for similar include headers. As an example, in `commissioner_api.cpp`:
```
#include "openthread-core-config.h"

#include <openthread/commissioner.h>

#include "common/instance.hpp"
#include "common/locator-getters.hpp"
```
And in `child_supervision_api.cpp`:
```
#include "openthread-core-config.h"
#include "openthread/child_supervision.h"

#include "common/instance.hpp"
#include "common/locator-getters.hpp"
```

When using `#include <filename>`, the compiler would normally search in directories pre-designated by compiler/IDE. So it's usually used to include standard library header files. When using `#include "filename"` the preprocessor searches first in the same directory as the file containing the directive. So I think in openthread, `#include "filename"` should be used in most cases, including the above example. The exception is when including standard library. Thoughts? @jwhui @abtink @bukepo @librasungirl @wgtdkp @simonlingoogle

jwhui, 2020-08-30:
See [STYLE_GUIDE.md](https://github.com/openthread/openthread/blob/master/STYLE_GUIDE.md):

> - Preprocessor #include directives shall use brace (“<”) and (“>”) style for all public headers, including C and C++ standard library, or other first- and third-party public library headers.
> - Preprocessor #include directives should use double quote (‘“‘) and (‘“‘) style for all private or relative headers.

Irving-cl, 2020-08-30:
@jwhui Thanks. Didn't notice that.

closed 2020-08-30 by Irving-cl