openthread/openthread #5492: Floating point in EFR32 builds

theHurc, 2020-09-03:
Is there a particular reason the EFR32 builds use the `-mfloat-abi=soft` option instead of `-mfloat-abi=softfp`? Isn't the EFR32 capable of hardware floating point?

dismirlian, 2020-09-05:
Yes, it's possible, in fact we have been runningr our tests with softfp. It would be even better to use the `hard` abi, if Silabs provided compatible versions of RAIL, which weren't available last time I checked.

romacdon, 2020-09-06:
Thanks for raising the question @theHurc.  You are correct.  For the series1 parts (efr32mg1, efr32mg12, and efrmg13) you can use `-mfloat-abi=softfp` and we will look into changing that in an upcoming PR.  For series2 parts (efr32mg21) we are correctly using `-mfloat-abi=hard`.

Also, thanks @dismirlian for your confirmation.

PR #5666 (2020-10-21): [efr32] common platform abstraction layer
This merges the PAL sources for all efr32-based platforms

# Summary of changes
- `examples/platforms/efr32mg*/` moved inside `examples/platforms/efr32/`
- Added `gecko_sdk_suite/v3.0/platform/radio/rail_lib/plugin/pa-conversions/pa_conversions_efr32.c` to `third_party/silabs/Makefile.am`
  - Helps with merging radio.c
- Closes #5492 - For series1 parts use -mfloat-abi=softfp
- Closes #5502 - UART Initialization overrides hal-config baud-rate value with a constant
- Closes #5493 - remove call to CMU_ClockSelectSet() in system.c
closed 2020-09-08 by theHurc