openthread/openthread #5493: EFR32 Clock set incorrectly

theHurc, 2020-09-03:
https://github.com/openthread/openthread/blob/debdb49b6a99fd59f3d030da28346e372aa37ebf/examples/platforms/efr32mg12/system.c#L91

In the referenced line the low frequency E clock is set to use the low frequency RC clock as it's reference. If you dig through the initialization function in the previous lines you'll see that the low frequency E clock is configured to run from the low frequency external crystal. Since the E clock drives the RTCC wouldn't you want it to use the external crystal which will be far more accurate?

silabs-yassine, 2020-09-03:
Hi,
On the data sheet efr32MG12, the LFRCO consumes more than the LFXO.
LFRCO consumes 370 to 500 nA and LFXO only 273 nA. The LFXAO is a crystal so it is accurate enough for the RTCC. LXAO has also a start up time shorter. So could you please copy paste your codes lines to be sure we have the same version of the file ? Thank you.
Regards.

theHurc, 2020-09-04:
Hi.

I'm sorry I don't quite understand what you're getting at here. You've said `LFXAO ` and `LXAO` which are not valid abbreviations as far as I know which is adding to the confusion. My version of the code just has the line referenced in my original question removed.

silabs-yassine, 2020-09-04:
Hi,
Forget my previous message. So if I understand, your question is:  why did not we set up the E clock with the external crystal ?
You said  **If you dig through the initialization function in the previous lines you'll see that the low frequency E clock is configured to run from the low frequency external crystal**

As you can see in the item below, (link of the data sheet [https://www.silabs.com/documents/public/data-sheets/mgm12p-datasheet.pdf]()), there are two externals crystal available to drive the RTCC. A low and high frequency. But the data sheet also says **_The RTCC can be clocked by any of the on-board oscillators with the exception of the AUXHFRCO_**
On the first page, you can also read **2.62 μA EM2 DeepSleep current (256 kB RAM retention and RTCC running from LFXO)**
So, yes we could have set up the RTCC with the low frequency external crystal and have a better accuracy but we are not supposed to set all example for a deep sleep mode. 

**Item**
The MGM12P fully integrates two crystal oscillators and four RC oscillators, listed below.
• **A 38.4MHz high frequency crystal oscillator (HFXO) provides a precise timing reference for the MCU and radio.
• A 32.768 kHz **crystal** oscillator (**LFXO**) provides an accurate timing reference for low energy modes.**
• An integrated high frequency RC oscillator (HFRCO) is available for the MCU system, when crystal accuracy is not required. The
HFRCO employs fast startup at minimal energy consumption combined with a wide frequency range.
• An integrated auxilliary high frequency RC oscillator (AUXHFRCO) is available for timing the general-purpose ADC and the Serial
Wire debug port with a wide frequency range.
• An integrated low frequency 32.768 kHz RC oscillator (**LFRCO**) can be used as a timing reference in low energy modes, when crystal accuracy is not required.
• An integrated ultra-low frequency 1 kHz RC oscillator (ULFRCO) is available to provide a timing reference at the lowest energy consumption in low energy modes.

Regards.

romacdon, 2020-09-06:
Hi @theHurc - Can you elaborate on this `If you dig through the initialization function in the previous lines you'll see that the low frequency E clock is configured to run from the low frequency external crystal.`?  I just dug through the code and I couldn't find that as the initialization stuff I found was all based on the macro define in hal-config.h, `#define HAL_CLK_LFECLK_SOURCE (HAL_CLK_LFCLK_SOURCE_LFRCO)`.   I just want to make sure we are looking at the same stuff.

theHurc, 2020-09-10:
In `gecko_sdk_suite\v2.7\hardware\kit\common\bsp\bsp_init.c` starting on line 305 I see that the `#define HAL_CLK_LFECLK_SOURCE` macro sets up the the clock source for the E clock.

So let me restate my question:

Why does the line referenced in my original question exist? It's just going to override the whatever was setup with the macro in `bsp_init.c`, right?

romacdon, 2020-09-10:
@theHurc I think I see your point.  If someone where to change the hal-config.h macro from `#define HAL_CLK_LFECLK_SOURCE (HAL_CLK_LFCLK_SOURCE_LFRCO)` to something else the code in `bsp_init.c` would do the proper initialization based on the new value but then would be overridden by the code in system.c.  As things stand with the default hal-config.h settings this is not a problem but could be for custom configurations.  I agree this needs to be updated and we will make this change in an upcoming PR.  Thanks for making us aware of this issue.

PR #5666 (2020-10-21): [efr32] common platform abstraction layer
This merges the PAL sources for all efr32-based platforms

# Summary of changes
- `examples/platforms/efr32mg*/` moved inside `examples/platforms/efr32/`
- Added `gecko_sdk_suite/v3.0/platform/radio/rail_lib/plugin/pa-conversions/pa_conversions_efr32.c` to `third_party/silabs/Makefile.am`
  - Helps with merging radio.c
- Closes #5492 - For series1 parts use -mfloat-abi=softfp
- Closes #5502 - UART Initialization overrides hal-config baud-rate value with a constant
- Closes #5493 - remove call to CMU_ClockSelectSet() in system.c
closed 2020-09-10 by theHurc