openthread/openthread #5502: EFR32MG12 UART Initialisation overrides hal-config baud-rate value with a constant

vijayrao1988, 2020-09-08:
**Description of the bug** 
The EFR32MG12 platform provides the hal-config.h file to define the pin usage, baud rates, etc. Using the HAL_SERIAL_APP_BAUD_RATE and HAL_USART0_BAUD_RATE does not impact the final build.

**To Reproduce**
To reproduce, modify the two definitions mentioned above to different values and the built executables do not reflect the change in the USART baud rate.

**Expected behavior**
USART baud rate must vary as per the definitions in the hal-config file.

**Source of the issue**
In uart.c in examples/platforms/efr32mg12/brd4304a,
#define USART_INIT
uses a constant for the Baud Rate, while the other variables are picked up from the hal-config.h file.
This needs to be corrected.

romacdon, 2020-09-08:
@vijayrao1988 thanks for bringing this to our attention. I agree this looks wrong and we'll update this in an upcoming PR. I tracked it back to this PR from Jan of 2018: https://github.com/openthread/openthread/pull/2401

silabs-yassine, 2020-09-25:
Hi, 
I modified `HAL_SERIAL_APP_BAUD_RATE ` in hal_config.h :
`#define HAL_SERIAL_APP_BAUD_RATE              (9600UL)`

Have you saved your file after modifying it ?
In my side It works properly. Could you please attached your file hal_config.h ?
Regards.

romacdon, 2020-09-28:
@silabs-yassine - please refer to this line: https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/uart.c#L55.  As you can see that line hard codes the value for baud rate. The `USART_INIT` macro where this hard coded value is specified is used to initialize the `uartInit` variable in `otPlatUartEnable`. See https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/uart.c#L155.  The `uartInit` variable is then used when initializing the UARTDRV driver. See https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/uart.c#L160. The suggestion, to which I agree, is to reference the `HAL_SERIAL_APP_BAUD_RATE` when defining the `USART_INIT` macro.

silabs-yassine, 2020-09-28:
@romacdon, I checked all lines that you gave me, thank you for the accuracy, as @vijayrao1988 said, UART baud speed can be done in hal-config.h. It is nice to know how the uart.c file handles the datas :) .

romacdon, 2020-09-28:
@silabs-yassine, just to be clear this has not yet been fixed.  It will be fixed in an upcoming PR.

silabs-yassine, 2020-09-28:
@romacdon thank you :)

PR #5666 (2020-10-21): [efr32] common platform abstraction layer
This merges the PAL sources for all efr32-based platforms

# Summary of changes
- `examples/platforms/efr32mg*/` moved inside `examples/platforms/efr32/`
- Added `gecko_sdk_suite/v3.0/platform/radio/rail_lib/plugin/pa-conversions/pa_conversions_efr32.c` to `third_party/silabs/Makefile.am`
  - Helps with merging radio.c
- Closes #5492 - For series1 parts use -mfloat-abi=softfp
- Closes #5502 - UART Initialization overrides hal-config baud-rate value with a constant
- Closes #5493 - remove call to CMU_ClockSelectSet() in system.c
closed 2020-10-21 by jwhui