openthread/openthread #5510: In EFR32MG12 (and probably other SL flavors) radio power level state variable out of sync

theHurc, 2020-09-10:
https://github.com/openthread/openthread/blob/cf1057b7806f6238404bf5c7340837c7f5a3cc04/examples/platforms/efr32mg12/radio.c#L1134

https://github.com/openthread/openthread/blob/cf1057b7806f6238404bf5c7340837c7f5a3cc04/examples/platforms/efr32mg12/radio.c#L1121

In both the `otPlatRadioSetTransmitPower` and `otPlatRadioGetTransmitPower` implementations a variable `sTxPowerDbm` is used to keep track of the radio power level. But this variable can easily get out of sync because the call `RAIL_SetTxPowerDbm` silently truncates any power level to the nearest power level available for the radio. Why not remove this extra state variable and use `RAIL_GetTxPowerDbm` (divided by 10 of course) to report the actual, set power level for the radio?

romacdon, 2020-09-11:
@theHurc it looks like `sTxPowerDbm` was introduced in this commit https://github.com/openthread/openthread/commit/eb7a9106388a4e43c5e43028a011fe9abe7befe0 and was used in more than just the `otPlatRadioSetTransmitPower` and `otPlatRadioGetTransmitPower`.  However, since then its use in those other locations has been removed so I agree we should eliminate `sTxPowerDbm` and simply call `RAIL_SetTxPowerDbm` and `RAIL_GetTxPowerDbm`.  We can make that change in an upcoming PR or if you want you can make the change.

theHurc, 2020-09-11:
I'll leave it to SL. Thanks.

PR #5701 (2020-10-26): [efr32] remove sTxPowerDbm and use RAIL_GetTxPowerDbm()
Closes #5510
closed 2020-09-11 by theHurc