openthread/openthread #5524: [cli] Add services with binary data

tlifschitz, 2020-09-14:
Hi all!

**Is your feature request related to a problem? Please describe.**
In our application the data related to services are binary protobuf-encoded messages. However the cli command  _service add  \<enterpriseNumber\> \<serviceData\> \<serverData\>_ directly copies data as a string.

**Describe the solution you'd like**
To have the possibility to parse the given data as binary and use the command like this:
`> service add 44970 0xDEADBEEF 0x123456789ABCDEF`

**Describe alternatives you've considered** 
Flag the binary payload with a '0x' prefix and in that case parse it using [Hex2Bin](https://github.com/openthread/openthread/blob/7266e1b909e9c6c147844584ac8518c4c679e86d/src/cli/cli.hpp#L178) in [Interpreter::ProcessService](https://github.com/openthread/openthread/blob/5bffd8bc4dadfedb6ce42c534e4ea33b2a3d1f10/src/cli/cli.cpp#L2391-L2441)
Note this should also be taken into account when removing services:
`> service remove 44970 0xDEADBEEF`

jwhui, 2020-09-15:
@tlifschitz , thanks for raising this issue! I agree that having a way to specify binary data is useful.

Rather than inferring binary from a leading `0x`, how about adding something more explicit in the case that someone truly wants to inject a string that leads with `0x`?

For example something like:
```
> service add-binary 44970 DEADBEEF 123456789ABCDEF
> service remove-binary 44970 DEADBEEF
```
or:
```
> service add 44970 -b DEADBEEF 123456789ABCDEF
> service remove 44970 -b DEADBEEF
```

Thoughts?

tlifschitz, 2020-09-15:
That would be definitely more explicit. However I was thinking in a generic case in which you can specify each individual string as binary. Maybe using the combination of both, the switch `-b` and a leading `0x`, would do the trick:
```
> service add 44970 -b 0xDEADBEEF bar
> service add 44970 -b foo 0x123456789ABCDEF
> service add 44970 -b 0xDEADBEEF 0x123456789ABCDEF
```
Or simply using or not the `-b` option before each string:
```
> service add 44970 -b DEADBEEF bar
> service add 44970 foo -b 123456789ABCDEF
> service add 44970 -b DEADBEEF  -b 123456789ABCDEF
```
Does that make any sense? Anyway this won't be really necessary in our use case, since both payloads are binary.

jwhui, 2020-09-15:
@tlifschitz , I think allowing a mix of string and binary makes sense.

Of the two options, I think I prefer having the `-b` before each option to make each one explicit.

edmont, 2020-09-18:
I came here to open an issue about this and found it already open in the top of the list :)

This is a totally needed feature. My question is, are the string arguments useful at all? Thread Specification doesn't define `S_service_data` or `S_server_data` as strings. I expect them being binary encoded information, same way as BBR Dataset is defined in Thread 1.2 Specification.

The only reason I see to keep string arguments would be backwards compatibility, but I doubt anyone is using them that way.

By the way, it would be useful to have a stand alone `service` command to list the services the device has announced, or the full list of services available in the Network Data.

jwhui, 2020-09-18:
> This is a totally needed feature. My question is, are the string arguments useful at all? Thread Specification doesn't define `S_service_data` or `S_server_data` as strings. I expect them being binary encoded information, same way as BBR Dataset is defined in Thread 1.2 Specification.

@edmont , that's a good point. Only supporting binary sounds good to me.

> By the way, it would be useful to have a stand alone `service` command to list the services the device has announced, or the full list of services available in the Network Data.

#5543 has the start to this. I can add ability to just list the services as well.

I can take a look at making the changes discussed in this issue. Thanks for everyone's input!

edmont, 2020-09-18:
> #5543 has the start to this. I can add ability to just list the services as well.

Having them listed along the rest of the Network Data sounds good to me as well.

jwhui, 2020-09-21:
@tlifschitz @edmont , submitted #5556, please review.

PR #5556 (2020-09-21): [cli] service and server data as hex-encoded binary values
Resolves #5524
closed 2020-09-21 by jwhui