openthread/openthread #5526: EFR32 Joining failure

theHurc, 2020-09-14:
The joining process between two EFR32 wireless dev kits fails.

Load one dev kit with ot-cli-mtd (joiner/child) and one with ot-cli-ftd (leader/commissioner).

## On the leader/commissioner

````
ifconfig up
thread start
commissioner start
commissioner joiner add * AAAABBBB
````

## On the joiner/child

````
ifconfig up
joiner start AAAABBBB
````

## CLI output on leader/commissioner

````
> Commissioner: Joiner start af5570f5a1810b7a
Commissioner: Joiner connect 3e15636229b773af
=========[[THCI] direction=recv | type=JOIN_FIN.req | len=051]==========
| 10 01 01 21 0D 53 4C 2D | 4F 50 45 4E 54 48 52 45 | ...!.SL-OPENTHRE
| 41 44 22 05 45 46 52 33 | 32 23 10 31 2E 30 2E 30 | AD".EFR32#.1.0.0
| 2E 32 20 47 69 74 48 75 | 62 2D 66 25 06 18 B4 30 | .2 GitHub-f%..40
| 00 00 10 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
=========[[THCI] direction=send | type=JOIN_FIN.rsp | len=003]==========
| 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
Commissioner: Joiner finalize 3e15636229b773af
[THCI] direction=send | type=JOIN_ENT.ntf
Commissioner: Joiner end 3e15636229b773af
Commissioner: Joiner start af5570f5a1810b7a
Commissioner: Joiner connect 3e15636229b773af
=========[[THCI] direction=recv | type=JOIN_FIN.req | len=051]==========
| 10 01 01 21 0D 53 4C 2D | 4F 50 45 4E 54 48 52 45 | ...!.SL-OPENTHRE
| 41 44 22 05 45 46 52 33 | 32 23 10 31 2E 30 2E 30 | AD".EFR32#.1.0.0
| 2E 32 20 47 69 74 48 75 | 62 2D 66 25 06 18 B4 30 | .2 GitHub-f%..40
| 00 00 10 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
=========[[THCI] direction=send | type=JOIN_FIN.rsp | len=003]==========
| 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
Commissioner: Joiner finalize 3e15636229b773af
[THCI] direction=send | type=JOIN_ENT.ntf
Commissioner: Joiner end 3e15636229b773af
````

## CLI output on joiner/child

````
> =========[[THCI] direction=send | type=JOIN_FIN.req | len=051]==========
| 10 01 01 21 0D 53 4C 2D | 4F 50 45 4E 54 48 52 45 | ...!.SL-OPENTHRE
| 41 44 22 05 45 46 52 33 | 32 23 10 31 2E 30 2E 30 | AD".EFR32#.1.0.0
| 2E 32 20 47 69 74 48 75 | 62 2D 66 25 06 18 B4 30 | .2 GitHub-f%..40
| 00 00 10 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
=========[[THCI] direction=recv | type=JOIN_FIN.rsp | len=003]==========
| 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
------------------------------------------------------------------------
Join failed [ResponseTimeout]
````

It seems the commissioner thinks the join process worked but the child/joiner has a timeout issue. What's going on here?

romacdon, 2020-09-15:
Hi @theHurc, we tried to recreate this issue but could not so we have a few questions.  1

1. What efr32 part are you using?
2. It appears from the version info that I see in the output that you may be using images built using Simplicity Studio.  Is that correct?  Not that there's anything wrong with that just that it would be good know.
3. What build config options are you using that are different from the default build options?
4. Could you also include a network analyzer trace of the issue?

JaneFromSilabs, 2020-09-15:
Hi @theHurc , Are you sending the device a "thread start" command after the joiner command?

theHurc, 2020-09-15:
1. I'm using 2 boards with the EFR32MG12 (BRD4162A Rev A01)
2. Yes, I found Simplicity Studio v5 and I'm using that.
3. I set the FTD to be the commissioner and the MTD to be a joiner. I also turned on full logging for both.
4. Some logs below

I found part of the issue. It seems that sometimes the FTD device is unable to form a thread network and become leader. Here are the commands:

```
dataset clear
dataset channel 26
dataset channelmask 0x07fff800
dataset panid 0x7447
dataset extpanid f1f802aabd5bef34
dataset masterkey a06fae9b2d269158a09a707d800ee49c
dataset networkname P4_OT_TEST_46
dataset meshlocalprefix fd44:d56a:2a61:5086:0000:0000:0000:0000
dataset securitypolicy 672 onrcb
dataset commit active
dataset active
ifconfig up
thread start
```

And Here is the associated log output. It just repeats over and over. The command `state` always returns `detached`

```
Channel: 26
Channel Mask: 07fff800
Ext PAN ID: f1f802aabd5bef34
Mesh Local Prefix: fd44:d56a:2a61:5086/64
Master Key: a06fae9b2d269158a09a707d800ee49c
Network Name: P4_OT_TEST_46
PAN ID: 0x7447
Security Policy: 672, onrcb
ifconfig up
[INFO]-CLI-----: execute command: ifconfig up
[INFO]-PLAT----: State=OT_RADIO_STATE_RECEIVE
[DEBG]-MAC-----: RadioState: Sleep -> Receive
[DEBG]-MAC-----: Idle mode: Radio receiving on channel 26
Done
> [INFO]-CORE----: Notifier: StateChanged (0x01001009) [Ip6+ LLAddr Ip6Mult+ NetifState]
thread start
[INFO]-CLI-----: execute command: thread start
[NOTE]-MLE-----: Role Disabled -> Detached
[INFO]-PLAT----: ShortAddr=FFFE
[DEBG]-MAC-----: RadioShortAddress: 0xfffe
[INFO]-MLE-----: AttachState Idle -> Start
Done
> [INFO]-CORE----: Notifier: StateChanged (0x00001015) [Ip6+ Role MLAddr Ip6Mult+]
[NOTE]-MLE-----: Attempt to attach - attempt 1, any-partition reattaching with Active Dataset
[INFO]-MLE-----: AttachState Start -> ParentReqRouters
[INFO]-MLE-----: Send Parent Request to routers (ff02:0:0:0:0:0:0:2)
[DEBG]-MAC-----: Request to start operation "TransmitDataDirect"
[DEBG]-MAC-----: Starting operation "TransmitDataDirect"
[DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[INFO]-PLAT----: State=OT_RADIO_STATE_RECEIVE
[DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[DEBG]-MAC-----: RadioState: Transmit -> Receive
==============================[TX len=063]===============================
| 41 D8 08 47 74 FF FF F1 | B8 4C 05 7C 45 46 A2 7F | AX.Gt..q8L.|EF".
| 3B 02 F0 4D 4C 4D 4C AA | C1 00 15 72 1F 00 00 00 | ;.pMLML*A..r....
| 00 00 00 01 D1 28 E8 93 | 29 F1 8C 8E 81 B7 D2 A9 | ....Q(h.)q...7R)
| 51 62 6E B2 A6 C3 ED 1B | 46 CA 1E 49 04 C4 45 .. | Qbn2&Cm.FJ.I.DE.
------------------------------------------------------------------------
[DEBG]-MAC-----: Finishing operation "[INFO]-MLE-----: AttachState ParentReqRouters -> ParentReqReeds
[INFO]-MLE-----: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
[DEBG]-MAC-----: Request to start operation "TransmitDataDirect"
[DEBG]-MAC-----: Starting operation "TransmitDataDirect"
[DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[INFO]-PLAT----: State=OT_RADIO_STATE_RECEIVE
[... 226 lines cut ...]
[DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[INFO]-PLAT----: State=OT_RADIO_STATE_RECEIVE
[DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[INFO]-MLE-----: Send Announce on channel 26
[DEBG]-MAC-----: Request to start operation "TransmitDataDirect"
[DEBG]-MAC-----: Starting operation "TransmitDataDirect"
[DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[INFO]-PLAT----: State=OT_RADIO_STATE_RECEIVE
[DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[DEBG]-MAC-----: RadioState: Transmit -> Receive
==============================[TX len=078]===============================
| 09 D8 1C FF FF FF FF 47 | 74 F1 B8 4C 05 7C 45 46 | .X.....Gtq8L.|EF
| A2 15 50 01 00 00 FF FF | FF FF FF 64 51 BE DC A8 | ".P........dQ>\(
| 78 55 F2 F4 E1 78 25 42 | D5 C4 9C 18 C1 AE C1 49 | xUrtax%BUD..A.AI
| 70 09 AB CD 0A 52 61 1F | 53 4F C4 9C 92 FD 80 DB | p.+M.Ra.SOD..}.[
| DE 68 A0 B6 77 49 52 D3 | 1A 63 1A 22 00 00 .. .. | ^h 6wIRS.c."....
------------------------------------------------------------------------
[DEBG]-MAC-----: Finishing operation "TransmitDataDirect"
[INFO]-MAC-----: Sent IPv6 UDP msg, len:83, chksum:6c8f, to:0xffff, sec:yes, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:a0[INFO]-MLE-----: AttachState Announce -> Idle
[INFO]-PLAT----: ShortAddr=FFFE
[DEBG]-MAC-----: RadioShortAddress: 0xfffe
[INFO]-MLE-----: AttachState Idle -> Start
[NOTE]-MLE-----: Attach attempt 2 unsuccessful, will try again in 0.547 seconds
```

theHurc, 2020-09-15:
Sorry I made a mistake. The MTD is BRD4162A Rev A01 and the FTD is BRD4161A Rev A02

romacdon, 2020-09-15:
Thanks @theHurc -  we'll take a look at the additional info you included and get back to you soon.  Also, with regards to a network analyzer trace.  Can you take a look at section 5.4 in this document - https://www.silabs.com/documents/public/quick-start-guides/qsg170-openthread-sdk-quick-start-guide.pdf.  It describes how to grab a trace.  Thanks.

JaneFromSilabs, 2020-09-15:
It looks to me like the issue is committing the dataset prior to starting thread.  If I commit the dataset prior to thread I am forever in a deteched state on what should be the leader.  But if I commit the dataset after thread has started it all works:
```
> ifconfig up
Done
>
> thread start
Done
>
> state
detached
Done
>
> state
leader
Done
>
> dataset active
Active Timestamp: 0
Channel: 11
Channel Mask: 0x07fff800
Ext PAN ID: dead00beef00cafe
Mesh Local Prefix: fdde:ad00:beef:0::/64
Master Key: 00112233445566778899aabbccddeeff
Network Name: OpenThread
PAN ID: 0x3e11
PSKc: f78a26b0ff7f2597ce5cafaca6da0528
Security Policy: 672, onrcb
Done
>
> dataset commit active
Done
>
> dataset active
Channel: 26
Channel Mask: 0x07fff800
Ext PAN ID: f1f802aabd5bef34
Mesh Local Prefix: fd44:d56a:2a61:5086::/64
Master Key: a06fae9b2d269158a09a707d800ee49c
Network Name: P4_OT_TEST_46
PAN ID: 0x7447
Security Policy: 672, onrcb
Done
>
>
> commissioner start
Commissioner: petitioning
Done
> Commissioner: active
>
>
> commissioner joiner add * S1LABS7
Done
>
> ~ Discovery Request from 3e4025344a09612b: version=2,joiner=1
Commissioner: Joiner start 63e4b719d19f361d
Commissioner: Joiner connect 63e4b719d19f361d
Commissioner: Joiner finalize 63e4b719d19f361d
Commissioner: Joiner end 63e4b719d19f361d
Commissioner: Joiner remove
```

And here is my joiner:
```
> factoryreset
channel 26
Done
>
> panid 0x7447
Done
>
> extpanid f1f802aabd5bef34
Done
>
> ifconfig up
Done
>
> joiner start S1LABS7
Done
>
> thread start
Done
>
> Join success
```

@jwhui Is there a reason why the "dataset commit active" command needs to come after the "thread start" command?

jwhui, 2020-09-15:
@theHurc @JaneFromSilabs , thanks for raising and investigating this issue.

The problem stems from how OpenThread interprets the "Active Timestamp" value. A value of 0 indicates that the Operational Dataset is not valid. As a result, the device can never become a leader on its own. To retrieve a valid Operational Dataset, the device first needs to successfully attach to an existing network.

If you really want to create a valid Active Operational Dataset, you should start with the following command:
```
dataset init new
```
Then modify the various parameters if you desire.

Hope that helps.

JaneFromSilabs, 2020-09-15:
@jwhui Thanks very much for your response.  

I used "dataset init new" in place of "dataset clear" and this configuration worked for me.

```
> dataset commit active
Done
>
> dataset active
Active Timestamp: 1
Channel: 26
Channel Mask: 0x07fff800
Ext PAN ID: f1f802aabd5bef34
Mesh Local Prefix: fd44:d56a:2a61:5086::/64
Master Key: a06fae9b2d269158a09a707d800ee49c
Network Name: P4_OT_TEST_46
PAN ID: 0x7447
PSKc: dec62c59f0254bf49bc48a3d197c359d
Security Policy: 672, onrcb
Done
>
> ifconfig up
Done
>
> thread start
Done
>
> state
leader
Done
```

jwhui, 2020-09-15:
@JaneFromSilabs , yes, `dataset clear` will work if you specify every value individually. The risk is that the Operational Dataset may not have some values specified.

`dataset init new` will ensure that every parameter has some value by initializing each parameter to some value.

theHurc, 2020-09-15:
I'm just getting started with Open Thread and some of the Dataset stuff is very confusing.What is the difference between the Operational Dataset, Active Dataset, and Pending Dataset and when are they used? What's the difference between the commands `dataset channel 12` and `channel 12`? When does the Active Dataset become the Operational Dataset? Or the Pending become the Active? I looked through the documentation and it doesn't say much about this.

Also, why isn't there any bounds checking in the CLI for the dataset commands? You can easily set a Mesh Local Prefix that is invalid (not fd00::/8) or a channel value that isn't valid. Then the Thread network won't work correctly. No specific error is returned though.

jwhui, 2020-09-15:
1. An Operational Dataset is the object that encapsulates a Thread network configuration.
2. An Active Operational Dataset is the Operational Dataset that is currently in use by the Thread device.
3. A Pending Operational Dataset is an Operational Dataset that is scheduled to be used some time in the near future. Pending operational Datasets are useful when changing Thread network parameters that may affect connectivity (e.g. channel, master key, etc.).
4. Often times, we say Active Dataset or Pending Dataset as short forms of Active Operational Dataset and Pending Operational Dataset, respectively.
5. The `dataset` command manipulates Operational Datasets.
6. The `channel` command manipulates the MAC channel directly (completely bypassing Operational Dataset). The `channel` command is useful for RCP, where the SoC only implements MAC/PHY.
7. As for parameter value validation, for completeness, we'd need to validate the Operational Dataset to ensure it is consistent. For example, the Channel must be a valid within the provided Channel Mask. That would imply updating the commands to require specifying multiple parameters at a time.
8. Full details regarding Operational Datasets are available in the [Thread 1.1 Specification](https://www.threadgroup.org/support#specifications).

Hope that helps.

theHurc, 2020-09-15:
@jwhui  That help a lot. Thanks!

schinchani, 2020-09-15:
Hi @theHurc 

The steps mentioned in the following link should be the precise steps required for the join process: https://openthread.io/guides/build/commissioning

I tried reproducing the issue at my end (on two 4161 nodes), for simplicity studio and GitHub builds but was not able to reproduce it. 

To investigate the issue further, can you please reply back with the following information:
- With the `dataset init new` command executed, are you still observing issues forming a Thread network?
- Once the network is formed successfully, do you still see the join issue? If yes, are you able to consistently reproduce the Response Timeout failure, or does it occur sporadically? I would recommend running a `factoryreset` command on the joiner, at the very beginning just to have a clean slate.

Thanks,
Sagar

theHurc, 2020-09-16:
Hey Sagar,

I was able to get the Thread network formation figured out. It was that I had the active timestamp set to 0.

Joining was consistent after I figured out that the Active Dataset must be completely clear. A `factoryreset` is one way to do this.

Thanks,
Jake

theHurc, 2020-09-16:
Hey again everyone,

After digging in deeper to the joining process I'm noticing some funkiness so I'm re-opening the issue. There are quite a few issues I'm seeing. It's possible I'm using the `ot` interface incorrectly, or calling something out of order. I'm not quite sure but the issues are concerning.

I got tired of typing in the CLI commands manually so I created files to add to the leader/commissioner (FTD)  and another for the child/joiner (MTD). I've added those files here. The both applications print very detailed information for debugging purposes. Just add the public function from `ServerApplication.h` to the FTD loop in main and the public function from `EdgeNodeApplication.h` to the MTD super loop in main. As before my MTD is BRD4162A Rev A01 and the FTD is BRD4161A Rev A02; both are EFR32MG12* chips.

The super loop for the FTD looks like this:

```
while (!otSysPseudoResetWasRequested())
{
    otTaskletsProcess(instance);
    otSysProcessDrivers(instance);
    RunServerApplication(instance);
}
```

The super loop for the MTD would look like this:

```
while (!otSysPseudoResetWasRequested())
{
    otTaskletsProcess(instance);
    otSysProcessDrivers(instance);
    RunEdgeNodeApplication(instance);
}
```

## FTD Application Information

On startup the FTD application generates a random, valid `otOperationalDataset`, sets it as the active DataSet, starts the interface and then the Thread network, waits to become leader, turns on the commissioner, and adds the join key `AAAABBBB` as a joiner with a 120 second timeout. Every time the joiner key expires it re-adds it. Reset the FTD to generate a new Thread network.

It should be equivalent to the CLI commands:

```
dataset init new
dataset commit active
ifconfig up
thread start
state // Repeated until leader
commissioner start
commissioner joiner add * AAAABBBB 120 //And doing this again and again after the key expires
```

## MTD Application Information

On startup the MTD application clears the active dataset (so the join process will work), turns on the network interface, then attempts to join with the key `AAAABBBB`. No matter the outcome of the join the application stops after this.  Resetting the MTD force another join attempt.

It should be equivalent to the CLI commands:

```
factoryreset
ifconfig up
joiner start AAAABBBB
```

## Issues I've seen:

- After a very short join process the MTD  sometimes reports a `OT_ERROR_NOT_FOUND` error. Seems too short
- Sometimes the `joinerCallback` in the MTD  doesn't get called at all and seems to hang forever. Yikes!
- The MTD  sometimes reports a `OT_ERROR_RESPONSE_TIMEOUT`
- The MTD  sometimes reports a `OT_ERROR_SECURITY`

As a side note: I've been using the Open Thread border router with an EFR32MG12 NCP for more than a year now and I've seen joining issues but have never had a chance to narrow down the issue. Hopefully this can help.

Please let me know if anyone can reproduce my issues and we can go from there.

[FTD_MTD_Test_Applications.zip](https://github.com/openthread/openthread/files/5234808/FTD_MTD_Test_Applications.zip)

Thanks,
Jake

theHurc, 2020-09-16:
One correction:

I see the `OT_ERROR_NOT_FOUND ` issue with the code I sent. The boards are sitting right next to each other. I don't think a join should ever fail.

If you comment line 391 in `EdgeNodeApplication.c` so the MTD actually attaches to the thread network as a child, then on subsequent resets you'll see the next 3 issues I mentioned. I don't think this is good. If a device is a child on a Thread network it shouldn't be prevented from re-joining should it? It's not impossible the device may reset or lose communication.

Thanks

jwhui, 2020-09-16:
@theHurc , my guess is that the PAN ID is still configured when you try to start the joining process. You can use `otLinkSetPanId()` to reset the PAN ID back to the invalid PAN ID (i.e. 0xffff).

When the PAN ID is set, the device will only discover neighbors with the same PAN ID. This is intended to allow devices to filter on PAN ID if they have some out-of-band way of knowing the PAN ID.

Hope that helps.

theHurc, 2020-09-17:
Is there a misuse of the ot API in the code I shared? I did exactly what a `factoryreset` does minus the chip reset: `otInstanceErasePersistentInfo`. A subsequent call to  `otDatasetGetActive` yields `OT_ERROR_NOT_FOUND`. Same thing you get after `factoryreset` which is what the SL team suggested I do.

I don't think this would explain the joining errors I see - `OT_ERROR_RESPONSE_TIMEOUT` and `OT_ERROR_SECURITY` - after the MTD is allowed to attach to the Thread network as a child, then reset, and attempts to join again.

jwhui, 2020-09-17:
> Is there a misuse of the ot API in the code I shared? I did exactly what a `factoryreset` does minus the chip reset: `otInstanceErasePersistentInfo`. A subsequent call to `otDatasetGetActive` yields [...]

The chip reset is an important missing piece. Calling `otInstanceErasePersistentInfo()` only erases the non-volatile storage. It does not, for example, reset the PAN ID to the invalid value in the MAC. You can validate whether this is occurring with `otLinkGetPanId()`.

theHurc, 2020-09-17:
I see. The API for the data sets and joining is very confusing to me. What is the Active DataSet I need in order to attempt to join on all channels 11 to 26?

jwhui, 2020-09-17:
> I see. The API for the data sets and joining is very confusing to me. What is the Active DataSet I need in order to attempt to join on all channels 11 to 26? I understand that Pan ID needs to be 0xFFFF but what about the channel or the channel mask. Anything else?

If you are calling `otInstanceErasePersistentInfo()`, then there is no Active Dataset.

Calling `otJoinerStart()` should always initiate a scan across all available channels.

See the call to `DiscoverScanner::Discover` - it passes an empty channel mask that indicates try all channels and the MAC PAN ID. These are **not** obtained from the Active Dataset.

https://github.com/openthread/openthread/blob/04ed75eb535aac6836a7df10ae1dc9f57f69555f/src/core/meshcop/joiner.cpp#L174-L176

theHurc, 2020-09-18:
Thanks for the response @jwhui!

It seems that when my device starts up I need to call `otInstanceErasePersistentInfo` and then `otLinkSetPanId(aInstance, 0xFFFF)` to get the subsequent call to `otJoinerStart` to search for a network on all channels and successfully join. The joining process is now stable. Thank you very much for your help.

My experience has revealed some documentation issues:

First, the instructions for the joining process here [Thread Commissioning](https://openthread.io/guides/build/commissioning) will really only succeed on a brand new device with no state saved. If a device has joined a network previously, then the user tries to join a different network (with a different pan ID), the joining process won't succeed. I don't know if the Open Thread community has an interest in updating that process in section 4 to be something like:

```
factoryreset
panid 0xFFFF
ifconfig up
joiner start J01NME
```

Second, the documentation on the `otJoinerStart` call itself is not great. It doesn't say anything about the state of the active dataset, the saved persistent info, or the PAN ID being relevant factors in this call. You have to dig deep into the source code to figure this out. An interface that is difficult to use is a huge code smell for me and having experienced these frustrations first hand I would guess others have struggled getting the joining process to behave as expected.

I'm new to this community so I'd appreciate some feedback on these documentation ideas as it aligns with community standards and/or established norms.

theHurc, 2020-09-18:
Ok, Now for the next issue....

My device can now successfully join a network consistently and then attach to that network as a child. If I reset the device and it restarts the joining process I now see a `OT_ERROR_RESPONSE_TIMEOUT`. Is this expected or intended behavior? My understanding is that the joining process allows a device to get the full thread network information so it can attach to that thread network. Seems to me the attachment state of a device shouldn't affect if a device is allowed to join or not but I'm not sure.

jwhui, 2020-09-18:
> First, the instructions for the joining process here [Thread Commissioning](https://openthread.io/guides/build/commissioning) will really only succeed on a brand new device with no state saved. If a [...]

Thanks for pointing this out. We have the `factoryreset` documented in other places such as [src/cli/README_COMMISSIONING.md](https://github.com/openthread/openthread/blob/master/src/cli/README_COMMISSIONING.md#start-joiner). We'll get the openthread.io docs updated.

> Second, the documentation on the `otJoinerStart` call itself is not great. It doesn't say anything about the state of the active dataset, the saved persistent info, or the PAN ID being relevant [...]

The only real thing that matters here is the PAN ID configured on the MAC. The Active Dataset persistently stored comes into play simply because the MAC PAN ID after boot will reflect what's stored in the Active Dataset. We can update the API doc to make clear that the MAC PAN ID is used to filter Discovery Responses.

> I'm new to this community so I'd appreciate some feedback on these documentation ideas as it aligns with community standards and/or established norms.

We're always looking for was to improve. Any feedback and contributions are welcome!

jwhui, 2020-09-18:
> My device can now successfully join a network consistently and then attach to that network as a child. If I reset the device and it restarts the joining process I now see a [...]

I have submitted #5550 that should allow the joiner process to succeed even if the device has been commissioned. Please review.

theHurc, 2020-09-21:
Perfect. Thanks.

closed 2020-09-21 by theHurc

schinchani, 2020-09-21:
Thank you @jwhui  for addressing @theHurc's follow-up questions. Really appreciate your quick and detailed responses.

jwhui, 2020-09-21:
> Thank you @jwhui for addressing @theHurc's follow-up questions. Really appreciate your quick and detailed responses.

@schinchani , my pleasure!

rohithanayady, 2021-06-18:
> Hey again everyone,
>
> After digging in deeper to the joining process I'm noticing some funkiness so I'm re-opening the issue. There are quite a few issues I'm seeing. It's possible I'm using the `ot` [...]
I have some doubts regarding this..
1. where we have to add the these files `ServerApplication.h` and `ServerApplication.c` aslo  `EdgeNodeApplication.h`  and `EdgeNodeApplication.c`.(Can you tell me the process before Build)
2. When I used this  `EdgeNodeApplication.c` file to join a device as router (after making some changes),it show join success but it is not attached as Router/Child

[routerapplication.zip](https://github.com/openthread/openthread/files/6676885/routerapplication.zip)

Hoping a solution for  the two issues

PR #5550 (2020-09-21): [mac] only apply neighbor check when key id mode is 1
Allows receiving a Joiner Entrust message even after the device has been commissioned.

Resolves #5526