openthread/openthread #5527: /96 based prefixes assign addresses but are failing to send out any udp/ping packets, where /64 works fine

JaneFromSilabs, 2020-09-14:
**Describe the bug** A clear and concise description of what the bug is.
I have a simple 2 node network with a BR and a joiner.  On my border router, I've set up a configuration with a /96 prefix:
```
> channel 11
Done
> panid 0xdf47
Done
> networkname R0GUE7
Done
> masterkey 46112233445566778899aabbccddeeff
Done
> extpanid 000df47000000000
Done
> prefix add 2002:222::/96 paros med
Done
> ifconfig up
Done
> thread start
Done
> state
leader
Done
> ipaddr
2002:222:0:0:6041:9b23:bd58:1260
fd00:df4:7000:0:0:ff:fe00:fc00
fd00:df4:7000:0:0:ff:fe00:2800
fd00:df4:7000:0:fe7c:30d0:47fa:4f8a
fe80:0:0:0:6c6a:ac71:e8af:1428
Done
```
On my other device:
```> channel 11
Done
>
> panid 0xdf47
Done
>
> masterkey 46112233445566778899aabbccddeeff
Done
>
> extpanid 000df47000000000
Done
>
> ifconfig up
Done
>
> thread start
Done
>
> ipaddr
2002:222:0:0:e100:ae08:5ccf:aeb5
fd00:df4:7000:0:0:ff:fe00:2801
fd00:df4:7000:0:93c1:b231:4f06:cf3f
fe80:0:0:0:dc7b:9bc:6a6f:f23f
Done
> udp open
Done
>
> udp bind :: 4242
Done
>
> ping 2002:222:0:0:6041:9b23:bd58:1260 0 1 1000
Done
```
Back on the OTBR:
```> ping 2002:222:0:0:e100:ae08:5ccf:aeb5 0 1 1000
Done
> ping 2002:222:0:0:e100:ae08:5ccf:aeb5 0 1 1000
Done
> udp send 2002:222:0:0:e100:ae08:5ccf:aeb5 4242 hello
Error 1: Failed
```
On the raspi:
```
pi@rpi1:~ $ ip -6 route show
2001:db8:1:ffff::/96 dev nat64 metric 1024  pref medium
2002:222::6041:9b23:0:0/96 dev wpan0 proto kernel metric 256  pref medium
fd00:df4:7000::/64 dev wpan0 proto kernel metric 256  pref medium
fdaa:bb:1::2 dev nat64 proto kernel metric 256  pref medium
fe80::/64 dev eth0 proto kernel metric 256  pref medium
fe80::/64 dev nat64 proto kernel metric 256  pref medium
fe80::/64 dev wpan0 proto kernel metric 256  pref medium
pi@rpi1:~ $ ping 2002:222:0:0:e100:ae08:5ccf:aeb5
connect: Network is unreachable
pi@rpi1:~ $ ifconfig
<clipped>
wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet6 fd00:df4:7000::ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        inet6 fd00:df4:7000::ff:fe00:2800  prefixlen 64  scopeid 0x0<global>
        inet6 2002:222::6041:9b23:bd58:1260  prefixlen 96  scopeid 0x0<global>
        inet6 fe80::6c6a:ac71:e8af:1428  prefixlen 64  scopeid 0x20<link>
        inet6 fd00:df4:7000:0:fe7c:30d0:47fa:4f8a  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 54  bytes 4874 (4.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 12  bytes 1864 (1.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
pi@rpi1:~ $ ping 2002:222::6041:9b23:bd58:1260
PING 2002:222::6041:9b23:bd58:1260(2002:222::6041:9b23:bd58:1260) 56 data bytes
64 bytes from 2002:222::6041:9b23:bd58:1260: icmp_seq=1 ttl=64 time=0.174 ms
64 bytes from 2002:222::6041:9b23:bd58:1260: icmp_seq=2 ttl=64 time=0.140 ms
```

In the network trace, I never see these requests getting sent.  However, if I switch to a /64 prefix, all of this works as expected.  

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 
pi@rpi1:~/ot-br-posix $ git show
commit e3cbbf74a8b65e0657dda6928382953756d6161d

2. IEEE 802.15.4 hardware platform
Silabs brd4161a on both devices

jwhui, 2020-09-15:
@JaneFromSilabs , this is not expected to work.

[RFC 4944 Section 6](https://tools.ietf.org/html/rfc4944#section-6) states the following:
```
   An IPv6 address prefix used for stateless autoconfiguration [RFC4862]
   of an IEEE 802.15.4 interface MUST have a length of 64 bits.
```

We could add a check that returns an error when trying to configure an IPv6 address prefix for SLAAC that is not 64 bits.

JaneFromSilabs, 2020-09-15:
@jwhui Thank you for the information.  I do agree that the configuration should be rejected since this prefix was picked up and used by the joiner.

PR #5530 (2020-10-23): [api] check on prefix length when slaac is enabled
Resolves https://github.com/openthread/ot-br-posix/issues/560
Resolves #5527
closed 2020-10-23 by jwhui