openthread/openthread #5554: "PRIx64" compile error on Ubuntu 20

nealjack, 2020-09-20:
**Describe the bug** 
Compiling openthread libraries and examples fails with the following error:
```
/openthread/examples/../src/cli/cli_joiner.cpp: In member function 'otError ot::Cli::Joiner::ProcessDiscerner(uint8_t, char**)':
/openthread/examples/../src/cli/cli_joiner.cpp:76:38: error: expected ')' before 'PRIx64'
   76 |         mInterpreter.OutputLine("0x%" PRIx64 "/%u", discerner->mValue, discerner->mLength);
      |                                ~     ^~~~~~~
      |                                      )
  CXX      libopenthread_cli_ftd_a-cli_network_data.o
```

It appears to be an issue with the PRIx64 macro and inttypes.h/cinttypes. Replacing the define with `<cinttypes>`, as well as defining the macro `__STDC_FORMAT_MACROS` as mentioned [here ](https://github.com/Microsoft/rDSN/issues/197) did not fix this issue.

**To Reproduce**

1. Git commit id: **0d8a99b**
2. IEEE 802.15.4 hardware platform: **NRF52840**
3. Build steps:
I am building on a newly created Ubuntu 20 VM, using these commands:
```
./script/bootstrap
./bootstrap
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COMMISSIONER=1 DISABLE_BUILTIN_MBEDTLS=1 DNS_CLIENT=1 DIAGNOSTIC=1 EXTERNAL_HEAP=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 COAPS=1 DHCP6_SERVER=1 DHCP6_CLIENT=1
```
4. Network topology: N/A

**Expected behavior** 
I would expect the build to complete successfully.

bukepo, 2020-09-21:
What compiler and version are you using?

nealjack, 2020-09-21:
I am using gcc/g++:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

```

jwhui, 2020-10-06:
@nealjack , thanks for raising this issue. I just submitted #5611. Can you indicate whether this change fixes the issue?

nealjack, 2020-10-06:
Hi @jwhui, I just tried to compile on the same VM as before, using the branch that #5611 is based on, and am still having a compile error. I fetched and checked out the `cli` branch, reran the boostrap scripts, made clean, and rebuilt with the same make command as before.

The error is different however, and is complaining that `<cinttypes>` is not defined. Is there some additional package for gcc/g++ that I must install?

```
~/openthread/examples/../src/cli/cli_joiner.cpp:36:10: fatal error: cinttypes: No such file or directory
   36 | #include <cinttypes>
```

I also tried this branch on my main development machine, which is using the same 9.2.1 gcc/g++ compilers and am still getting the same issue with `PRIx64`.

lord-potato, 2020-10-06:
Hi, I'm having the same issue (EFR32 platform, same compiler version, also ubuntu 20.04), and #5611 didn't solve the error.

lmnotran, 2020-10-06:
I am still having this issue as well

lmnotran, 2020-10-06:
Just as a sanity check, I did the following:

1. Spun up a `ubuntu:20.04` docker container
  a. my clone of `openthread:master` was mounted to `/openthread` in the container. This includes the setup needed for the efr32 parts
2. Ran `./script/bootstrap` (removed all instances of `sudo` since i was `root`)
3. Ran `./bootstrap && make -f examples/Makefile-efr32mg21 BOARD=BRD4180A`

To my surprise, the docker container builds fine. I'm a bit lost. Can anyone else try to reproduce with a fresh Ubuntu 20.04 docker container?

lord-potato, 2020-10-06:
@lmnotran try with `JOINER=1`

lmnotran, 2020-10-07:
@lord-potato Oh oops. I totally forgot that. Thanks for the reminder! 😄 

I just tried again with `JOINER=1` and with @jwhui 's changes in #5611 and got the error @nealjack  is getting 
```
~/openthread/examples/../src/cli/cli_joiner.cpp:36:10: fatal error: cinttypes: No such file or directory
   36 | #include <cinttypes>
```

jwhui, 2020-10-07:
@nealjack @lmnotran @lord-potato , I've tracked it down to an issue with the toolchain installation. In #5613, I've updated `script/bootstrap` to download directly from developer.arm.com. Please review.

lord-potato, 2020-10-07:
I'm getting the same error with #5613 in a fresh ubuntu docker container
```
/ot/examples/../src/cli/cli_joiner.cpp: In member function 'otError ot::Cli::Joiner::ProcessDiscerner(uint8_t, char**)':
/ot/examples/../src/cli/cli_joiner.cpp:76:38: error: expected ')' before 'PRIx64'
   76 |         mInterpreter.OutputLine("0x%" PRIx64 "/%u", discerner->mValue, discerner->mLength);
      |                                ~     ^~~~~~~
      |                                      )
```

abtink, 2020-10-07:
Submitted https://github.com/openthread/openthread/pull/5615 which replaces the `PRIx64` use in the CLI code.

I recall, in some earlier project (not OpenThread), we faced similar issues with `PRIx64` not being defined (or in one case defined but not behaving correctly) on some of the older embedded toolchains.

While using these should technically help with portability of the code, practically (and unfortunately) it rather adds a limitation and makes the code less portable. I would suggest we try to avoid using `PRi{len}` macros when/if we can.

lmnotran, 2020-10-07:
@abtink #5615 works for me. 

Thanks for the quick help everyone! 😃

closed 2020-10-07 by jwhui

bukepo, 2022-11-03:
It seems `PRi{len}` is c++11 and c99 standard, and OpenThread forces C++11 and c99. Do we want to support compiler not compatible with them?

* c++: https://en.cppreference.com/w/cpp/header/cinttypes
* c: https://en.cppreference.com/w/c/types/integer

PR #5613 (2020-10-07): [script] fix arm-gcc toolchain install on ubuntu
Fixes #5554