openthread/openthread #5568: [nRF52840] device easily hangs for 1.2 build when continuous traffic since #5184

librasungirl, 2020-09-24:
**Describe the bug** 
When testing continuous Multicast traffic (MA with scope = 5), nRF52840 DK with 1.2 firmware easily hang
```
(gdb) bt
#0  0x0003f958 in mutex_lock ()
#1  0x00031508 in ProcessAesDrv ()
#2  0x00030ebc in cc310_mbedtls_internal_aes_encrypt ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

J-Link RTT Viewer logs
```
00> [0000050596] [DEBG]-M[0000050603] [DEBG]-MAC-----: Request to start operation "TransmitDataDirect"
00> [0000050603] [DEBG]-MAC-----: Starting operation "TransmitDataDirect"
00> [0000050603] [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
00> [0000050603] [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
00> [0000050604] [DEBG]-MAC-----: Received frame from short address 0x1400
00> [0000050604] [DEBG]-MAC-----: Rx security - frame counter 41128
```

**To Reproduce** Information to reproduce the behavior, including
1. Git commit id
`master branch since #5184`

2. IEEE 802.15.4 hardware platform
`nRF52840 DK`

3. Build steps
`make -f examples/Makefile-nrf52840 USB=1 THREAD_VERSION=1.2 DUA=1 MLR=1 FULL_LOGS=1 LOG_OUTPUT=PLATFORM_DEFINED`

**Additional context** 
The issue occurs since #5184

brianljt, 2020-09-24:
Thanks for pointing this out @librasungirl . From backtrace, looks like it is caused by duplicated access to cc310. When a frame is received, it will use CC310 to do decryption. However, in #5184, the enh-ACK will be generated in interrupt context and will also use CC310 to do encryption. A possible solution is to use software AES instead of CC310 if the mutex cannot be acquired.

simonlingoogle, 2020-09-24:
> A possible solution is to use software AES instead of CC310 if the mutex cannot be acquired.

Is it possible not to do decryption in interrupt handler but post it to the main loop?

brianljt, 2020-09-24:
> > A possible solution is to use software AES instead of CC310 if the mutex cannot be acquired.
> 
> Is it possible not to do decryption in interrupt handler but post it to the main loop?

Enh-ACK is time critical and cannot withstand the latency of main loop task scheduling. So this should be handled in interrupt context

closed 2020-09-30 by jwhui

Feng573795502, 2020-12-22:
> Enh-ACK is time critical and cannot withstand the latency of main loop task scheduling. So this should be handled in interrupt context

如果使用nrf的ses开发软件时候需要替换掉cc310 不知道要修改哪些静态链接库