openthread/openthread #5570: Update EFR32 to Gecko SDK 3.0 and redundant implementation cleanup

theHurc, 2020-09-24:
The current Open Thread EFR32 builds reference Gecko SDK 2.7. However, Silicon Labs have done a beta release of Simplicity Studio v5 which now targets the Gecko SDK 3.0. The interface for the initialization and sleep timing are quite a bit different than old Gecko SDK 2.7. There are a few outstanding bugs that will be irrelevant with the new SDK 3.0; #5493 and possibly #5502 for example.

As an interested party I would like to see the Open Thread versions in this repository synced up with what is supported in Simplicity Studio v5. Otherwise support for the EFR32 implementations are going to quickly get out of hand.

While this is cleanup is taking place Silicon Labs may as well remove redundant source files. The `nrf528xx` implementations, for example have a `src/` file containing all of the platform specific implementation files. There's no reason there couldn't be a similar structure in a new generic `efr32mgxx/src` directory. As an example, right now these files are exactly the same:

`efr32mg1/alarm.c`
`efr32mg12/alarm.c`
`efr32mg13/alarm.c`
`efr32mg21/alarm.c`

Any change to `alarm.c` has to be changed in 4 places. This goes for most other files as well.

romacdon, 2020-09-24:
Hi @theHurc - please see this PR which was merged a week ago: https://github.com/openthread/openthread/pull/5529 It brings moves the GSDK support from 2.7 to 3.0.  We are in the process of working on a single set of files for all efr32 platforms.  You should see that PR shortly.

theHurc, 2020-09-24:
Oh great! I must have missed that. Thanks.

closed 2020-09-24 by theHurc

romacdon, 2020-09-24:
No worries @theHurc.  We do very much appreciate the feedback.  Keep your eyes open for the PR which reduces the platform files down to a single set of files.  We are getting very close to completion of that work.

lmnotran, 2020-10-19:
@theHurc Just a heads up: The PR that merges all the efr32 platforms is in review #5666

theHurc, 2020-10-20:
That's great. Thanks!