openthread/openthread #5594: OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN does not seem to have any effect

Feng573795502, 2020-09-29:
No matter what value I configure, the parent is still far away, and the nearer cannot become the parent. The signal strength at the far end is -81dbm, and the signal strength at the near is -31dbm.

NRF52840+2401C  

f45b588e119491d8fd365e584d210ab49e0aae6a

jwhui, 2020-09-29:
@Feng573795502 , I suspect the reason why your device is not attaching is unrelated to `OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN`.

Can you provide more information as below?

**Describe the bug** A clear and concise description of what the bug is.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2020-10-14:
Closing stale issue.

closed 2020-10-14 by jwhui