openthread/openthread #5609: [question] error using  "PRIx64" when compiling some platforms

lmnotran, 2020-10-06:
# Description

I get the following error when building the platforms below. **This happens on one of my workstations and I am unable to reproduce on any other workstations.** 

## Error

```make
Making all in cli
  CXX      libopenthread_cli_ftd_a-cli.o
...
  CXX      libopenthread_cli_ftd_a-cli_joiner.o
/home/mason/repos/openthread/examples/../src/cli/cli_joiner.cpp: In member function 'otError ot::Cli::Joiner::ProcessDiscerner(uint8_t, char**)':
/home/mason/repos/openthread/examples/../src/cli/cli_joiner.cpp:76:38: error: expected ')' before 'PRIx64'
   76 |         mInterpreter.OutputLine("0x%" PRIx64 "/%u", discerner->mValue, discerner->mLength);
      |                                ~     ^~~~~~~
      |                                      )
make[4]: *** [Makefile:809: libopenthread_cli_ftd_a-cli_joiner.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/home/mason/repos/openthread/examples/../src/cli/cli_joiner.cpp: In member function 'otError ot::Cli::Joiner::ProcessDiscerner(uint8_t, char**)':
/home/mason/repos/openthread/examples/../src/cli/cli_joiner.cpp:76:38: error: expected ')' before 'PRIx64'
   76 |         mInterpreter.OutputLine("0x%" PRIx64 "/%u", discerner->mValue, discerner->mLength);
      |                                ~     ^~~~~~~
      |                                      )
make[4]: *** [Makefile:949: libopenthread_cli_mtd_a-cli_joiner.o] Error 1
make[3]: *** [Makefile:433: all-recursive] Error 1
make[2]: *** [Makefile:586: all-recursive] Error 1
make[1]: *** [Makefile:512: all] Error 2
make: *** [examples/Makefile-efr32mg12:295: do-build-cortex-m4] Error 2
```

# To Reproduce

### Git commit id
3d22a37267a198c4077e4de2f24521391c4ad3bc (head of master, though it doesn't seem to matter)
### IEEE 802.15.4 hardware platform
- efr32mg12
- efr32mg13
- efr32mg21
- jn5189
- k32w061
### Build steps
```
make -f examples/Makefile-<PLATFORM> 
```

# Expected behavior
The platforms should build successfully

# Machine info
```
➜  openthread git:(master) ✗ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

➜  openthread git:(master) ✗ arm-none-eabi-g++ --version
arm-none-eabi-g++ (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

➜  openthread git:(master) ✗ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

```

jwhui, 2020-10-06:
@lmnotran , thanks for raising this issue. I believe this is a duplicate of #5554. That said, @nealjack , I just submitted #5611. Can you indicate whether this change fixes the issue?

nealjack, 2020-10-06:
@jwhui the change did not fix my issue - please see the details at #5554

lmnotran, 2020-10-06:
Closing as a duplicate of #5554

closed 2020-10-06 by lmnotran