openthread/openthread #5612: Radio ON and Radio OFF callbacks

theHurc, 2020-10-06:
Maybe this already exists but I'm not finding it. It's common to have to toggle an external gpio to turn on a regulator to power a power amplifier which powers the radio. Then to need to turn that regulator off after the usage is finished to conserve power. Does this functionality already exist? If not, it would be nice to add this type of feature to the interface.

jwhui, 2020-10-07:
@theHurc , FEM support should reside within platforms-specific code. For example, the [nRF IEEE 802.15.4 radio driver has support includes FEM support](https://infocenter.nordicsemi.com/index.jsp?topic=%2F15.4_radio_driver_v1.8.0%2Frd_fem.html).

theHurc, 2020-10-07:
Ooooh. That's what FEM is for. Ok. Thanks.

closed 2020-10-07 by theHurc