openthread/openthread #5619: Assert failure in HandleTransmitDone with operation kOperationIdle

eujean, 2020-10-08:
**Description** 
Occassionaly while using the ot-cli-ftd device with two of the cc1352 launchpads, I will get an assertion failure in `HandleTransmitDone()` on one of them. For some reason the operation type is `kOperationIdle`. 

**To Reproduce** 

1. Git commit id: e0bc277ccbf7c22ea6dee55144008388c1723ec0
2. Platform CC1352
3. make -f examples/Makefile-cc1352 JOINER=1 COMMISSIONER=1
4. Network topology:
The topology consists of a border router using an NRF52840 dongle as a NCP with a raspberry pi, which will for example execute a ping command as follows to each of the two identical cc1352 launchpad devices within its neighbor table sequentially.
`ping -6 -s 10 -c 20 <ip of first cc1352>`
then
`ping -6 -s 10 -c 20 <ip of second cc1352>`

the second one will consistently fail and go into that assert failure in `src/core/mac/mac.cpp:1488`

**Expected behaviour**
It is expected that both devices respond to the ping request and are not influenced by each other.

**Additional Notes**
If there is only one of these cc1352 launchpads in the same network as the border router, then it works perfectly. As soon as a second one is added, the above steps cause the error.

Also if I don't do anything with the devices i.e. no explicit ping commands or packets, then, they stay connected indefinitely.

Any suggestions on what could be causing this or where to start looking?

srickardti, 2020-10-08:
@eujean , do you have a sniffer log of this happening? I'll setup my own network to investigate, but that may take some time.

My hunch is that the ack frame from node A is being picked up by node B and the sequence number matches the last transmitted frame of node B. Unlikely, but not impossible.

For the platform driver for the TI SDK we make sure to only call `otPlatRadioTxDone` when the radio was requested to transmit. It should be sufficient to check `cc1352_stateTransmit == sState` before [notifying the upper layers](https://github.com/openthread/openthread/blob/master/examples/platforms/cc1352/radio.c#L1954). If the problem is what I think it is.

eujean, 2020-10-09:
@srickardti Find attached a capture of when this occured. In this capture, the border router is 0x8400, the first node being pinged is 0x0800 and the node which fails is 0x5400. Please let me know if you need more information regarding this.

[assert_failure_capture.zip](https://github.com/openthread/openthread/files/5354394/assert_failure_capture.zip)

Now that you mention it, when I started with the cc1352 I used the examples provided by the simplelink sdk. And looking at their platform driver for the radio, they have a `platformRadio_phyState_Transmit != sState` with the comment "We only care about ACKS in TX state". their `radio.c` is somewhat different so not sure if I'm looking in the right place, but I don't recall having seen this problem with their example, so I will do a few quick tests and let you know.

srickardti, 2020-10-09:
I think either the thread masterkey is not the OpenThread default, or we missed some packets at the beginning. I'm unable to fully decode the capture.

But it looks like the node in question is ext addr `6a:93:64:b1:f4:79:92:b7` and advertises becoming a router in packets 39 and 43, based on the sequence numbers. Packet 43 has sequence number 44 and is a broadcast, that would drop us out of transmit state as soon as the packet is transmitted. But the transmit buffer isn't cleared out. It takes a while, but the node with short address `0x8400` wraps its sequence number and counts up to 44. Once that happens, the first node hears the ack with a matching sequence number and blindly tells the upper layers that a transmit was done.

You are correct, the `radio.c` in the SimplelinkSDK is a bit more complicated. It interacts with the RF driver to manage the radio's power states and apply the latest RF characterization settings. But yeah, that's the comment I was referencing. I implemented ack matching in the SimplelinkSDK first and then back ported it to the github project. I think this was just carelessness on my part.

srickardti, 2020-10-09:
These changes might fix your issue: https://github.com/srickardti/openthread/tree/openthread-next

eujean, 2020-10-12:
Yes sorry I forgot that I changed the masterkey during my testing.

Thank you for the fix, I will try it out and let you know if it solves the problem.

eujean, 2020-10-13:
@srickardti It seems to work with that change. 

Thank you!

PR #5640 (2020-10-13): [cc1352,cc2652] add check before claiming transmit done
Check the global state in the radio when an ack frame is found before
reporting a transmit was completed. This change avoids the case where
an acknowledgment frame from another transmission is mistaken for our
acknowledgment frame.

Resolves #5619
closed 2020-10-13 by eujean