openthread/openthread #5628: [codecov] code coverage threshold check is failing

jwhui, 2020-10-09:
https://github.com/openthread/openthread/pull/5627
https://github.com/openthread/openthread/pull/5624
https://github.com/openthread/openthread/pull/5623
https://github.com/openthread/openthread/pull/5614

simonlingoogle, 2020-10-10:
I found that #5613 and #5616 uploaded no code coverage data. Why was that? @jwhui

Irving-cl, 2020-10-10:
I checked the coverage report of some new PRs and found 2 causes for a huge decrease of the code coverage. However I don't understand why would that happen for either of them.
1. A decrease in `ncp` code. In #5623, if we sort the files by 'miss lines':
[image]
We can see a big decrease in `ncp_base_mtd`, `ncp_base_ftd` and `ncp_base`. When I checked `ncp_base_mtd.cpp` for detail, I found functions like `otError NcpBase::HandlePropertyGet<SPINEL_PROP_JAM_DETECT_RSSI_THRESHOLD>(void)` are not hit. And many other properties get/set functions. But they should be executed by case in toranj (`test-001-get-set.py`). I also checked the CI build for this PR and the report seems to be uploaded successfully. #5624 shares the same problem. I cannot find any causes for such missing of these code. As a contrast, in #5622:
[image]
The decreasing of ncp code is not large and hence the PR passed coverage report. It feels like the hit/miss of ncp code are not stable. However I cannot find out the reason.
2. Code wrapped with `LCOV_EXCL_XXX` are tracked (they shouldn't be tracked). In #5629: 
[image]
We can see a great number of `lines missed` in `spinel.c`. And I found those code are wrapped with `LCOV_EXCL_XXX` so they shouldn't be taken into account. #5573 have the same problem. And I don't understand why `LCOV_EXCL_START` and `LCOV_EXCL_STOP` doesn't work. This also seems to happen occasionally.

These are some observations. Any idea about what recent changes could cause these problems?

jwhui, 2020-10-11:
It appears other project have better success generating lcov output first then uploading the lcov output to codecov.

See:

- https://github.com/zephyrproject-rtos/zephyr/pull/5937
- https://github.com/zephyrproject-rtos/zephyr/blob/0d04754652e9655c442b7ca730fdf599ce80a1a0/scripts/ci/run_ci.sh#L37-L77

In the meantime, I submitted #5631 to disable project threshold until we understand how to resolve the issue.

wgtdkp, 2020-10-11:
@Moandor-y @jwhui Not sure if I have mentioned this already, there was a coverage issue when I tried to migrate OT Commissioner from Travis CI to GitHub Action (PR: https://github.com/openthread/ot-commissioner/pull/104). The case is that the `codecov/codecov-action@v1` job generates definitely wrong coverage data for OT Commissioner (It used to be working with Travis CI but has problem with GitHub Action, not sure how this could happen).

The Solution I came out at the end is to use `lcov` to generate the coverage report at the local and upload the report directly:
```yaml
- name: Upload coverage to Codecov
    run: |
        lcov --directory build/ --capture --output-file coverage.info
        lcov --list coverage.info
        bash <(curl -s https://codecov.io/bash) -f coverage.info
```

This works for OT Commissioner and there is unstable coverage issues it OT Commissioner anymore.

Moandor-y, 2020-10-12:
The coverage data on the master branch is also unstable: <https://codecov.io/gh/openthread/openthread/branch/master/commits>. We can see +- thousands of lines in recent commits, and coverage dropped to 78%. Also, it shows "processing reports" for recent commits. I guess there are some temporary issues with codecov.
It seems the OT commissioner isn't affected thanks to @wgtdkp 's solution. I'll send a PR to use the same strategy and see if this makes it stable.

Moandor-y, 2020-10-14:
For the large number of commits stuck in "Processing" [here](https://codecov.io/gh/openthread/openthread/commits), I believe this instability is caused by Codecov itself. Maybe related to the [outage](https://status.codecov.io/incidents/wz84gcst2jff) several days ago. I think we can wait until next week and see if it gets fixed.

Moandor-y, 2020-10-15:
From Codecov:
> looks like this is an issue on our side. It’s going to take some time to investigate and get a fix in. The main issue is that the repository is hitting a limit in number of uploads, so I would suggest, in the meantime if it’s possible, to add a step that uploads a few of the coverage reports to Codecov as a single step.

So confirmed it's an issue on their side. We can also see if it's possible to combine multiple report uploads into one while waiting for their solution.

Moandor-y, 2020-10-20:
The issue isn't fixed yet. I found a way to combine reports from multiple jobs into one report using the artifact uploading/downloading in GitHub Actions, I'll try it in https://github.com/openthread/openthread/pull/5635.
Also, I'll see if we can switch to Coveralls.

Moandor-y, 2020-10-22:
I've changed https://github.com/openthread/openthread/pull/5635 to merge uploads into one (except 1.2 backbone tests which upload data in Docker containers), and the Codecov report seems working again (shows 84%+).

@simonlingoogle will make 1.2 backbone tests send coverage reports to artifacts and let them be merged in the same way. With both changes the number of uploads can be reduced to less than 10 and code coverage shall be stable again.

simonlingoogle, 2020-10-22:
> I've changed #5635 to merge uploads into one (except 1.2 backbone tests which upload data in Docker containers), and the Codecov report seems working again (shows 84%+).
>
> @simonlingoogle will make [...]

It sounds great! And we are going to have a fixed number of code coverage uploads, and we can set `after_n_builds` correctly in codecov.yml.

Moandor-y, 2020-10-28:
Updated https://github.com/openthread/openthread/pull/5635. Now it merges reports before uploading, so there will be only 4 uploads.

Also, the new uploading method can handle `LCOV_EXCL_START` correctly. We can see -2500 total line count in reports.

PR #5635 (2020-10-29): [continuous-integration] use lcov to generate coverage data
This PR makes it use `lcov` to generate the coverage report and upload using the bash script, which may be more stable.

Resolves #5628.
closed 2020-10-29 by jwhui