openthread/openthread #5647: src/posix/Makefile-posix DAEMON=1 fails to build

vke, 2020-10-14:
Hello, I'm trying to build posix tools with DAEMON=1 using the latest commit on a freshly installed RPi4 and I get the following errors:

```/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp: In function ‘void UpdateUnicastLinux(const otIp6AddressInfo&, bool)’:
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp:354:119: error: cast from ‘char*’ to ‘rtattr*’ increases required alignment of target type [-Werror=cast-align]
     rta           = reinterpret_cast<struct rtattr *>((reinterpret_cast<char *>(&req)) + NLMSG_ALIGN(req.nh.nlmsg_len));
                                                                                                                       ^
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp:366:105: error: cast from ‘char*’ to ‘rtattr*’ increases required alignment of target type [-Werror=cast-align]
         rta           = reinterpret_cast<struct rtattr *>((reinterpret_cast<char *>(rta)) + rta->rta_len);
                                                                                                         ^
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp: In function ‘void processNetlinkEvent(otInstance*)’:
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp:1041:75: error: cast from ‘char*’ to ‘nlmsghdr*’ increases required alignment of target type [-Werror=cast-align]
     for (struct nlmsghdr *msg = reinterpret_cast<struct nlmsghdr *>(buffer); NLMSG_OK(msg, static_cast<size_t>(length));
                                                                           ^
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp: In function ‘void processMLDEvent(otInstance*)’:
/home/pi/openthread/src/posix/../../src/posix/platform/netif.cpp:1191:98: error: cast from ‘sockaddr*’ to ‘sockaddr_in6*’ increases required alignment of target type [-Werror=cast-align]
             struct sockaddr_in6 *addr6 = reinterpret_cast<struct sockaddr_in6 *>(ifAddr->ifa_addr);
                                                                                                  ^
  CXX      libopenthread_posix_a-virtual_time.o
/home/pi/openthread/src/posix/../../src/posix/platform/udp.cpp: In function ‘otError transmitPacket(int, uint8_t*, uint16_t, const otMessageInfo&)’:
/home/pi/openthread/src/posix/../../src/posix/platform/udp.cpp:131:86: error: cast from ‘const otIp6Address*’ to ‘const in6_addr*’ increases required alignment of target type [-Werror=cast-align]
     if (!IsMulticast(reinterpret_cast<const struct in6_addr &>(aMessageInfo.mSockAddr)) &&
                                                                                      ^
```

Steps to reproduce:
1) get fresh sources
2) run bootstrap scripts
3) build using `make -f src/posix/Makefile-posix DAEMON=1`

Raspberry Pi OS version (fully updated): `Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`
Git commit: `690ddbf8817b62413a2444e0963c7d96496b0c98`

x893, 2020-10-14:
You use -Wcast-align in configure.ac and code
rta           = reinterpret_cast<struct rtattr *>((reinterpret_cast<char *>(rta)) + rta->rta_len);
notify about cast-align warning on ARM platform.
So need rewrite code or -Wno-error=cast-align

x893, 2020-10-14:
Of course -Wno-error=cast-align remove errors.
But need correct implementation of

rta           = reinterpret_cast<struct rtattr *>((reinterpret_cast<char *>(&req)) + NLMSG_ALIGN(req.nh.nlmsg_len));

because you can have problem with alignement on ARM.
I hope G guru programmers resolve it.

bukepo, 2020-10-15:
@x893 I think the current implementation is a correct version. I found a similar discussion about [lwIP](http://lwip.100.n7.nabble.com/patch-5846-struct-sockaddr-in-has-different-alignment-than-struct-sockaddr-td13388.html). Throughts?

x893, 2020-10-15:
May be. But the compiler warns that the alignment can be violated during such an operation and a hardfault can occur. It would be perfectly correct to copy the bytes into the rtattr structure, but this takes more time. It is possible to change -Wcast-align (configure.ac) to -Wno-error=cast-align for compilation.

Moandor-y, 2020-10-20:
The implementation is correct, but the compiler isn't smart enough to check the alignment. Because the usage of `rtnetlink` is incompatible with the `memcpy` solution, I think the only solution is to ignore these errors. I've sent a PR to ignore `cast-align` at these locations.

closed 2020-10-28 by jwhui