openthread/openthread #5662: Daemon: RadioSpinelNoResponse or radio tx timeout

grja, 2020-10-18:
**Describe the bug** 
After a few minuts ot-daemon is crashing with various error messages on efr32 platform.

**To Reproduce**
1. Git commit id: 5c7850ebe5aa5d73189dd2b9d8e060c72abb31b3
2. Hardware platform: EFR32MG12 (Thunderboard Sense 2)
3. Build steps for daemon: 
```make -f src/posix/Makefile-posix DAEMON=1```
4. Build steps for Thunderboard Sense 2: 
Gecko SDK Version: 3.0.2
```make -f examples/Makefile-efr32mg12 BOARD=BRD4166A  ```

4. Network topology:
- RCP on EFR32MG12 (Thunderboard Sense 2) + one ot-cli-ftd on EFR32MG12 (Thunderboard Sense 2) 
- Daemon and RCP run on Ubuntu 18.04.4 LTS
- Command to run Daemon: 
``` sudo ./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200' -v```

**Console/log output**
Two examples of logs:
First run:
```./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: Running OPENTHREAD/20191113-01298-g5c7850ebe; POSIX; Oct 16 2020 13:59:36
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: Thread version: 2
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: RCP version: OPENTHREAD/20191113-01298-g5c7850ebe; EFR32; Oct 16 2020 13:33:13
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: =========[[THCI] direction=recv | type=JOIN_FIN.req | len=048]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: | 10 01 01 21 0A 4F 50 45 | 4E 54 48 52 45 41 44 22 | ...!.OPENTHREAD"
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: | 05 45 46 52 33 32 23 10 | 32 30 31 39 31 31 31 33 | .EFR32#.20191113
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: | 2D 30 30 39 39 39 2D 67 | 25 06 18 B4 30 00 00 10 | -00999-g%..40...
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: =========[[THCI] direction=send | type=JOIN_FIN.rsp | len=003]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: | 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: [THCI] direction=send | type=JOIN_ENT.ntf
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: [THCI] direction=recv | type=JOIN_ENT.rsp
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: [CRIT]-PLAT----: exit(1): ProcessRadioStateMachine line 999, radio tx timeout, Failure
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[4536]: [CRIT]-PLAT----: Removing daemon socket: /tmp/openthread.sock
```
Second run:
```./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: Running OPENTHREAD/20191113-01298-g5c7850ebe; POSIX; Oct 16 2020 13:59:36
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: Thread version: 2
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: RCP version: OPENTHREAD/20191113-01298-g5c7850ebe; EFR32; Oct 16 2020 13:33:13
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: =========[[THCI] direction=recv | type=JOIN_FIN.req | len=048]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 10 01 01 21 0A 4F 50 45 | 4E 54 48 52 45 41 44 22 | ...!.OPENTHREAD"
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 05 45 46 52 33 32 23 10 | 32 30 31 39 31 31 31 33 | .EFR32#.20191113
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 2D 30 30 39 39 39 2D 67 | 25 06 18 B4 30 00 00 10 | -00999-g%..40...
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: =========[[THCI] direction=send | type=JOIN_FIN.rsp | len=003]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [THCI] direction=send | type=JOIN_ENT.ntf
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [THCI] direction=recv | type=JOIN_ENT.rsp
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: =========[[THCI] direction=recv | type=JOIN_FIN.req | len=048]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 10 01 01 21 0A 4F 50 45 | 4E 54 48 52 45 41 44 22 | ...!.OPENTHREAD"
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 05 45 46 52 33 32 23 10 | 32 30 31 39 31 31 31 33 | .EFR32#.20191113
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 2D 30 30 39 39 39 2D 67 | 25 06 18 B4 30 00 00 10 | -00999-g%..40...
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: =========[[THCI] direction=send | type=JOIN_FIN.rsp | len=003]==========
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: | 10 01 01 .. .. .. .. .. | .. .. .. .. .. .. .. .. | ................
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: ------------------------------------------------------------------------
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [THCI] direction=send | type=JOIN_ENT.ntf
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [THCI] direction=recv | type=JOIN_ENT.rsp
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [CRIT]-PLAT----: WaitResponse() at /home/g/work/openthread-git/openthread/src/posix/../../src/lib/spinel/radio_spinel_impl.hpp:1388: RadioSpinelNoResponse
./output/posix/x86_64-unknown-linux-gnu/bin/ot-daemon[14201]: [CRIT]-PLAT----: Removing daemon socket: /tmp/openthread.sock

```

dismirlian, 2020-10-19:
Hello, make sure you are using hardware flow control (eg. `'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200&uart-flow-control'`).

grja, 2020-10-20:
Unfortunately my development board (Thunderboard Sense 2) doesn't support hardware flow control. Now, I am preparing another development board with supports hardware flow control. Then I will start testing daemon again.

grja, 2020-10-28:
I tested with the dev board that supports hw flow control and it works fine. Issue is not seen anymore.
Issue can be closed.
Thank you for help and quick response.

closed 2020-10-28 by grja