openthread/openthread #5667: [docs] CLI docs out of date

lmnotran, 2020-10-19:
**Is your feature request related to a problem? Please describe.** 
#5587 changes the API definitions for various CLI functions and the CLI docs haven't been regenerated to reflect these changes yet.

Example: https://openthread.io/reference/group/api-cli#otclisetusercommands

**Describe the solution you'd like**
Short term: regenerate the docs. 

Long term: Would it be possible to add a CI job that regenerates the API docs every time [the line](https://github.com/openthread/openthread/blob/master/include/openthread/instance.h#L56) defining `OPENTHREAD_API_VERSION` is updated?

Vyrastas, 2020-10-19:
The reference docs for openthread.io have to be kicked off manually, and we generally do a run every other week.  The process is managed outside of GitHub and we also do a manual review of some files as part of the process... so I don't believe there's a CI job that can fully automate the update.

That said, we're scheduled to run the reference this week so it should be updated by tomorrow.

lmnotran, 2020-10-19:
Gotcha! I didn't know that there was a manual review involved, but I guess it makes sense for quality control, etc. 

Thanks @Vyrastas !

closed 2020-10-19 by lmnotran

Vyrastas, 2020-10-21:
@lmnotran the openthread.io reference has been updated, so should have the changes you were looking for.