openthread/openthread #5676: Coap command not transmit by RCP with ml-EID ip

nicho2, 2020-10-21:
I send a command coap with no response:
coap-client -m get coap://[fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832]/ble_name -N

I have 4 products and a BR with RCP dongle nrf52840 (with build 5c7850ebe..dcebe97f8)

the 4 products are nrf52840_SDK

IPs:
nrf52840_sdk_1:
     [ "fdbb:ddbd:7576:8ac9:0:ff:fe00:c00", "fd11:22:0:0:ea00:a829:4777:904d", "fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832", "fe80:0:0:0:6817:9951:3cab:bd1e" ]

Border router:
	[ "fdbb:ddbd:7576:8ac9:0:ff:fe00:2800", "fd11:22:0:0:f239:9caa:3059:66a7", "fdbb:ddbd:7576:8ac9:b6af:bb5a:2634:99fa", "fe80:0:0:0:d872:ea23:54e4:2713" ]

nrf52840_sdk_2
[ "fdbb:ddbd:7576:8ac9:0:ff:fe00:4400", "fd11:22:0:0:e980:e73b:f474:d970", "fdbb:ddbd:7576:8ac9:8eaf:866:fd26:7834", "fe80:0:0:0:7810:787e:e820:806" ]

nrf52840_sdk_3
[ "fdbb:ddbd:7576:8ac9:0:ff:fe00:b400", "fd11:22:0:0:32fd:bde2:2215:4964", "fdbb:ddbd:7576:8ac9:2b42:a87a:e912:a1d", "fe80:0:0:0:44ac:793a:59f5:f269" ]

nrf52840_sdk_4
[ "fdbb:ddbd:7576:8ac9:0:ff:fe00:e400", "fd11:22:0:0:942d:bef:7820:bb20", "fdbb:ddbd:7576:8ac9:fbaa:62ea:3d32:22a", "fe80:0:0:0:5803:40e:f9e2:34f2" ]

For all products, i request the same Coap ressource with ml-eid and i haven't answer with the product with "fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832" ipv6

sudo ot-ctl coap get fdbb:ddbd:7576:8ac9:0000:00ff:fe00:0c00 ble_name
coap response from fdbb:ddbd:7576:8ac9:0:ff:fe00:c00 with payload: 5468726561645f47505f32314643

sudo ot-ctl coap get fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832 ble_name
no response

coap-client -m get coap://[fdbb:ddbd:7576:8ac9:0000:00ff:fe00:0c00]/ble_name -N
Thread_GP_21FC

coap-client -m get coap://[fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832]/ble_name -N
no response

When i see the frame with wireshark, i observe the request frame is never sending, so i think the RCP don't relayed this frame.

Also, Sometimes, it's work well!

coap-client -m get coap://[fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832]/ble_name -N
Thread_GP_21FC

route data:

{
  "rloc16": "0c00",
  "mlip": "fdbb:ddbd:7576:8ac9:0000:00ff:fe00:0c00",
  "br": false,
  "Ext_Address": "'6a1799513cabbd1e'",
  "Rloc16": "0x0c00",
  "Route": {
    "IdSequence": "112",
    "RouteData": [
      {
        "RouteId": "0x03",
        "LinkQualityOut": "0",
        "LinkQualityIn": "0",
        "RouteCost": "1"
      },
      {
        "RouteId": "0x0a",
        "LinkQualityOut": "0",
        "LinkQualityIn": "1",
        "RouteCost": "2"
      },
      {
        "RouteId": "0x11",
        "LinkQualityOut": "1",
        "LinkQualityIn": "1",
        "RouteCost": "2"
      },
      {
        "RouteId": "0x21",
        "LinkQualityOut": "0",
        "LinkQualityIn": "0",
        "RouteCost": "2"
      },
      {
        "RouteId": "0x2d",
        "LinkQualityOut": "3",
        "LinkQualityIn": "3",
        "RouteCost": "1"
      },
      {
        "RouteId": "0x39",
        "LinkQualityOut": "3",
        "LinkQualityIn": "3",
        "RouteCost": "1"
      }
    ]
  },
  "Leader_Data": {
    "PartitionId": "0x4d788d3d",
    "Weighting": "64",
    "DataVersion": "233",
    "StableDataVersion": "59",
    "LeaderRouterId": "0x21"
  }
}

jwhui, 2020-10-21:
Can you provide the output of `eidcache`? I'm wondering if somehow the Address Query did not succeed.

Can you also provide the sniffer packet capture?

What version of the Nordic SDK are you using?

nicho2, 2020-10-22:
following, your request

 sudo ot-ctl eidcache
fdbb:ddbd:7576:8ac9:fbaa:62ea:3d32:22a e400
fdbb:ddbd:7576:8ac9:2b42:a87a:e912:a1d b400
fdbb:ddbd:7576:8ac9:8eaf:866:fd26:7834 9800
fdbb:ddbd:7576:8ac9:4503:be71:4b49:f4b5 8400
fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832 fffe
Done

[coap frame not sending.zip](https://github.com/openthread/openthread/files/5420317/coap.frame.not.sending.zip)
dataset:
Active Timestamp: 1
Channel: 11
Channel Mask: 07fff800
Ext PAN ID: f515a3a51f0d87d5
Mesh Local Prefix: fdbb:ddbd:7576:8ac9/64
Master Key: 1ca00c3430a05390f60d736bfb776f03
Network Name: OT_AD_Demo
PAN ID: 0xdad4
PSKc: dfd75d9ed470b2c8c83b2e2dd3a17376
Security Policy: 0, onrcb

nordic Dongle : nrf52840
nordic sdk : nrf5_thread_4.1.0

the problem is with fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832 (in the capture we don't see coap request ble_name for this ip)
the RLOC here is ec00

jwhui, 2020-10-23:
@nicho2 , my guess is that you had tried to contact that IPv6 address when the device was not attached and the Address Query did not get a response. As a result, the Address Query component is backing off. Currently, the Thread Specification specifies a maximum backoff time of 8 hours.

https://github.com/openthread/openthread/blob/aaf553ee7d5ac42e35155551414fa9dde7731bf3/src/core/config/tmf.h#L97-L107

In practice, we have found that this is far too long. In Nest products, we use a much smaller value of 120 seconds (2 minutes).

I would suggest changing the above config to 120 to see if that helps resolve your issue.

nicho2, 2020-10-23:
Hello,

If you observed the capture, at the line 40, i send a  networkdiagnostic get  fdbb:ddbd:7576:8ac9::ff:fe00:ec00  0 1 5 6 16 
and the product is answered good (line 42)

and normaly at the 55, i send a get coap for ressource /ble_name on this product and i use its ipv6  fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832  but as you see the frame is not transmit by the dongle ( Border Router).
It's not a problem with the attachment of the product. it's a problem with the RCP.

jwhui, 2020-10-23:
@nicho2 , when sending to an RLOC, no Address Query is required. However, when sending to an EID (i.e. not RLOC), the device will need to perform an Address Query to discover the RLOC. I do not see an Address Query message emitted for `fdbb:ddbd:7576:8ac9:e6ac:f767:1c27:c832`. Can you try changing the config for `OPENTHREAD_CONFIG_TMF_ADDRESS_QUERY_MAX_RETRY_DELAY` as suggested above?

nicho2, 2020-10-23:
Hello Jonathan,

I have changed this line 
#define OPENTHREAD_CONFIG_TMF_ADDRESS_QUERY_MAX_RETRY_DELAY 120 //28800

for the device and for the RCP, and i have always the problem:

sudo ot-ctl eidcache
fdbb:ddbd:7576:8ac9:fbaa:62ea:3d32:22a e400
fdbb:ddbd:7576:8ac9:2b42:a87a:e912:a1d b400
fdbb:ddbd:7576:8ac9:8eaf:866:fd26:7834 9800
fdbb:ddbd:7576:8ac9:4503:be71:4b49:f4b5 8400
fdbb:ddbd:7576:8ac9:25dc:f4ce:1969:a081 fffe

Is it possible there is a relation with the topology  because between the RCP and the device there is 2 hops (the other products have a direct link with RCP)?

jwhui, 2020-10-23:
@nicho2 , the fact that (1) the target is multiple hops away and (2) the target is an EID (not an RLOC) requires the BR to perform an Address Query to determine how to reach the device.

It is not clear to me why the Address Query is not happening or not succeeding. The pcap file is a bit short. Would you be able to collect a packet capture that includes the network formation itself?

Also, can you enable DEBG level logging on the BR and provide the logs?

nicho2, 2020-10-26:
here, you will found a first log file

[log.txt](https://github.com/openthread/openthread/files/5439677/log.txt)

Tomorrow, i'm doing all you are requested

nicho2, 2020-10-27:
Hello @jwhui, following the trace 
[log_RCP_join_and_coap.txt](https://github.com/openthread/openthread/files/5443557/log_RCP_join_and_coap.txt)

[join_and_coap_complet.zip](https://github.com/openthread/openthread/files/5443562/join_and_coap_complet.zip)

the product has now these ipv6 :

55 | dc00 | [ "fdbb:ddbd:7576:8ac9:0:ff:fe00:dc00", "fdbb:ddbd:7576:8ac9:7137:2781:adc0:aeb", "fe80:0:0:0:6cff:646e:c3b7:fb66" ]

jwhui, 2020-10-28:
@nicho2 , thank you for raising this issue and working with us to pinpoint the issue.

The sniffer packet capture clearly shows the Address Query message being sent with an IPv6 Hop Limit value set to 1. As a result, the Address Query message is never forwarded to devices that are multiple hops away.

The root cause is when `PLATFORM_UDP` is enabled, the Address Query message is sent using the Linux kernel's default IPv6 hop limit value. For multicast messages, the default is set to 1.
https://github.com/openthread/openthread/blob/e95d7b676140740a0a3f200be93602a9eb459d92/src/posix/platform/udp.cpp#L119

One possible way to fix this is by using `OPENTHREAD_CONFIG_IP6_HOP_LIMIT_DEFAULT` rather than the Linux kernel default. This would make the behavior consistent with OpenThread devices that do not have `PLATFORM_UDP` enabled. See #5736 for the proposed fix. Let me know if this addresses your issue.

nicho2, 2020-10-29:
hello @jwhui, 
i have put this modification:
int hopLimit = (aMessageInfo.mHopLimit ? aMessageInfo.mHopLimit : OPENTHREAD_CONFIG_IP6_HOP_LIMIT_DEFAULT);

and now, all works fine.

Thanks.

jwhui, 2020-10-29:
> hello @jwhui,
> i have put this modification:
> int hopLimit = (aMessageInfo.mHopLimit ? aMessageInfo.mHopLimit : OPENTHREAD_CONFIG_IP6_HOP_LIMIT_DEFAULT);
> 
> and now, all works fine.

@nicho2 , thanks for confirming!

PR #5736 (2020-10-29): [posix] default hop limit to OPENTHREAD_CONFIG_IP6_HOP_LIMIT_DEFAULT
Resolves #5676
closed 2020-10-29 by jwhui