openthread/openthread #5678: [Daemon]  Error decoding hdcl frame: nobuffs

tlifschitz, 2020-10-21:
**Describe the bug**
--------------
While running ot-daemon, hdlc frames are discarded in `HandleHdlcFrame` with error `OT_ERROR_NO_BUFS` when hdlc decoder tries to write decoded bytes from the RCP. This happens when multiple SEDs (10) attempt to attach at the same time or during intense data upload from SEDs to the FTD, making the network very unstable, and eventually making ot-daemon crash.

**To Reproduce**
-------------
1. Git commit id **9e3b450218582e721b102683de9786e7be5ef0b2**
2. IEEE 802.15.4 hardware platform **EFR32MG13/EFR32MG12**
3. Build steps
Linux recipe for ot-daemon
```
set(OT_BORDER_AGENT ON CACHE BOOL "enable border agent" FORCE)
set(OT_BORDER_ROUTER ON CACHE BOOL "enable border router feature" FORCE)
set(OT_BUILTIN_MBEDTLS_MANAGEMENT OFF CACHE BOOL "diable mbedTLS management" FORCE)
set(OT_COMMISSIONER ON CACHE BOOL "enable commissioner" FORCE)
set(OT_DAEMON ON CACHE BOOL "enable daemon mode" FORCE)
set(OT_JOINER ON CACHE BOOL "enable joiner" FORCE)
set(OT_LEGACY ON CACHE STRING "enable legacy network support" FORCE)
set(OT_MAC_FILTER ON CACHE STRING "enable MAC filter" FORCE)
set(OT_MBEDTLS_CONFIG_FILE "\"${PROJECT_SOURCE_DIR}/mbedtls-config.h\"" CACHE STRING "use mbedtls-config.h of this project" FORCE)
set(OT_PLATFORM "posix" CACHE STRING "use posix platform" FORCE)
set(OT_PLATFORM_NETIF ON CACHE BOOL "enable platform netif" FORCE)
set(OT_PLATFORM_UDP ON CACHE BOOL "enable platform UDP" FORCE)
set(OT_UDP_FORWARD OFF CACHE BOOL "diable udp forward" FORCE)
set(OT_SERVICE ON CACHE BOOL "enable services" FORCE)
set(OT_FULL_LOGS ON CACHE BOOL "enable full logs" FORCE)
set(OT_LOG_LEVEL_DYNAMIC ON CACHE BOOL "enable dynamic log level" FORCE)
set(OT_LOG_LEVEL "DEBG" CACHE STRING "debug log level" FORCE)

list(APPEND OT_PLATFORM_DEFINES "-DOPENTHREAD_CONFIG_POSIX_SETTINGS_PATH=\"/data/thread\"")

add_subdirectory(git)

target_compile_definitions(ot-config INTERFACE
"-DOPENTHREAD_CONFIG_RADIO_915MHZ_OQPSK_SUPPORT=1"
"-DOPENTHREAD_CONFIG_RADIO_2P4GHZ_OQPSK_SUPPORT=0"
"-DOPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_MAX_ALOCS=32"
"-DOPENTHREAD_CONFIG_MPL_DYNAMIC_INTERVAL_ENABLE=1"
"-DOPENTHREAD_CONFIG_MLE_MAX_CHILDREN=64         "
"-DOPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS=1024    "
"-DOPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES=256   "
"-DOPENTHREAD_CONFIG_COAP_SERVER_MAX_CACHED_RESPONSES=128"
"-DOPENTHREAD_CONFIG_LOG_LEVEL=OT_LOG_LEVEL_DEBG "
"-DOPENTHREAD_CONFIG_LOG_API=1                   "
"-DOPENTHREAD_CONFIG_LOG_ARP=1                   "
"-DOPENTHREAD_CONFIG_LOG_BBR=1                   "
"-DOPENTHREAD_CONFIG_LOG_CLI=1                   "
"-DOPENTHREAD_CONFIG_LOG_COAP=1                  "
"-DOPENTHREAD_CONFIG_LOG_DUA=1                   "
"-DOPENTHREAD_CONFIG_LOG_ICMP=1                  "
"-DOPENTHREAD_CONFIG_LOG_IP6=1                   "
"-DOPENTHREAD_CONFIG_LOG_MAC=1                   "
"-DOPENTHREAD_CONFIG_LOG_MEM=1                   "
"-DOPENTHREAD_CONFIG_LOG_MESHCOP=1               "
"-DOPENTHREAD_CONFIG_LOG_MLE=1                   "
"-DOPENTHREAD_CONFIG_LOG_MLR=1                   "
"-DOPENTHREAD_CONFIG_LOG_NETDATA=1               "
"-DOPENTHREAD_CONFIG_LOG_NETDIAG=1               "
"-DOPENTHREAD_CONFIG_LOG_PKT_DUMP=1              "
"-DOPENTHREAD_CONFIG_LOG_PLATFORM=1              "
"-DOPENTHREAD_CONFIG_LOG_PREPEND_LEVEL=1         "
"-DOPENTHREAD_CONFIG_LOG_PREPEND_REGION=1        ")
```

4. Network topology **1 FTD -> 10 SEDs**

Set up network leader and attempt to attach multiple devices in a couple of seconds windows, or produce a considerable amount of traffic from SEDs to FTD (we are sending ~30Kb per device using CoAP )

**Expected behavior**
------
[... 153 lines cut ...]
Oct 19 20:59:23 [WARN]-PLAT----: Error decoding hdlc frame: NoBufs

------------------------------ 180 repeated lines ----------------------------------

Oct 19 20:59:23 [WARN]-PLAT----: Error decoding hdlc frame: NoBufs
Oct 19 20:59:23 [WARN]-PLAT----: Error decoding hdlc frame: NoBufs
Oct 19 20:59:23 ot-daemon[21199]: [CRIT]-PLAT----: WaitResponse() at ../git/src/lib/spinel/radio_spinel_impl.hpp:1371: RadioSpinelNoResponse
Oct 19 20:59:23 ot-daemon[21199]: [CRIT]-PLAT----: Removing daemon socket: /tmp/openthread.sock
Oct 19 20:59:23 systemd[1]: ot-daemon.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Oct 19 20:59:23 systemd[1]: ot-daemon.service: Unit entered failed state.
Oct 19 20:59:23 systemd[1]: ot-daemon.service: Failed with result 'exit-code'.
Oct 19 20:59:28 systemd[1]: ot-daemon.service: Service hold-off time over, scheduling restart.
Oct 19 20:59:29 systemd[1]: Stopped ot-daemon.

Oct 19 20:59:29 systemd[1]: Started ot-daemon.
Oct 19 20:59:29 ot-daemon[21565]: Running OPENTHREAD/20191113-01279-g9e3b45021; POSIX; Oct 19 2020 20:15:50
Oct 19 20:59:29 Running OPENTHREAD/20191113-01279-g9e3b45021; POSIX; Oct 19 2020 20:15:50
Oct 19 20:59:29 ot-daemon[21565]: Thread version: 2
Oct 19 20:59:29 ot-daemon[21565]: [DEBG]-PLAT----: Wait response: tid=1 key=54
Oct 19 20:59:29 ot-daemon[21565]: [DEBG]-MAC-----: RadioPanId: 0x1111
```

**Additional context** 
---------------------
We are using a custom platform running embedded linux and EFR32MG13 as RCP and can’t find what could be failing. Has daemon+RCP architecture been thoroughly tested for a production environment or is it still not that mature? For the moment we will continue to use wpantund but with an older version of NCP build since it is not compatible with master branch in openthread repo anymore. However, any insight to solve this would be greatly appreciated, so that we can migrate to using ot-daemon and keep up to date with the upstream changes.

jwhui, 2020-10-21:
The HDLC decoder reports `OT_ERROR_NO_BUFS` when there is no more room in the receive frame buffer to accept new bytes.

- https://github.com/openthread/openthread/blob/dcebe97f82fc30474237ea92cff8345d20c75119/src/lib/hdlc/hdlc.cpp#L270
- https://github.com/openthread/openthread/blob/dcebe97f82fc30474237ea92cff8345d20c75119/src/lib/hdlc/hdlc.cpp#L290

The daemon will exit if it is expecting a response from the RCP and does not receive one:

- https://github.com/openthread/openthread/blob/dcebe97f82fc30474237ea92cff8345d20c75119/src/lib/spinel/radio_spinel_impl.hpp#L1385

My guess is that the UART communication is not reliably delivering bytes from RCP to the daemon. Is there any way for you to validate whether UART communication is reliable?

/cc/ @romacdon

dismirlian, 2020-10-21:
Hi @jwhui, thanks for your response. I'm working with @tlifschitz on this.

> My guess is that the UART communication is not reliably delivering bytes from RCP to the daemon. Is there any way for you to validate whether UART communication is reliable?

Yes, we thought the same. We have conducted the following tests to validate the UART communication:
- RCP Rx: We transmitted a ~10MB file from the host to the RCP. We inserted delays in the RCP to put pressure on the hardware flow control. We validated that the host and RCP behave correctly in this direction.
- RCP Tx: We transmitted a similar file from the RCP to the host. The host made use of hardware flow control to stop the RCP in some circumstances. We validated that the host and RCP behave correctly in this direction also.

Also, we have been testing wpantund with the same children, and with very heavy traffic and the same RCP firmware. We observed no faults with wpantund.

We will try to run a longer UART test.

Any other ideas?

jwhui, 2020-10-21:
@dismirlian , thanks for the additional context.

To help isolate the issue, can you print out and inspect the read bytes? I'm thinking observed the read bytes here:
https://github.com/openthread/openthread/blob/ff76aac3db99c815045a549666137ee6a0723b2f/src/posix/platform/hdlc_interface.cpp#L196

I'd like to see if the issue is occurring before or after the bytes are received via the call to `read()`.

dismirlian, 2020-10-21:
@jwhui, thanks for the incredibly fast answer. We will do as you suggest and also stress the serial port with some tool, eg. https://github.com/cbrake/linux-serial-test.

latheepanm, 2020-10-22:
Just curious, did you try increasing the buffer size? We've had similar issues during load/stress tests and that usually helps. It appears as jwhui mentioned, that the crash is because the response for tid=3, key=4868 gets dropped with the hundred other spinel/hdlc messages. It is interesting that this is more prominent in the RCP case, maybe because significantly more data gets passed between RCP and HOST compared to NCP and HOST?

dismirlian, 2020-10-22:
Hi @latheepanm thanks for the pointer. No, we didn't try, because on inspecting the code it seems that hardware flow control should take care of limiting the traffic if the host can't keep up. I can't find any reason on the host

latheepanm, 2020-10-22:
@dismirlian we had to increased the OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE within ncp_config.h, but we had the OT_ERROR_NO_BUFS in a slightly different place. If I was to guess, I would suggest increasing kMaxFrameSize within spinel_interface.hpp. I'm not too familiar with the specifics of UART comms but wondering if after the data is read from the UART buffer, openthread has no place to put it (since it's buffers are full) and so it drops it.

dismirlian, 2020-10-22:
@jwhui I attach two short log that dumps the read buffer, [here](https://github.com/openthread/openthread/files/5425024/log-ot-daemon-NoBufs-2.txt).

This is the code that generates the dump:
```   C++
    if (rval > 0)
    {
        otDumpInfo(OT_LOG_REGION_PLATFORM, "HdlcInterface::Read", buffer, rval);
        Decode(buffer, static_cast<uint16_t>(rval));
    }
```

Thanks!

dismirlian, 2020-10-22:
@latheepanm, thanks for the info. I'll study your suggestions and report back.

jwhui, 2020-10-22:
> @jwhui I attach a short log that dumps the read buffer, [here](https://github.com/openthread/openthread/files/5425024/log-ot-daemon-NoBufs-2.txt).

@dismirlian , thanks for providing the log. It appears that when things are working well, the bytes read from UART include one HDLC frame at a time:
```
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: =====================[HdlcInterface::Read len=072]======================
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: | 7E 83 06 34 2E CE C8 31 | 44 80 F4 4D A1 EE 7E 7E | ~..4.NH1D.tM!n~~
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: | 80 06 60 FE 80 00 00 00 | 00 00 00 2C CE C8 31 44 | ..`~.......,NH1D
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: | 80 F4 4D 2B E3 7E 7E 80 | 06 63 19 00 FE 80 00 00 | .tM+c~~..c..~...
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: | 00 00 00 00 2C CE C8 31 | 44 80 F4 4D 40 FF FF FF | ....,NH1D.tM@...
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: | FF FF FF FF FF 5A 49 7E | .. .. .. .. .. .. .. .. | .....ZI~........
Oct 22 18:38:22 /usr/sbin/ot-daemon[14652]: ------------------------------------------------------------------------
```

However, where we get into trouble is when there are many UART bytes buffered when we read:
```
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: =====================[HdlcInterface::Read len=2048]======================
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: | 01 D4 B2 7D 31 02 02 00 | 00 00 00 01 00 00 37 0C | .T2}1.........7.
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: | 7E 7E 80 06 71 16 00 6B | 98 64 7D 31 7D 31 00 E8 | ~~..q..k.d}1}1.h
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: | 0E E8 0D 2A 4A 01 00 02 | 04 08 55 EA E0 49 6F E7 | .h.*J.....Uj`Iog
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: | 80 10 00 0A 00 01 D6 FB | AD 02 02 00 00 00 00 01 | ......V{-.......
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: | 00 00 45 74 7E 7E 80 06 | 71 16 00 6B 98 02 7D 31 | ..Et~~..q..k..}1
...
```

And where things fall apart is when we are waiting for a response from the RCP:
```
Oct 22 18:38:57 /usr/sbin/ot-daemon[14652]: [DEBG]-PLAT----: Wait response: tid=12 key=4868
Oct 22 18:38:58 /usr/sbin/ot-daemon[14652]: [WARN]-PLAT----: Error decoding hdlc frame: NoBufs
```

What's happening here is that we have a bunch of messages buffered from the RCP, but we are looking for a particular response. Unfortunately, we cannot receive the response because there are no buffers to receive it.

/cc/ @abtink , thoughts?

dismirlian, 2020-10-22:
Ok, so the HDLC frames are not drained while waiting a specific response? Ah, maybe the frames that come in are "unsolicited", and fill up some buffer?

abtink, 2020-10-22:
@dismirlian thanks for posting the logs and helping debug this.

As @jwhui mentioned, it seems to be an overflow of the spinel HDLC buffer that is causing this.

We see towards the end  of logs (where things go bad) that we are reading large chunks from UART `[HdlcInterface::Read len=2048]`. The read content seems to be fine containing multiple HDLC frames (`7E` is the flag indicating start/end of each frame) - it seems to be mostly the received 15.4 frames.

Here is a quick overview of the buffering model for RCP host:
- We have a buffer to save the decoded HDLC frames
- As `Hdlc` decodes the received bytes from UART, they are written into this buffer (we get callback at end of each fully decoded (spinel) frame).
- In most situations, in the callback, we can directly process the spinel frame and then free up the HDLC buffer for next spinel frame.
- But if we are in middle of a spinel command-request/response with RCP, we would need to wait to see the response.
- In this case, we would keep the decoded spinel frame in the buffer to be processed later (after the response is received).
   - This is needed since OT stack is in general not reentrant and all the calls/callbacks need to happen from same OS context (task/thread).

The situation here seems to be that we are waiting for response for spinel `key=4868` (which is `SPINEL_PROP_MAC_SRC_MATCH_SHORT_ADDRESSES`, i.e. we are updating src-match table on RCP) and while host is waiting for response, we get a lot of received 15.4 frames from RCP (which we would then need to keep in the HDLC buffer which overflows).

------------------

The immediate solution (and I would say proper way for now) would be to increase the HDLC buffer size.
I looked it up and unfortunately noticed that it seems to be hardcoded as `2048` in `spinel_interface.hpp`.

I will submit a PR to add a config option to allow this be changed.

We can consider later if other enhancements/solutions can help with this.

dismirlian, 2020-10-22:
@latheepanm, if "unsolicited" frames can't be processed from the Hdlc RX handler while OT is waiting "synchronously" for a response, it becomes clear that your suggestion of increasing `kMaxFrameSize` is correct. I increased it from 2k to 10k, and the problem goes away. Thank you!

I think that there should be a way of configuring this constant, instead of having it hardcoded. Also, even if I increase it, how can I *guarantee* that there will be no issues?

dismirlian, 2020-10-22:
Hi @abtink, thanks for looking into this. The explanation is very clear, and in line with our observations.

I'll be expecting the PR. Thank you!

abtink, 2020-10-22:
Just submitted https://github.com/openthread/openthread/pull/5687 (to add config option for spinel rx frame buffer size)

On how/if we can guarantee avoiding buffer overflow, I am not sure if we can get this 100% but we can consider some guidelines in terms of what rx buffer size to use to be safer:

- On RCP side, we have existing mechanism/code that when preparing a spinel response  would check if tx buffer is full, and if so we would wait for it be drained and then prepare and send the response.
- So we should not see a drop (i.e. not emitting) a spinel response from RCP due to tx buffer being full.
- However, the response can end at the end of tx buffer after a bunch of other spinel frames.
- So It would be good to make sure that the "spinel rx buffer" size on the host side is larger than the RCP tx spinel buffer.

Another thing is the data may be buffered on the host side as well. We may have the data waiting in a UART buffer on the host side (to be read by OT `HdlcInterface::Read()`). Ideally, I think we may want to use a "spinel rx buffer" that is larger than sum of the two: UART buffer on host and tx buffer on size (though I am not sure if host UART buffer can be known?).

latheepanm, 2020-10-22:
@dismilian glad that you were able to get it to work! We are currently running the stress/load tests to determine what an optimal value would be for our applications but it cannot be guaranteed. 

@abtink that was a great explanation on the root cause of the issue! Would it make sense to use something akin to a priorityqueue (where the priority is the prop key or range of prop keys) to avoid the waitresponse() from timing out? Just thinking out loud

abtink, 2020-10-22:
>  @abtink that was a great explanation on the root cause of the issue! Would it make sense to use something akin to a priorityqueue (where the priority is the prop key or range of prop keys) to avoid the waitresponse() from timing out? Just thinking out loud

Thanks @latheepanm. 

We actually already have such a feature/behavior in the code. 
- `Spinel::Buffer` provides two priority levels (high/low) for frames.
   - The buffer is actually implemented as circular/ring buffer where we write frames in forward or backward directions based on frame priority (low or high). 
https://github.com/openthread/openthread/blob/a04b6ea75362e2cb209f86d8e73e8edba7440623/src/lib/spinel/spinel_buffer.hpp#L494-L511
- When writing frames, spinel responses always use high priority and all events use low priority
https://github.com/openthread/openthread/blob/a04b6ea75362e2cb209f86d8e73e8edba7440623/src/lib/spinel/spinel_encoder.cpp#L49-L69
- So spinel responses should be sent out before other queued event frames (which include 15,4 rx frames) but potentially not before any frame being processed/sent at the time and/or the ones already buffered on the host side (e.g. UART buffer).

I think/guess in the situation from this issue/setup, the overflow may be mainly due to data being buffered on host (before being processed).

latheepanm, 2020-10-23:
> Thanks @latheepanm.
>
> We actually already have such a feature/behavior in the code.
>
>     * `Spinel::Buffer` provides two priority levels (high/low) for frames.
>
>       * The buffer is actually [...]

@abtink thank you for that! I was also wondering if it makes sense on the host side to have SpinelInterface::RxFrameBuffer (MultiFrameBuffer) be implemented as a priorityqueue as well? Would having higher priority frames overwrite lower priority frames be helpful in this situation? Again, just thinking out loud and thank you for the detailed responses!

abtink, 2020-10-23:
@latheepanm, aways happy to discuss and all ideas are always welcome.  😃 

On host side when we are waiting for response and run out of buffer space we can consider freeing/overwriting some existing ones (practically drop them before processing them). I would guess these would likely be received 15.4 frames (potentially acked by RCP already). I agree it may help avoid the response timeout and potential asserts (on missing timeout), but I am not sure if it would be a good idea.

My concern is that it may mask issue in the situations where the `RxFrameBuffer` size is indeed too small and make it look like things are working while dropping  frames (often). I would say in such cases it may be better to fail/assert so indicate a potential issue and bad configuration. 

I think we should be able to allocate large enough buffer for practical traffic situations on host side, specially since running on posix environments, I think/expect we would not be as constrained in terms of memory.

jwhui, 2020-10-23:
@tlifschitz @dismirlian @latheepanm , thanks for raising this issue and all of your help to resolve this!

latheepanm, 2020-10-23:
> @latheepanm, always happy to discuss and all ideas are always welcome. 😃
> 
> On host side when we are waiting for response and run out of buffer space we can consider freeing/overwriting some existing ones (practically drop them before processing them). I would guess these would likely be received 15.4 frames (potentially acked by RCP already). I agree it may help avoid the response timeout and potential asserts (on missing timeout), but I am not sure if it would be a good idea.
> 
> My concern is that it may mask issue in the situations where the `RxFrameBuffer` size is indeed too small and make it look like things are working while dropping frames (often). I would say in such cases it may be better to fail/assert so indicate a potential issue and bad configuration.
> 
> I think we should be able to allocate large enough buffer for practical traffic situations on host side, specially since running on posix environments, I think/expect we would not be as constrained in terms of memory.

@abtink I agree that it would mask the issue and posix systems would not be as constrained as embedded systems. The last two things I would add is,

1) There may be instances when a ton of packets get dropped but the application does not assert/fail (we've witnessed this during our stress tests; not at this layer, but at different layers)
2) From an end user perspective, the error could be handled a bit more gracefully, but I guess that could be done if/as needed.

dismirlian, 2020-10-23:
@abtink, thanks for the detailed explanation on buffering. I agree that it won't make a difference if the default buffer size is 2k, 8k or even 64k. I guess there is a limit to its size because of the `uint16_t`s.

If I'm not mistaken, at least in Linux the UART buffer size is equal to one page (4096 bytes on ARM). I would bump the default frame buffer size to even more than 8k.

Two other comments:
1. From a user's perspective, it is difficult to know *why* the packets get dropped, and which buffer size to tune. I think this is true for most configuration options; a new user won't know what to tune for their specific use case. Would a "tuning guide" make sense? I'm thinking of a guide that outlines the different options that one would need to tune in common usage scenarios.
2. In this specific case (NoBuffs), a communication problem is hard to distinguish from a buffer misconfiguration. For example, the HDLC decoder drops "inter-frame" bytes without any warning.

Thanks for your support!

abtink, 2020-10-23:
Thanks @dismirlian 

> From a user's perspective, it is difficult to know why the packets get dropped, and which buffer size to tune. I think this is true for most configuration options; a new user won't know what to tune [...]

I fully agree. I think this would be very helpful.

With OT now having/adding many features and many config parameters (a lot of features to enable/disable and knobs to control), it's getting more complicated for users (project/product integrators) to learn about all OT configs and where/how to use/set them.

We have been considering for a while to provide a bunch of OT config set examples targeted towards different type of practical use-cases (device types) like:  a posix FTD border router device config example, an NCP border router one , an NCP FTD, an NCP MTD (battery-powered), SoC MTD battery-powered, etc. 

The idea/plan is to build these example configs from the products/projects where we integrate OT internally. I think this can be a useful starting point as an "OT config guide" and I hope we can submit a first PR on this soon. Inputs from other developers/users of OT would be very helpful on this (hopefully we can build on this and make it easier for users/developers)

dismirlian, 2020-10-24:
@abtink, great to hear about the "OT config guide", I think it would be a huge help to developers. Those example use-cases you mention seem to be comprehensive enough to get started.

Thanks!

jwhui, 2020-11-04:
Resolved by #5687

closed 2020-11-04 by jwhui