openthread/openthread #5679: efr32: radio: asserts during data requests

dismirlian, 2020-10-21:
**Describe the bug** A clear and concise description of what the bug is.
With very heavy traffic and very heavy MCU load, the following asserts trigger (lines 911, 919, 926):

https://github.com/openthread/openthread/blob/831406ce9055d622a3258a82a1a6ee556530fc49/examples/platforms/efr32mg12/radio.c#L902-L929

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: 831406ce9055d622a3258a82a1a6ee556530fc49
2. IEEE 802.15.4 hardware platform: efr32mg12
3. Build steps: `make -f examples/Makefile-efr32mg12 BOARD=BRD4161A`
4. Network topology: 1 NCP + 10 SEDs

For this to happen, the MCU has to be *very heavily loaded*, so that the RAIL interrupt fires late. These asserts are due to the `RAIL_IEEE802154_GetAddress` and `RAIL_IEEE802154_SetFramePending` returning `RAIL_STATUS_INVALID_STATE` because it is too late to modify the outgoing Immediate ACK.

I could submit a PR addressing this, but it's not clear to me if we should inform the upper layers that the frame was acked with FP or not, even if these calls failed.

**Expected behavior** A clear and concise description of what you expected to happen.
No asserts.

**Console/log output** If applicable, add console/log output to help explain your problem.
None.

**Additional context** Add any other context about the problem here.
None.

romacdon, 2020-10-22:
@dismirlian - thanks finding and reporting this.  You are exactly correct.  The asserts here are far too extreme.  One of our RAIL experts looked over the code and this is what they had to say, "[These RAIL_ functions] can return an error if it's called too late to modify the outgoing ACK due to interrupt latency. While unlikely we probably shouldn't assert because of it we should just make sure we tell the upper layer that we did not set frame pending on this ACK so it knows the end device won't be staying on to listen."  So if you don't mind submitting a PR that would be great.  Thanks again.

dismirlian, 2020-10-22:
Hi @romacdon, thanks for your answer. I'll submit a PR shortly.

PR #5697 (2020-10-23): [efr32] avoid asserts during data requests
This PR fixes #5679.

If the RAIL interrupt fires late (for example, when the MCU is heavily loaded), then `RAIL_IEEE802154_GetAddress` and `RAIL_IEEE802154_SetFramePending` return `RAIL_STATUS_INVALID_STATE` because it is too late to modify the outgoing Immediate ACK. The proposed solution is to avoid asserting and print a warning to make the user aware of such problems.
closed 2020-10-23 by jwhui