openthread/openthread #5684: [efr32] wrong MAC capabilities

dismirlian, 2020-10-22:
When merging the efr32 platform unification (#5666) the following constants were changed from 0 to 1:
`OPENTHREAD_CONFIG_MAC_SOFTWARE_CSMA_BACKOFF_ENABLE` and `OPENTHREAD_CONFIG_MAC_SOFTWARE_ENERGY_SCAN_ENABLE`, [here](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/src/openthread-core-efr32-config.h).

I think this is not correct. Could someone confirm?

Thanks!

lmnotran, 2020-10-23:
@dismirlian I think you're right. I believe both CSMA and energy scan are handled by RAIL and that leaving the `OPENTHREAD_CONFIG_MAC_SOFTWARE_CSMA_BACKOFF_ENABLE` option would reintroduce #2379.

Thanks for catching this!

dismirlian, 2020-10-25:
Great, thanks.

PR #5700 (2020-10-26): [efr32] disable OT software CSMA backoff and energy scan
Closes #5684

CC: @romacdon
closed 2020-10-26 by jwhui