openthread/openthread #5688: Reattach backoff usage

theHurc, 2020-10-22:
I'm trying to use the `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF` feature. I'm just using the default minimum interval (251 ms) and the maximum interval (1200000 ms). It seems the reattach attempts stop after about 11 attempts only getting to a 257024 ms backoff before it doesn't try any longer. Is there a setting I'm missing or is the behavior intended?

abtink, 2020-10-23:
@theHurc from what I recall/expect from the code, it should not stop attach attempts (unless device gets attached). 
 
I know internally we do use/enable this feature on some MTD devices (battery-powered) and if I recall correctly, there were test-cases verifying its behavior (in the battery stress test and some in-field tests). If I am not mistaken it is being used with same OT default config values.

We can try to run in simulation (with speed-up) to verify the behavior. If you have logs or other info, please share. They may be helpful.

theHurc, 2020-10-24:
I would expect the attach attempts to continue indefinitely. I am using this in a FreeRTOS context so that may be an issue. I'll try and run my application layer in the super-loop context and see if I can reproduce. Thanks

abtink, 2020-10-24:
@theHurc, I ran a build with `MLE_ATTACH_BACKOFF` enabled and default configs in simulation with time speed-up. The logs seem to show correct behavior. 
- It keeps trying to attach, increasing wait time between attempts, till it reaches the max limit of 1200 second 
- We also see a random jitter added when we hit the max.

```
abtink@ ~/sw/openthread/tests/toranj (master) $ sudo rm *.log; sudo TORANJ_VERBOSE=1 TORANJ_POSIX_RCP_MODEL=0 python test-attach-backoff.py 
------------------------------------------------------------------------------------------------------------------------
Starting 'test-attach-backoff'
$ Node1.__init__() cmd: /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:../../examples/apps/ncp/ot-ncp-ftd 1 400" -o Config:TUN:InterfaceName wpan1 -o Config:NCP:DriverName spinel -o Daemon:SyslogMask "all -debug"
$ Node1.wpanctl('leave') -> 'Leaving current WPAN. . .'
$ Node1.wpanctl('form "attch-backoff"'):
     Forming WPAN "attch-backoff" as node type "router"
     Successfully formed!
$ Node1.wpanctl('set Thread:DeviceMode -v 1') -> ''
'test-attach-backoff' passed.

abtink@ ~/sw/openthread/tests/toranj (master) $ cat wpantund-logs1.log  | grep "Attach attempt"
wpantund[1200553]: NCP => [0000469692611][NOTE]-MLE-----: Attach attempt 1 unsuccessful, will try again in 0.287 seconds
wpantund[1200553]: NCP => [0000469697810][NOTE]-MLE-----: Attach attempt 2 unsuccessful, will try again in 0.541 seconds
wpantund[1200553]: NCP => [0000469702857][NOTE]-MLE-----: Attach attempt 3 unsuccessful, will try again in 1.045 seconds
wpantund[1200553]: NCP => [0000469708210][NOTE]-MLE-----: Attach attempt 4 unsuccessful, will try again in 2.052 seconds
wpantund[1200553]: NCP => [0000469714189][NOTE]-MLE-----: Attach attempt 5 unsuccessful, will try again in 4.056 seconds
wpantund[1200553]: NCP => [0000469722264][NOTE]-MLE-----: Attach attempt 6 unsuccessful, will try again in 8.055 seconds
wpantund[1200553]: NCP => [0000469734385][NOTE]-MLE-----: Attach attempt 7 unsuccessful, will try again in 16.074 seconds
wpantund[1200553]: NCP => [0000469754542][NOTE]-MLE-----: Attach attempt 8 unsuccessful, will try again in 32.141 seconds
wpantund[1200553]: NCP => [0000469790728][NOTE]-MLE-----: Attach attempt 9 unsuccessful, will try again in 64.286 seconds
wpantund[1200553]: NCP => [0000469859087][NOTE]-MLE-----: Attach attempt 10 unsuccessful, will try again in 128.529 seconds
wpantund[1200553]: NCP => [0000469991646][NOTE]-MLE-----: Attach attempt 11 unsuccessful, will try again in 257.040 seconds
wpantund[1200553]: NCP => [0000470252771][NOTE]-MLE-----: Attach attempt 12 unsuccessful, will try again in 514.060 seconds
wpantund[1200553]: NCP => [0000470770962][NOTE]-MLE-----: Attach attempt 13 unsuccessful, will try again in 1028.109 seconds
wpantund[1200553]: NCP => [0000471803065][NOTE]-MLE-----: Attach attempt 14 unsuccessful, will try again in 1198.470 seconds
wpantund[1200553]: NCP => [0000473005561][NOTE]-MLE-----: Attach attempt 15 unsuccessful, will try again in 1201.757 seconds
wpantund[1200553]: NCP => [0000474211468][NOTE]-MLE-----: Attach attempt 16 unsuccessful, will try again in 1199.667 seconds
wpantund[1200553]: NCP => [0000475415217][NOTE]-MLE-----: Attach attempt 17 unsuccessful, will try again in 1202.011 seconds
wpantund[1200553]: NCP => [0000476621308][NOTE]-MLE-----: Attach attempt 18 unsuccessful, will try again in 1199.179 seconds
wpantund[1200553]: NCP => [0000477824595][NOTE]-MLE-----: Attach attempt 19 unsuccessful, will try again in 1199.457 seconds
wpantund[1200553]: NCP => [0000479028133][NOTE]-MLE-----: Attach attempt 20 unsuccessful, will try again in 1199.546 seconds
wpantund[1200553]: NCP => [0000480231740][NOTE]-MLE-----: Attach attempt 21 unsuccessful, will try again in 1201.013 seconds
wpantund[1200553]: NCP => [0000481436772][NOTE]-MLE-----: Attach attempt 22 unsuccessful, will try again in 1199.021 seconds
```

theHurc, 2020-10-27:
Yes, I see that too when I run with the base project. In FreeRTOS I think I have memory corruption issue somewhere. That's for taking a look. I seems to be my bug.

closed 2020-10-27 by theHurc