openthread/openthread #5696: Implementing support for multi-protocol chip

nikita-s-wrk, 2020-10-23:
Analyzing ways to implement driver support for radio+mcu SoC to be used as Border Router. As I can see, currently most common way to do that currently is to build firmware for SoC and then connect it to host CPU using SPINEL-over-UART/SPI 
I.e. posix application expects uart transport: ```spinel+hdlc+uart:///dev/ttyACM0```

But since we use proprietary serialization protocol over SPI that also includes BLE/ZigBee/etc protocols, we could not reuse this model.
However, we have proprietary 802.15.4 MAC inteface available on host CPU that could be used in openthread by implementing corresponding drivers. We are currently using it this way:

```[OT CLI with drivers] <--Proprietary ioctl interface--> [MAC interface in kernel] <-SPI-> [SoC with Radio] ```

ot-br-posix implementation expects to have SPINEL-over-SPI/UART and we are looking for a ways to be compatible with mainline openthread, so adding additional SPINEL-over-proprietary is not a good option.

I guess most appropriate way to do that would be to somehow build RCP with mac drivers for host CPU in that way:

```[ot-br-posix binary] <-SPINEL-> [ot-rcp binary] <--Proprietary ioctl interface--> [MAC interface in kernel] <-SPI-> [SoC with Radio]```
Similar to how it is used in simulation, but with real calls to MAC. Before wpantund got deprecated, I was able to build ot-ncp-ftd for host and use it for border router.

Is that right way to go from openthread perspective? 
Are there any existing or in-progress implementation for multi-protocol chips?

jwhui, 2020-10-23:
@nikita-s-wrk , if the IEEE 802.15.4 MAC functionality is directly callable on the host, I suggest directly the `otPlatRadio` APIs such that they directly call your IEEE 802.15.4 MAC APIs. This would avoid the overhead of dealing with spinel.

You can find one example here: [examples/platforms/qpg6095/radio.c](https://github.com/openthread/openthread/blob/master/examples/platforms/qpg6095/radio.c)

nikita-s-wrk, 2020-10-23:
@jwhui 
Thanks for reply. 
But in that case it will require to have "special" build for applications like ot-br-posix, right?

jwhui, 2020-10-23:
> @jwhui
> Thanks for reply.
> But in that case it will require to have "special" build for applications like ot-br-posix, right?

@nikita-s-wrk , it depends on what your goals are. If you prefer to keep spinel (and the overhead of additional IPC), then you can follow a model similar to how we run [ot-daemon with a "simulated" RCP](https://github.com/openthread/openthread/tree/master/src/posix#daemon-mode-support). In your case, replace the simulated RCP executable with one that calls your IEEE 802.15.4 MAC APIs.

nikita-s-wrk, 2020-11-03:
> @nikita-s-wrk , it depends on what your goals are. If you prefer to keep spinel (and the overhead of additional IPC), then you can follow a model similar to how we run [ot-daemon with a "simulated" [...]

I see doing it like this and keeping spinel is the only option to deliver back to mainline openthread as platform, right?

bukepo, 2020-11-03:
We intentionally keeps the spinel protocol in the URL, aiming at being extensible for other radio protocols.

jwhui, 2020-11-03:
> I see doing it like this and keeping spinel is the only option to deliver back to mainline openthread as platform, right?

@nikita-s-wrk , I mentioned two different approaches above:

- Implement `otPlatRadio*` APIs directly.
- Implement RCP+spinel and use IPC.

Both approaches are currently used within the GitHub repo.

Is there an alternative approach you'd like to consider/propose?

nikita-s-wrk, 2020-11-19:
Hi Jonathan,

Implementing otPlatRadio directly will break compatibility with apps i.e. otbr-posix that I see as crucial part to be aligned to openthread project.

RCP+spinel on host side is looking good to me. However, we should ensure it will not bring significant latency due to IPC.

Different option from that could be: implement uart-over-proprietary in kernel driver and bridge SPINEL to SoC over it, running RCP as part of firmware. Downside of that - we'll be not able to update openthread without rebuilding firmware and integrating rebuild in openthread build system would be not trivial. Also, memory-constrained or fw-less chips will not work that way obviously.

jwhui, 2020-11-19:
@nikita-s-wrk , I agree that it seems the RCP is the quickest path forward for you. Note that the quickest path to implementing RCP is by implementing the `otPlatRadio*` APIs - doing so will allow you to leverage the RCP implementation in [src/ncp](https://github.com/openthread/openthread/tree/master/src/ncp).

jwhui, 2021-05-27:
Closing stale issue.

closed 2021-05-27 by jwhui