openthread/openthread #5698: ot-daemon drops responses with InvalidSourceAddress when using service ALOCs

dismirlian, 2020-10-23:
Communication from host to SEDs fails if using service ALOCs, from commit d0d2b71af38733fd539ce939f1c10f05316a678f onwards. I don't know if this is a bug or a misconfiguration on our side.

- Configuration: ot-daemon+RCP -> 10 SEDs.
- Host adds service ALOCs, eg. a CoAP server and a NTP server.
- Devices reach the servers using their service ALOCs.

Before that commit, everything worked as expected. From this commit onwards, we see that every server response is dropped with `[[WARN]]-PLAT----: processTransmit: InvalidSourceAddress`.

Thanks!

jwhui, 2020-10-23:
@bukepo , what was the motivation for #5632?

jwhui, 2020-10-30:
Resolved by #5747

closed 2020-10-30 by jwhui