openthread/openthread #5712: [ot-cli] no-reset option causes app to crash

latheepanm, 2020-10-26:
Description:
When launching the ot-cli posix app with the no-reset option as a command line argument, the app crashes immediately.

Command:
`./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli 'spinel+hdlc+uart:///dev/ttyACM2?no-reset'`

Logs:
```
Oct 26 13:39:13 latheepan-Latitude-7400 ./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[8830]: Running OPENTHREAD/20191113-01338-g12b967cce; POSIX; Oct 26 2020 12:53:47
Oct 26 13:39:13 latheepan-Latitude-7400 ./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[8830]: Thread version: 2
Oct 26 13:39:15 latheepan-Latitude-7400 ./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[8830]: [[CRIT]]-PLAT----: WaitResponse() at /home/latheepan/githubs/openthread/src/posix/../../src/lib/spinel/radio_spinel_impl.hpp:1388: RadioSpinelNoRespo
```

bukepo, 2020-10-26:
It is necessary to reset RCP on start so that RCP and host are in sync.

latheepanm, 2020-10-27:
@bukepo thank you for that! After reading your comments here and on the PR, it appears that we have misinterpreted/misunderstood the purpose of the no-reset feature (maybe the documentation could highlight that a reset of the RCP is still required through other means?). For our application, it would be desirable to not send a reset to the RCP when an app launches. Is there any other way to ensure that they are in sync? Is there a recommended set/order of commands that should be sent to the RCP?

bukepo, 2020-11-02:
@latheepanm I don't think it's easy to make host and RCP work properly without sync at start. RCP holds quite a few states, such as short address, extended address, source match table, transmit frame, frame counters and so on. Besides, the spinel protocol also keeps track of next expected transaction id, which is unknown if RCP is not reset. May I know the reason you don't want the reset of RCP?

latheepanm, 2020-11-02:
@bukepo We are working on a project to add multi-PAN support to the RCP, so that multiple host applications can use a single RCP for different networks concurrently.  (The networks have to be on the same channel.)  We're planning on opening an issue soon to contribute our work on this to OpenThread. 

One of the issues we are facing is that, after the first application launches and sets up some of the parameters on the RCP, the second application launches and issues a RESET command. However, this RESET command causes the first application and RCP to become out-of-sync (erases the parameters that were set by the first application). We need the second application to not issue the RESET command and just set its own parameters on the RCP. One alternative (maybe) is to allow the RESET to happen and then catch and handle the response, similar to PR #5694 . Do you believe PR #5694  is "complete" or would you advise something else?

bukepo, 2020-11-03:
I don't think spinel was designed to support multiple hosts. How does all the applications sync transaction ids?

https://github.com/openthread/openthread/blob/master/src/ncp/ncp_base.cpp#L718

bukepo, 2020-11-03:
@latheepanm What's the maximum number of host applications? The spinel protocol has a [`IID`](https://github.com/openthread/openthread/blob/master/src/lib/spinel/spinel.h#L79) field in its frame header, which might be used as an identifier to distinguish the source of an spinel command. The RCP may take different actions based on the identifier.

latheepanm, 2020-11-03:
@bukepo you are correct, at the moment we leverage the IID field within the spinel header to distinguish between host applications. However, the IID field is only 2 bits so we do have an upper limit there (on the host, we filter data based on IID and we kept IID 0 to represent "unsolicited" messages so that both applications can receive this data; therefore the upper limit is 3 in our case). 

We had to do a bit of modifications to support this and in regards to syncing the transaction ids, what we did was we augmented the ResponseEntry struct with an IID field/member and stored that along with the transaction ids. At the moment, the if condition that you refer to will fail most of the time, but the entry will only be marked removed if the tid and iid both match. This needs to be improved to avoid the if condition you pointed to from always/mostly failing.

bukepo, 2020-11-04:
@latheepanm is it possible to submit your full modifications? This can help us understand the impact.

latheepanm, 2020-11-05:
@bukepo we will post an issue as per the contribution guidelines. We'll highlight this issue #5712 and maybe we can continue the conversation there? 

Would you recommend including the code as well? If so, where?

jwhui, 2021-01-29:
Closing stale issue.

closed 2021-01-29 by jwhui