openthread/openthread #5724: [style] llvm@10 brew formula not found

lmnotran, 2020-10-27:
**Describe the bug**
`./script/bootstrap`[ tries to install the](https://github.com/openthread/openthread/blob/master/script/bootstrap#L108) `llvm@10` brew formula, but I don't think it exists. It's not listed in [the formulas available](https://formulae.brew.sh/formula/) in brew's core tap
[image]

**To Reproduce**
Run `./script/bootstrap` on a mac where `clang-format` is not installed

**Expected behavior**
llvm version 10 should be installed

**Console/log output**
```java
$ brew install llvm@10
Error: No available formula or cask with the name "llvm@10".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

wgtdkp, 2020-10-28:
@lmnotran To unblock yourself from this issue, you can remove the llvm@10 installing in the bootstrap script as a workaround.

bukepo, 2020-10-28:
Did you try update brew repo first?

```bash
brew update
```

lmnotran, 2020-10-28:
Yup! It's the first thing I tried. I think the problem is that the `llvm@10` package doesn't seem to exist in the `brew` core tap, as mentioned above.

```java
> brew update
Already up-to-date.

> brew install llvm@10
Error: No available formula or cask with the name "llvm@10".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

wgtdkp, 2020-11-01:
@jwhui I think it is true that homebrew doesn't provide `llvm@10`.  `clang-format-9` is available with `apt-get` and `llvm@9` is available with `brew`, should we use `clang-format-9` instead?

simonlingoogle, 2020-11-01:
> @jwhui I think it is true that homebrew doesn't provide `llvm@10`. `clang-format-9` is available with `apt-get` and `llvm@9` is available with `brew`, should we use `clang-format-9` instead?

I think it's a good idea to downgrade to `llvm@9` for both `clang-tidy` and `clang-format`.

jwhui, 2020-11-02:
> @jwhui I think it is true that homebrew doesn't provide `llvm@10`. `clang-format-9` is available with `apt-get` and `llvm@9` is available with `brew`, should we use `clang-format-9` instead?

Agree that we should downgrade to clang-format-9 and clang-tidy-9. Just submitted #5761.

PR #5761 (2020-11-03): [pretty] downgrade to clang-format/tidy v9.0
Resolves #5724
closed 2020-11-03 by jwhui