openthread/openthread #5726: CodeLab Formatting Issues: "Simulating a Thread network using OpenThread in Docker"

wabrams, 2020-10-27:
On Step 3 (Simulate a Thread network):
[image]
"1**. Start Node 1**" (formatting showing ** instead of bolding)
[image]
"If you don't see the &gt; prompt after running this command, press enter**.**" (errors with the &gt character code (should be >) and bolding the period (shouldn't see ** for bolding))

wabrams, 2020-10-27:
apologies if this isn't the correct place, but the "Report a mistake" widget in the corner directs me here.

Vyrastas, 2020-10-28:
This is the right place... thanks for letting us know.  They migrated all Codelabs to a new backend recently, I'll have to see why that particular section isn't formatting correctly.

wabrams, 2020-10-28:
yea no problem. tested it in firefox, edge, and chrome (no extensions) before posting. didn't find the source for that page, otherwise i would've just submitted a pull request.

Vyrastas, 2020-10-29:
This is updated now!

closed 2020-10-29 by Vyrastas