openthread/openthread #5746: [nrf] Trying to do something after a frame received in radio but before ack sent

Irving-cl, 2020-10-30:
I'm now implementing Enhanced-ACK Based Probing. For this feature we need to aggregate Link Metrics data from frames sent from a specific Initiator, and put the aggregated data into the Enhanced-ACK sent to that Initiator.

My plan is to put the process in platform radio (I.E. implement the aggregation and fetch of data in `examples/platforms/xxx/radio.c`). The main reason for doing this is to ensure that the Link Metrics data in the latest Enh-ACK would contain the information of the latest received frame. 

For example, after Enhanced-ACK Based Probing is configured, the Initiator sends a 2015 frame **A** to the Subject. And the Subject sends an Enh-ACK containing Vendor-Specific IE. We want the Link Metrics data of frame **A** aggregated in the IE. If we aggregate it in OpenThread core, I think we won't be able to make this work. (The ack would have already been sent out when the code reaches higher layer).

So if I do the aggregation and fetch in platform radio, maybe do it in `nrf_802154_received_timestamp_raw`, can we ensure the process to be completed before ack being sent out? Or it's better to do this in `nrf_802154_tx_ack_started`? But as in `nrf_802154_tx_ack_started` we only have the generated ack, instead of the original received frame, I would prefer to do this process with the received frame. 

In short, I want to do something immediately after a frame is received in radio but before the ack sent out.

What would be your suggestions on this? @hubertmis @edmont

hubertmis, 2020-10-30:
> The main reason for doing this is to ensure that the Link Metrics data in the latest Enh-ACK would contain the information of the latest received frame.

I think that this requirement is the reason why the feature should be implemented in the 802.15.4 driver. The API of 802.15.4 driver is designed in the way that it does not guarantee strict timings between calls and real radio operations. It is possible to call 802.15.4 driver API on one CPU core, while radio operations are executed on other one with nondeterministic timings of IPC between. When OpenThread receives notification that a frame was received, the notification can be delayed by IPC.

We have a proof of concept injecting CSL data to CSL IEs in the 802.15.4 driver. I think similar feature should be implemented to inject Link Metrics data to IEs in the 802.15.4 driver. I can share this PoC with you if you are interested.

Irving-cl, 2020-10-30:
@hubertmis 

I agree to inject Link Metrics data using the the similar method as we did for CSL. However we need the `Lqi` and `Rssi` information for the latest received frame. But it seems that they can only be got in `nrf_802154_received_timestamp_raw`. And in `nrf_802154_tx_ack_started`(where we can inject the IEs), we only have the generated ACK. How can we access `Lqi` and `Rssi` in this IPC?

hubertmis, 2020-10-30:
@Irving-cl 
The code that you use currently to update CSL IE data won't work in nRF53. `nrf_802154_tx_ack_started()` function won't be available there. We've prepared a PoC in the 802.15.4 driver that prepares full Enh-ACK, including valid CSL IE data. We'll have to replace `nrf_802154_tx_ack_started()` based CSL IE preparation with new functions in 802.15.4 driver, to support Thread 1.2 in nRF53.

Similarly, we should not use `nrf_802154_tx_ack_started()` to prepare Link Metrics IEs in Enh-ACK. It could work in nRF52 family, but won't work in nRF53. I think it would be a waste to prepare temporary solution (based on `_tx_ack_started()`)  now and later move implementation to 802.15.4 driver. I suggest to implement injecting Link Metrics data to Enh-ACKs in the 802.15.4 as a first step.

Irving-cl, 2020-10-30:
@hubertmis 

That sounds great. I'm looking forward to that. I'm still going to implement it on nRF52 first. I would inject the data in 802.15.4 driver and make the code as reusable as possible when new APIs are introduced. And the solution on nrf52 would only give the data for frames before the latest one.

Of course I'd like to see your PoC. I'm recently drafting a document for platform requirements to support Low Power features. So I'd like to have a general idea about your solution and think how we would later implement it with new APIs.

hubertmis, 2020-11-01:
@Irving-cl , I think your implementation must be part of the 802.15.4 driver anyway. I don't think public API of the driver allows higher layers to access LQI or dBm data before ACK is transmitted.

Irving-cl, 2020-11-02:
@hubertmis 

I will make the implementation as part of the 802.15.4 driver. We don't need to allow higher layers to access LQI or dBm before ACK is transmitted. But I think we may need the 802.15.4 driver to do that. The driver itself should be able to get LQI and dBm for a received frame, and inject IE that related to that data to the ACK it transmit soon.

hubertmis, 2020-11-02:
@Irving-cl , we've just decided that @MarekPorwisz and his team are going to take care of the PoC I've mentioned. They are going to clean it up and share in a week or so.

Irving-cl, 2020-11-05:
@hubertmis @MarekPorwisz

Please let me know if you have any updates for the new PoC. I'm very interested in the new PoC and want to see how we could implement low power features with it. And I would still implement Enhanced-ACK Probing on nrf52 (not perfect, with the problem I described above).

closed 2020-11-05 by Irving-cl