openthread/openthread #5759: port unreachable after get observable coap command

nicho2, 2020-11-02:
sometimes, i can't register a ressource observable coap, and i have an ICMP message (port unreachable) but the device seems answer correctly.

below, you will find a wireshark capture:

the BR has ips : [ "fdbb:ddbd:7576:8ac9:0:ff:fe00:2800", "fdbb:ddbd:7576:8ac9:b6af:bb5a:2634:99fa", "fe80:0:0:0:d872:ea23:54e4:2713" ]
the device with coap server : [ "fdbb:ddbd:7576:8ac9:0:ff:fe00:800", "fdbb:ddbd:7576:8ac9:c8da:c843:1758:1f2e", "fe80:0:0:0:5c2b:7c26:aff:1a2b" 

between lines  1 - 43 : case where all works fine

after line 55 : case where register is not possible

[coap_en_observer.zip](https://github.com/openthread/openthread/files/5473618/coap_en_observer.zip)

jwhui, 2020-11-03:
@nicho2 , thanks for raising this issue. It seems that the UDP Port numbers are not matching up. Looking at packet 55, the CoAP GET message is sent with source port 53653. However, the CoAP Content messages in packet 56 is being sent to UDP port 46200.

What CoAP implementation are you using? Can you provide more details on your setup and steps to reproduce the issue?

nicho2, 2020-11-04:
hello @jwhui, thanks for your observation. Perhaps , when i cancelled the observe (line 43) the device don't realize correctly this operation.

for the client, i use the library python Coapthon (in python 3) and this code to try

from coapthon.client.helperclient import HelperClient

class clientObs():
    def __init__(self,host,port=5683):
        self.client = HelperClient(server=(host, port))
        self.last_response = None

    def observe(self,path):
        response = self.client.observe(path,self.callback)
        if response is not None:
            print(response.pretty_print())
        return response

    def callback(self,response):
        print('reception callback')
        if response is not None:
            self.last_response = response
            print(response.pretty_print())

    def observe_stop(self):
        self.client.cancel_observing(self.last_response,send_rst=True)

host ='fdbb:ddbd:7576:8ac9:c8da:c843:1758:1f2e'
path = "ble_name"
c = clientObs(host)
c.observe("act")

and device side:
it's sdk nrf52840 with nordic coap server and stack nrf5_thread_v4_1_0

jwhui, 2020-11-04:
@nicho2 , thanks for the additional detail. This appears to be an issue with Coapthon.

Looking at the packet trace again, this is what I observe:

1. Packet 40, the server sends a CoAP confirmable notification.
2. Packet 41, the client sends a CoAP ACK.
3. Packet 43, the client sends a CoAP Reset.

It seems that the CoAP Client is trying to cancel the subscription simply by sending a CoAP Reset message. However, this behavior does not conform to [RFC 7641 Section 3.6](https://tools.ietf.org/html/rfc7641#section-3.6):

```
3.6 Cancellation

   A client that is no longer interested in receiving notifications for
   a resource can simply "forget" the observation.  When the server then
   sends the next notification, the client will not recognize the token
   in the message and thus will return a Reset message.  This causes the
   server to remove the associated entry from the list of observers.
   The entries in lists of observers are effectively "garbage collected"
   by the server.

   Implementation Note:  Due to potential message loss, the Reset
      message may not reach the server.  The client may therefore have
      to reject multiple notifications, each with one Reset message,
      until the server finally removes the associated entry from the
      list of observers and stops sending notifications.

   In some circumstances, it may be desirable to cancel an observation
   and release the resources allocated by the server to it more eagerly.
   In this case, a client MAY explicitly deregister by issuing a GET
   request that has the Token field set to the token of the observation
   to be cancelled and includes an Observe Option with the value set to
   1 (deregister).  All other options MUST be identical to those in the
   registration request except for the set of ETag Options.  When the
   server receives such a request, it will remove any matching entry
   from the list of observers and process the GET request as usual.
```

In other words, the client should:
1. send a Reset in response to receiving a notification
2. send a GET request with an Observe Option

I don't see either of the above behaviors in the packet trace.

nicho2, 2020-11-05:
@jwhui ,

Yes, it's an issue with Coapthon3.
I have modify this part to send a frame with observe = 1
and now , it's OK

Thank you very much!!

jwhui, 2020-11-05:
@nicho2 , great to hear that it is working now! Thanks.

closed 2020-11-05 by jwhui