openthread/openthread #5767: [nrf52840] first packet after reboot has wrong security

brianljt, 2020-11-03:
**Describe the bug**
Enable Thread 1.2. Leader using RCP platform. After leader rebooted(ctrl-c and restart), first packet sent out will have wrong security. Second and the following packets are correct. Issue is gone when using builtin mbedtls.

**To Reproduce**

1. Git commit id
    tip of master. I'm using f886f34d0062a6e89639324c9d9f8b7bb0d67430
2. IEEE 802.15.4 hardware platform
    nrf52840 DK
3. Build steps
    make -f examples/Makefile-nrf52840 THREAD_VERSION=1.2
4. Network topology
    4.1 Start nrf52840DK as RCP leader: ./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli -v "spinel+hdlc+uart:///dev/ttyACM1"
    4.2 Start another nrf52840DK as child. Child attach to the leader. Ping can go through from leader to child
    4.3 Kill posix app(ctrl - c) and restart. Start Thread.
    4.4 Ping from this leader to child.

**Expected behavior**
Expect ping response from child.

**Console/log output**
```bash
# initial setup, first ping can get ping response
> factoryreset
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[475264]: Running OPENTHREAD/20191113-01381-g0b0d6eef1-dirty; POSIX; Nov  3 2020 14:52:31
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[475264]: Thread version: 3
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[475264]: RCP version: OPENTHREAD/20191113-01381-g0b0d6eef1-dirty; NRF52840; Nov  3 2020 14:53:25
> panid 0xbeef
Done
> ifconfig up
Done
> thread start
Done
> ping fe80:0:0:0:9cc5:a92c:227e:cd02
Done
> 16 bytes from fe80:0:0:0:9cc5:a92c:227e:cd02: icmp_seq=1 hlim=64 time=26ms
>
# Kill posix app and restart
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli -v "spinel+hdlc+uart:///dev/ttyACM1"
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[488654]: Running OPENTHREAD/20191113-01381-g0b0d6eef1-dirty; POSIX; Nov  3 2020 14:52:31
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[488654]: Thread version: 3
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[488654]: RCP version: OPENTHREAD/20191113-01381-g0b0d6eef1-dirty; NRF52840; Nov  3 2020 14:53:25
>
> ifconfig up
Done
> thread start
Done
>
# no response for first ping
> ping fe80:0:0:0:9cc5:a92c:227e:cd02
Done
# From second frame onward, ping is OK
> ping fe80:0:0:0:9cc5:a92c:227e:cd02
Done
> 16 bytes from fe80:0:0:0:9cc5:a92c:227e:cd02: icmp_seq=2 hlim=64 time=34ms
ping fe80:0:0:0:9cc5:a92c:227e:cd02
Done
> 16 bytes from fe80:0:0:0:9cc5:a92c:227e:cd02: icmp_seq=3 hlim=64 time=26ms
```
Sniffer reports "No encryption key set - can't decrypt" for this packet
[image]

**Additional context**
Issue is gone if CC310 library is not used: make -f examples/Makefile-nrf52840 THREAD_VERSION=1.2 DISABLE_BUILTIN_MBEDTLS=0

brianljt, 2020-11-03:
Thank to @Moandor-y for spotting this issue.

LuDuda, 2020-11-03:
Thanks for spotting this. We'll take a look.

MarekPorwisz, 2020-11-05:
Could you clarify what is flashed on your DKs? As I'm guessing DK 1 uses rcp, but I'm not sure about DK 2. My guess is CLI sample since RCP should not be affected by crypto libs on the board, but I would like to confirm that.
Thanks,
/\/\arek

brianljt, 2020-11-06:
@MarekPorwisz Leader is RCP, child is ot-cli-ftd. This problem is at leader side. With OT1.2 enabled, transmission packet encryption will be done at RCP side. https://github.com/openthread/openthread/pull/5596

MarekPorwisz, 2020-11-06:
@brianljt I must be missing something. I'm not even able to get thread working. Do I need any additional switches to build posix app?
I have my nrf52840 DK connected via debug port.

running following commands:
git clean -f -d -X
./bootstrap
make -f examples/Makefile-nrf52840 THREAD_VERSION=1.2 
make -f src/posix/Makefile-posix THREAD_VERSION=1.2

arm-none-eabi-objcopy -O ihex output/nrf52840/bin/ot-rcp ot-rcp.hex
arm-none-eabi-objcopy -O ihex output/nrf52840/bin/ot-cli-ftd ot-cli-ftd.hex
nrfjprog --program ot-rcp.hex --chiperase --reset
nrfjprog --program ot-cli-ftd.hex --chiperase --reset
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli -v "spinel+hdlc+uart:///dev/serial/by-id/usb-SEGGER_J-Link_000683770451-if00"

I get the following:
> factoryreset
/home/map5/repo/openthread/output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[19767]: Running OPENTHREAD/20191113-01383-gf886f34d0; POSIX; Nov  6 2020 12:41:32
/home/map5/repo/openthread/output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[19767]: Thread version: 2
/home/map5/repo/openthread/output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[19767]: RCP version: OPENTHREAD/20191113-01383-gf886f34d0; NRF52840; Nov  6 2020 09:55:17
> channel 11
Done
> panid 0xabcd
Done
> ifconfig up
Done
> thread start
Done
> /home/map5/repo/openthread/output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[19767]: [CRIT]-PLAT----: exit(1): ProcessRadioStateMachine line 999, radio tx timeout, Failure

brianljt, 2020-11-06:
@MarekPorwisz My steps are similar to yours. It is working OK on my setup. Here is my steps,

```bash
git clean -dfx
./bootstrap
make -f examples/Makefile-nrf52840 THREAD_VERSION=1.2
make -f src/posix/Makefile-posix THREAD_VERSION=1.2
arm-none-eabi-objcopy -O ihex output/nrf52840/bin/ot-rcp ot-rcp.hex
nrfjprog -f nrf52 --chiperase --reset --program ot-rcp.hex -s 683368873
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli -v "spinel+hdlc+uart:///dev/ttyACM1"
```

```bash
 ./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli -v "spinel+hdlc+uart:///dev/ttyACM1"
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[2778977]: Running OPENTHREAD/20191113-01403-gebb2fd53e; POSIX; Nov  6 2020 20:55:27
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[2778977]: Thread version: 3
./output/posix/x86_64-unknown-linux-gnu/bin/ot-cli[2778977]: RCP version: OPENTHREAD/20191113-01403-gebb2fd53e; NRF52840; Nov  6 2020 20:56:02
>
> panid 0xbeef
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
leader
Done
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread-4254  | 1dbe03f027572743 | abcd | e214d27069a239a7 | 11 | -55 | 152 |
| 0 | OpenThread-4254  | 1dbe03f027572743 | abcd | cfe0d8dd4efc2973 | 11 | -38 | 220 |
| 0 | OpenThread-4254  | 1dbe03f027572743 | abcd | 2e109f19b573ecef | 11 | -35 | 232 |
| 0 | NEST-PAN-4B0F    | 7c37b717a7727dc2 | cd1d | 0390eaf92723ee50 | 13 | -57 | 144 |
Done
```

The only differences I noticed are,
1. Maybe we are using different git commit. I have synced to ebb2fd53e.
2. I use `nrfjprog` with parameter `-f nrf52`
3. The timestamp for posix app version and RCP version has 3 hours diff in your log. Are you using same codebase for building posix app and RCP firmware?

MarekPorwisz, 2020-11-13:
Replacing the board helped with my issue.

As for the first packet not being encrypted (It is encrypted but after it is sent ;) )
The encryption is handled by the tx_started radio event that is called when transmission starts. Is is actually a race condition, because the radio does not wait for the function to finish, so in case of any delays the data may be encrypted after the radio transmission.

As for why this is observed only with nrf_scurity and not with builtin mbedtls (this is only a guess): In both cases mbedtls is used for AES but in case of nrf_security this might be the first call to mbedtls so there might by some additional initialization performed extending the processing time.

brianljt, 2020-11-16:
Thank you @MarekPorwisz , that make sense. Can we make this part of initialization process so that the first packet out will use similar processing time like others?

MarekPorwisz, 2020-11-20:
To keep you updated: Extended init time was caused by the AES tables being generated in RAM during the first use, instead of being stored in ROM (MBEDTLS_AES_ROM_TABLES option). I've managed to port everything to our latest nrf_security available in NCS as the older one used in OpenThread did not support changing this. But since this new nrf_security supports additional software crypto backend that is faster than mbedtls, I've switched to using it. Everything seems to be working fine, but I steel need to do a little cleanup. I' hope to create PR today or on Monday.

jwhui, 2020-11-20:
@MarekPorwisz , thanks for the update and working to address this issue!

MarekPorwisz, 2020-11-20:
PR created: https://github.com/openthread/openthread/pull/5837

closed 2020-11-25 by jwhui