openthread/openthread #5773: [github-actions] expect/tun-realm-local-multicast.exp fails intermittently

jwhui, 2020-11-05:
**Describe the bug**

tests/scripts/expect/tun-realm-local-multicast.exp fails intermittently

**Console/log output**

Log archive: [logs_36395.zip](https://github.com/openthread/openthread/files/5492203/logs_36395.zip)

```
# 1 rcp
spawn /home/runner/work/openthread/openthread/build/cmake/openthread-posix-1.1/src/posix/ot-cli spinel+hdlc+uart:///home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.1/examples/apps/ncp/ot-rcp?forkpty-arg=1
routerselectionjitter 1
Done

> panid 0xface
Done

> ifconfig up
Done

> thread start
Done

> state
detached

Done

> state
detached

Done

> state
detached

Done

> state
leader

Done

> extaddr
9284c96b9a2aa25e

Done

> 
# 2 cli
spawn /home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.1/examples/apps/cli/ot-cli-ftd 2
routerselectionjitter 1
routerselectionjitter 1
Done
> panid 0xface
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
child
Done
> state
router
Done
> 
[... 446 lines cut ...]
expect: set expect_out(spawn_id) "exp8"
expect: set expect_out(buffer) "\r\nfdde:ad00:beef:0:f55a:fcc0:f244:81fe"

expect: does "\r\nDone\r\n> " (spawn_id exp8) match glob pattern "Done"? yes
expect: set expect_out(0,string) "Done"
expect: set expect_out(spawn_id) "exp8"
expect: set expect_out(buffer) "\r\nDone"
send: sending "ping fdde:ad00:beef:0:f55a:fcc0:f244:81fe\n" to { exp3 }

expect: does "\r\r\n> " (spawn_id exp3) match glob pattern "Done"? no

expect: does "\r\r\n> ping fdde:ad00:beef:0:f55a:fcc0:f244:81fe\r\n" (spawn_id exp3) match glob pattern "Done"? no

expect: does "\r\r\n> ping fdde:ad00:beef:0:f55a:fcc0:f244:81fe\r\nDone" (spawn_id exp3) match glob pattern "Done"? yes
expect: set expect_out(0,string) "Done"
expect: set expect_out(spawn_id) "exp3"
expect: set expect_out(buffer) "\r\r\n> ping fdde:ad00:beef:0:f55a:fcc0:f244:81fe\r\nDone"

expect: does "" (spawn_id exp3) match glob pattern "16 bytes from fdde:ad00:beef:0:f55a:fcc0:f244:81fe: icmp_seq=1"? no

expect: does "\r\r\n> " (spawn_id exp3) match glob pattern "16 bytes from fdde:ad00:beef:0:f55a:fcc0:f244:81fe: icmp_seq=1"? no
expect: timed out
> FAIL tests/scripts/expect/tun-realm-local-multicast.exp
Error: Process completed with exit code 1.
```

bukepo, 2020-11-06:
still debugging.

librasungirl, 2020-11-06:
The root cause is ping (R2->E4) traffic occurs before thread links are synced given the linear topology R1/L - R2 - R3- E4.

Observations:
1) For the multicast Link Request from R3 when it newly becomes Router, R2 fails to process it here 
https://github.com/openthread/openthread/blob/ebb2fd53efe465fc253dd884794d909836070311/src/core/thread/mle_router.cpp#L604

It is because R2 hasn't get the updated route info from Leader yet (Leader resets the advertise trickle timer, the first MLE advertisement happens around 1s after leader allocates router id for R3), while R3 send multicast Link Request immediately after receiving the `a/as` response.

2) routing info not established between R3 and Leader. In some failing runs, noticing that
  - R3 may not forward a/aq and not send a/an
  - R3 may resend a/an as it doesn't receive ack for the first a/an
  - Leader may still drop the ping after acked a/an from R3

#5776 introduces some time delay to allow route info propagation before ping for now.

However there should be general issues for linear topology, 
a) new router(R3) knows the latest active routers (via a/as response) earlier than its attaching parent (R2) which should wait MLE advertisement
b)  the initial application traffic for the newly attached device may fail as the route information might not be propagated fully yet. the time required depends on the topology. 

It seems to me the simple intuitive suggestion is just wait for application on top of linear topology.

Thoughts?

@jwhui @bukepo

bukepo, 2020-11-09:
Fixed by #5776

jwhui, 2020-11-10:
It appears intermittent failures are still occuring:

https://github.com/openthread/openthread/runs/1380481216

[logs_37203.zip](https://github.com/openthread/openthread/files/5518558/logs_37203.zip)

```
# 1 rcp
spawn /home/runner/work/openthread/openthread/build/cmake/openthread-posix-1.1/src/posix/ot-cli spinel+hdlc+uart:///home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.1/examples/apps/ncp/ot-rcp?forkpty-arg=1
routerselectionjitter 1
Done

> panid 0xface
Done

> ifconfig up
Done

> thread start
Done

> state
detached

Done

> state
detached

Done

> state
detached

Done

> state
leader

Done

> extaddr
1ada4bd496a8877a

Done

> 
# 2 cli
spawn /home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.1/examples/apps/cli/ot-cli-ftd 2
routerselectionjitter 1
Done
> panid 0xface
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
detached
Done
> state
child
Done
> state
router
Done
> 
[... 184 lines cut ...]
"[\r\n]FAILURE_NOT_EXPECTED[\r\n]"? Gate "?FAILURE_NOT_EXPECTED?"? gate=no
expect: timed out
send: sending "state\n" to { exp6 }
Gate keeper glob pattern for 'router' is 'router'. Activating booster.
Gate keeper glob pattern for '[\r\n]FAILURE_NOT_EXPECTED[\r\n]' is '?FAILURE_NOT_EXPECTED?'. Activating booster.

expect: does "\r\n> state\r\ndetached\r\nDone\r\n> state\r\ndetached\r\nDone\r\n> state\r\nchild\r\nDone\r\n> " (spawn_id exp6) match regular expression "router"? Gate "router"? gate=no
"[\r\n]FAILURE_NOT_EXPECTED[\r\n]"? Gate "?FAILURE_NOT_EXPECTED?"? gate=no

expect: does "\r\n> state\r\ndetached\r\nDone\r\n> state\r\ndetached\r\nDone\r\n> state\r\nchild\r\nDone\r\n> state\r\nrouter\r\nDone\r\n> " (spawn_id exp6) match regular expression "router"? Gate "router"? gate=yes re=yes
expect: set expect_out(0,string) "router"
expect: set expect_out(spawn_id) "exp6"
expect: set expect_out(buffer) "\r\n> state\r\ndetached\r\nDone\r\n> state\r\ndetached\r\nDone\r\n> state\r\nchild\r\nDone\r\n> state\r\nrouter"
parent: waiting for sync byte
parent: telling child to go ahead
parent: now unsynchronized from child
spawn: returns {18880}
send: sending "routerselectionjitter 1\n" to { exp7 }

expect: does "" (spawn_id exp7) match glob pattern "Done"? no

expect: does "routerselectionjitter 1\r\n" (spawn_id exp7) match glob pattern "Done"? no
expect: timed out
Error: Process completed with exit code 1.
```

librasungirl, 2020-11-17:
Updated the test script and will submit after more runs of local verification.

librasungirl, 2020-11-17:
#5821 is submitted after all ( >500 runs) pass over ~ 3 hours

closed 2020-11-17 by jwhui