openthread/openthread #5774: [github-actions] v1_2_LowPower_7_2_01_ForwardTrackingSeries.py fails intermittently

jwhui, 2020-11-05:
**Describe the bug**

tests/scripts/thread-cert/v1_2_LowPower_7_2_01_ForwardTrackingSeries.py fails intermittently.

**Console/log output**

[logs_36424.zip](https://github.com/openthread/openthread/files/5492306/logs_36424.zip)

```
==============================
!!! FAIL:./tests/scripts/thread-cert/v1_2_LowPower_7_2_01_ForwardTrackingSeries.py
==============================
2020-11-05 05:08:23,869 - INFO - Creating node 1: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': 'rdn', 'panid': 64206, 'allowlist': [2, 3], 'version': '1.2', 'name': 'LEADER'}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 1
1: extpanid 000db80000000000
1: extaddr 166e0a0000000001
1: panid 64206
1: mode rdn
2020-11-05 05:08:23,876 - INFO - Creating node 2: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': '-', 'panid': 64206, 'allowlist': [1], 'version': '1.2', 'name': 'SED_1'}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 2
2: extpanid 000db80000000000
2: extaddr 166e0a0000000002
2: panid 64206
2: mode -
2020-11-05 05:08:23,881 - INFO - Creating node 3: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': '-', 'panid': 64206, 'allowlist': [1], 'version': '1.2', 'name': 'SSED_1'}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 3
3: extpanid 000db80000000000
3: extaddr 166e0a0000000003
3: panid 64206
3: mode -
2: extaddr
1: macfilter addr add 166e0a0000000002
3: extaddr
1: macfilter addr add 166e0a0000000003
1: macfilter addr allowlist
1: extaddr
2: macfilter addr add 166e0a0000000001
2: macfilter addr allowlist
1: extaddr
3: macfilter addr add 166e0a0000000001
3: macfilter addr allowlist
1: extaddr
2: extaddr
3: extaddr
1: ifconfig up
1: thread start
1: state
2: pollperiod 2000
2: ifconfig up
2: thread start
ActiveOperationalDataset is not implemented yet.
2: state
3: csl period 3125
3: ifconfig up
3: thread start
ActiveOperationalDataset is not implemented yet.
CslSynchronizedTimeout is not implemented yet.
3: state
1: ipaddr
2: ping fe80:0:0:0:146e:a00:0:1
3: ping fe80:0:0:0:146e:a00:0:1
2: linkmetrics mgmt fe80:0:0:0:146e:a00:0:1 forward 1 r p
LinkMetricsManagement is not implemented yet.
LinkMetricsManagement is not implemented yet.
2: linkmetrics query fe80:0:0:0:146e:a00:0:1 forward 1
LinkMetricsQuery is not implemented yet.
LinkMetricsReport is not implemented yet.
2: linkmetrics mgmt fe80:0:0:0:146e:a00:0:1 forward 1 X 
LinkMetricsManagement is not implemented yet.
LinkMetricsManagement is not implemented yet.
3: linkmetrics mgmt fe80:0:0:0:146e:a00:0:1 forward 2 d m
LinkMetricsManagement is not implemented yet.
LinkMetricsManagement is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics probe fe80:0:0:0:146e:a00:0:1 2 1
LinkProbe is not implemented yet.
3: linkmetrics query fe80:0:0:0:146e:a00:0:1 forward 2
LinkMetricsQuery is not implemented yet.
LinkMetricsReport is not implemented yet.
3: linkmetrics mgmt fe80:0:0:0:146e:a00:0:1 forward 2 X 
LinkMetricsManagement is not implemented yet.
LinkMetricsManagement is not implemented yet.
3: linkmetrics query fe80:0:0:0:146e:a00:0:1 forward 2
EE
======================================================================
ERROR: test (__main__.LowPower_7_2_01_ForwardTrackingSeries)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./tests/scripts/thread-cert/v1_2_LowPower_7_2_01_ForwardTrackingSeries.py", line 163, in test
    self.simulator.go(5)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 509, in go
    self.receive_events()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 262, in receive_events
    msg, addr = self.sock.recvfrom(self.MAX_MESSAGE)
socket.timeout: timed out

======================================================================
ERROR: test (__main__.LowPower_7_2_01_ForwardTrackingSeries)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 250, in tearDown
    node.stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 598, in stop
    self.thread_stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 614, in thread_stop
    self.send_command('thread stop')
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 574, in send_command
    self.pexpect.send(cmd + '\n')
  File "/home/runner/.local/lib/python3.6/site-packages/pexpect/popen_spawn.py", line 142, in send
    return self.proc.stdin.write(b)
BrokenPipeError: [Errno 32] Broken pipe

----------------------------------------------------------------------
Ran 1 test in 13.272s

FAILED (errors=2)
LinkMetricsQuery is not implemented yet.
Current nodeid:
None
Current awake:
set()
Current time:
114497208
Current event:
(114497208, 2305, ('127.0.0.1', 9001), 0, 0)
Events:
(114499840, 2304, ('127.0.0.1', 9003), 0, 0)
(115010207, 2302, ('127.0.0.1', 9002), 0, 0)
1: thread stop
Error: Process completed with exit code 1.
```

Irving-cl, 2020-11-06:
Debugging now.

Irving-cl, 2020-11-08:
I ran the action locally with the lastest master branch for about 20 times but cannot reproduce the error. Which PRs are experiencing this failure? @jwhui 

The pcap seems correct. From the log, it looks like a node crashes in the end. I'm thinking that we may need a way to upload core dump files so that we can debug such cases.

jwhui, 2020-12-07:
Resolved by #5921

closed 2020-12-07 by jwhui