openthread/openthread #5783: [thread-cert] LowPower_5_3_01 failed when VIRTUAL_TIME=0

Irving-cl, 2020-11-07:
**Describe the bug**
When running the case `v1_2_LowPower_5_3_01_SSEDAttachment.py` with `VIRTUAL_TIME=0` it fails. Currently the cases are all run with `VIRTUAL_TIME=1`. 

**To Reproduce** Information to reproduce the behavior, including:
```
THREAD_VERSION=1.2 VIRTUAL_TIME=0 CSL_RECEIVER=1 ./script/test build cert tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py
```

1. Git commit id: f886f34d0
2. IEEE 802.15.4 hardware platform: simulation
3. Build steps: `THREAD_VERSION=1.2 VIRTUAL_TIME=0 CSL_RECEIVER=1 ./script/test build`

simonlingoogle, 2020-11-09:
Is it a 100% chance fail?

Irving-cl, 2020-11-12:
> Is it a 100% chance fail?

As I tried, yes.

PR #5785 (2020-11-12): [low-power] fix csl simulation
This PR fixes #5783.

The cause is that, on nRF platform, `otPlatRadioGetNow` always gets the LSB part of an `uint64_t` time. However, on simulation platform, this is not the case. So in
```
            if (otPlatRadioGetNow(&GetInstance()) <
                mTransmitFrame.mInfo.mTxInfo.mTxDelayBaseTime + mTransmitFrame.mInfo.mTxInfo.mTxDelay)
```
we don't get the result expected. `otPlatRadioGetNow(&GetInstance())` is a very big number on simulation platform, and `mTxDelayBaseTime` is the LSB part of the rx time.
closed 2020-11-12 by jwhui