openthread/openthread #5795: [github-actions] cli-ftd-otns failing intermittently

jwhui, 2020-11-10:
**Describe the bug**

cli-ftd-otns failing intermittently

**Console/log output**

https://github.com/openthread/openthread/runs/1380487504

[logs_37213.zip](https://github.com/openthread/openthread/files/5518572/logs_37213.zip)

```
======================================================================
FAIL: testCommissioning (__main__.BasicTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/otns/pylibs/unittests/test_commissioning.py", line 90, in testCommissioning
    self.assertFormPartitions(1)
  File "/tmp/otns/pylibs/unittests/OTNSTestCase.py", line 53, in assertFormPartitions
    self.assertTrue(len(pars) == count and 0 not in pars, pars)
AssertionError: False is not true : {827550411: [1], 1363485286: [2]}

----------------------------------------------------------------------
Ran 3 tests in 0.279s

FAILED (failures=1)
countes {'AlarmEvents': 587, 'RadioEvents': 47, 'StatusPushEvents': 105, 'UartWriteEvents': 380, 'DispatchByExtAddrSucc': 4, 'DispatchByExtAddrFail': 0, 'DispatchByShortAddrSucc': 0, 'DispatchByShortAddrFail': 0, 'DispatchAllInRange': 0}
joins [(2, 0.0, 4.819)]
Error: Process completed with exit code 1.
```

simonlingoogle, 2020-11-10:
[cli-ftd-otns-pcaps.zip](https://github.com/openthread/openthread/files/5518611/cli-ftd-otns-pcaps.zip)

simonlingoogle, 2020-11-10:
This test has been failing from time to time. 
Seems the joiner failed to be commissioned even in a very simple setting with just two nodes.

jwhui, 2021-08-16:
[logs_91982.zip](https://github.com/openthread/openthread/files/6994119/logs_91982.zip)

jwhui, 2022-03-01:
Have not seen this issue recently. Closing.

closed 2022-03-01 by jwhui