openthread/openthread #5801: Validate network name as UTF-8 string

bukepo, 2020-11-12:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Missing validation of Thread network name as a UTF-8 string.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
Validate it.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
None

**Additional context** Add any other context or screenshots about the feature request here.

simonlingoogle, 2020-11-12:
Why is it necessary?

bukepo, 2020-11-12:
> Why is it necessary?

Thread Network Name is defined as UTF-8 string.

abtink, 2020-11-12:
We added and used a `spinel_validate_utf8()` implementation (since spinel the gatekeeper to the OT API on NCP build it checks all inputs parsed as `SPIENL_DATATYPE_UTF8`).

I think we can re-use the same implementation within OT core (and perform common checks in OT).

closed 2020-11-30 by jwhui