openthread/openthread #5802: [github-actions] v1_2_LowPower_5_3_01_SSEDAttachment.py fails intermittently

jwhui, 2020-11-12:
**Describe the bug**

tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py fails intermittently

**Console/log output**

- [logs_37428.zip](https://github.com/openthread/openthread/files/5528157/logs_37428.zip)
- [packet-verification-low-power-pcaps.zip](https://github.com/openthread/openthread/files/5528159/packet-verification-low-power-pcaps.zip)

```
Using virtual time
Using Thread 1.2 stack
File "tests/scripts/thread-cert/run_cert_suite.py", line 114, in parse_args
2020-11-12 03:47:32,938 - INFO - Max jobs: 4
File "tests/scripts/thread-cert/run_cert_suite.py", line 115, in parse_args
2020-11-12 03:47:32,939 - INFO - Multiply: 1
File "tests/scripts/thread-cert/run_cert_suite.py", line 116, in parse_args
2020-11-12 03:47:32,939 - INFO - Test scripts: 4
File "tests/scripts/thread-cert/run_cert_suite.py", line 186, in main
2020-11-12 03:47:32,939 - INFO - Has Backbone tests: False
2020-11-12 03:47:33,327 - INFO - Creating node 1: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': 'rdn', 'panid': 64206, 'allowlist': [2, 3], 'version': '1.2', 'name': 'LEADER'}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 1
1: extpanid 000db80000000000
1: extaddr 166e0a0000000001
1: panid 64206
1: mode rdn
2020-11-12 03:47:33,340 - INFO - Creating node 2: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': 'rdn', 'panid': 64206, 'allowlist': [1], 'version': '1.2', 'name': 'ROUTER', 'router_selection_jitter': 1}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 2
2: extpanid 000db80000000000
2: extaddr 166e0a0000000002
2: panid 64206
2: mode rdn
2: routerselectionjitter 1
2020-11-12 03:47:33,360 - INFO - Creating node 3: {'is_mtd': False, 'is_bbr': False, 'is_otbr': False, 'is_host': False, 'mode': '-', 'panid': 64206, 'allowlist': [1], 'version': '1.2', 'name': 'SSED_1'}
/home/runner/work/openthread/openthread/build/cmake/openthread-simulation-1.2/examples/apps/cli/ot-cli-ftd 3
3: extpanid 000db80000000000
3: extaddr 166e0a0000000003
3: panid 64206
3: mode -
2: extaddr
1: macfilter addr add 166e0a0000000002
3: extaddr
1: macfilter addr add 166e0a0000000003
1: macfilter addr allowlist
1: extaddr
2: macfilter addr add 166e0a0000000001
2: macfilter addr allowlist
1: extaddr
3: macfilter addr add 166e0a0000000001
3: macfilter addr allowlist
1: extaddr
2: extaddr
3: extaddr
1: ifconfig up
1: thread start
1: state
2: ifconfig up
2: thread start
ActiveOperationalDataset is not implemented yet.
2: state
3: csl period 3125
3: ifconfig up
3: thread start
ActiveOperationalDataset is not implemented yet.
CslSynchronizedTimeout is not implemented yet.
3: state
1: rloc16
2: rloc16
3: rloc16
1: ipaddr rloc
2: ipaddr rloc
3: ipaddr rloc
3: ipaddr
2: ping fd00:db8:0:0:0:ff:fe00:5c02
3: ipaddr
2: ping fd00:db8:0:0:0:ff:fe00:5c02 128
EE
======================================================================
ERROR: test (__main__.LowPower_5_3_01_SSEDAttachment)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py", line 98, in test
    timeout=timeout * 2))
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 1342, in ping
    self.simulator.go(1)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 509, in go
    self.receive_events()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 262, in receive_events
    msg, addr = self.sock.recvfrom(self.MAX_MESSAGE)
socket.timeout: timed out

======================================================================
ERROR: test (__main__.LowPower_5_3_01_SSEDAttachment)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 250, in tearDown
    node.stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 598, in stop
    self.thread_stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 614, in thread_stop
    self.send_command('thread stop')
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 574, in send_command
    self.pexpect.send(cmd + '\n')
  File "/home/runner/.local/lib/python3.6/site-packages/pexpect/popen_spawn.py", line 142, in send
    return self.proc.stdin.write(b)
BrokenPipeError: [Errno 32] Broken pipe

----------------------------------------------------------------------
Ran 1 test in 13.176s

FAILED (errors=2)
Current nodeid:
None
Current awake:
set()
Current time:
22497324
Current event:
(22497324, 585, ('127.0.0.1', 9035), 0, 0)
Events:
(22499840, 583, ('127.0.0.1', 9037), 0, 0)
(23066324, 584, ('127.0.0.1', 9036), 0, 0)
1: thread stop
File "tests/scripts/thread-cert/run_cert_suite.py", line 76, in run_cert
2020-11-12 03:47:46,539 - ERROR - Run test v1_2_LowPower_5_3_01_SSEDAttachment_1 failed, please check the log file: v1_2_LowPower_5_3_01_SSEDAttachment_1.log
Traceback (most recent call last):
  File "tests/scripts/thread-cert/run_cert_suite.py", line 73, in run_cert
    env=env)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['python3', './tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py']' returned non-zero exit status 1.
PASS 1 FAIL 0 ./tests/scripts/thread-cert/v1_2_LowPower_test_forward_tracking_series.py
PASS 1 FAIL 0 ./tests/scripts/thread-cert/v1_2_LowPower_7_2_01_ForwardTrackingSeries.py
PASS 1 FAIL 0 ./tests/scripts/thread-cert/v1_2_LowPower_6_1_07_PreferringARouterOverAReed.py
PASS 0 FAIL 1 ./tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py
Error: Process completed with exit code 1.
```

Irving-cl, 2020-11-12:
I think this may relate to #5783.

Irving-cl, 2020-12-06:
The failure is indeed caused by crashed program. And I got the core dump files now:
[image]
I'm trying to find out the cause now.

jwhui, 2020-12-07:
Resolved by #5921

closed 2020-12-07 by jwhui