openthread/openthread #5805: [github-actions] upload core dump file and binaries when test program crashes

Irving-cl, 2020-11-12:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.

We got some cases that fail intermittently, like #5802, #5774. From the logs, it looks like some node crashes:
```
ERROR: test (__main__.LowPower_5_3_01_SSEDAttachment)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./tests/scripts/thread-cert/v1_2_LowPower_5_3_01_SSEDAttachment.py", line 98, in test
    timeout=timeout * 2))
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 1342, in ping
    self.simulator.go(1)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 509, in go
    self.receive_events()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/simulator.py", line 262, in receive_events
    msg, addr = self.sock.recvfrom(self.MAX_MESSAGE)
socket.timeout: timed out

======================================================================
ERROR: test (__main__.LowPower_5_3_01_SSEDAttachment)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 250, in tearDown
    node.stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 598, in stop
    self.thread_stop()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 614, in thread_stop
    self.send_command('thread stop')
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 574, in send_command
    self.pexpect.send(cmd + '\n')
  File "/home/runner/.local/lib/python3.6/site-packages/pexpect/popen_spawn.py", line 142, in send
    return self.proc.stdin.write(b)
BrokenPipeError: [Errno 32] Broken pipe
```
And I'm not sure if it's the program's problem or for some reason the process crashes in github-action's environment. And it's pretty hard to reproduce these failures locally.

**Describe the solution you'd like** A clear and concise description of what you want to happen.

I'm thinking if we can upload core dump file for crashed programs and the binaries so that we can debug the programs locally. I'll check if we can do this. Thoughts? @bukepo @simonlingoogle @Moandor-y @jwhui

jwhui, 2020-11-12:
Uploading core dump files as artifacts on failure seems like a good idea.

Irving-cl, 2020-12-02:
#5861 merged.

closed 2020-12-02 by Irving-cl