openthread/openthread #5806: [github-actions] test_dua_dad.py fails intermittently

jwhui, 2020-11-12:
**Describe the bug**

tests/scripts/thread-cert/backbone/test_dua_dad.py fails intermittently.

**Console/log output**

- [logs_37550.zip](https://github.com/openthread/openthread/files/5532910/logs_37550.zip)
- [thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/5532911/thread-1-2-backbone-results.zip)
- https://github.com/openthread/openthread/runs/1392025462?check_suite_focus=true

```
======================================================================
FAIL: test (__main__.TestDuaDad)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 265, in tearDown
    self._verify_packets(test_info_path)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 293, in _verify_packets
    self.verify(pv)
  File "./tests/scripts/thread-cert/backbone/test_dua_dad.py", line 230, in verify
    pv.verify_dua_registration(ROUTER2, ROUTER2_DUA, pbbr_eth=PBBR2_ETH, pbbr_src64=PBBR2)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/pktverify/packet_verifier.py", line 311, in verify_dua_registration
    assert bq1.sniff_timestamp - dr.sniff_timestamp <= 1.0, bq1.sniff_timestamp - dr.sniff_timestamp
AssertionError: 2.4843719005584717

----------------------------------------------------------------------
Ran 1 test in 167.911s

FAILED (failures=1)
File "tests/scripts/thread-cert/run_cert_suite.py", line 76, in run_cert
2020-11-12 19:29:34,473 - ERROR - Run test test_dua_dad_3 failed, please check the log file: test_dua_dad_3.log
Traceback (most recent call last):
  File "tests/scripts/thread-cert/run_cert_suite.py", line 73, in run_cert
    env=env)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['python3', './tests/scripts/thread-cert/backbone/test_dua_dad.py']' returned non-zero exit status 1.
PASS 0 FAIL 1 ./tests/scripts/thread-cert/backbone/test_dua_dad.py
PASS 1 FAIL 0 ./tests/scripts/thread-cert/backbone/test_mlr_multicast_routing.py
PASS 1 FAIL 0 ./tests/scripts/thread-cert/backbone/test_mlr_multicast_routing_timeout.py
PASS 1 FAIL 0 ./tests/scripts/thread-cert/backbone/test_mlr_multicast_routing_commissioner_timeout.py
File "tests/scripts/thread-cert/run_cert_suite.py", line 88, in cleanup_backbone_env
2020-11-12 19:34:37,062 - INFO - Cleaning up Backbone testing environment ...
Error: Process completed with exit code 1.
```

simonlingoogle, 2020-11-13:
This test verify that the DAD process sends first BB.qry within 1 second after sending DUA.rsp. 

I had seen this failure before and thought it was due to performance issue of Github Action Runner VMs. 

But with an analysis of the pcap, I found that the DUA.req is transmitted. The first DUA.req message is 802.15.4 ACKed but not processed (no DUA.rsp). Then ROUTER2 retransmitted DUA.req and succeeded. 

From the log I found that the DUA.req is dropped due to `UnknownNeighbor`:
```
Nov 12 19:28:14 a96b091e7dff otbr-agent[92]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:67, seqnum:116, type:Data, src:0x1400, dst:0x1000, sec:yes, ackreq:yes
```

This was because the ROUTER2 has not established a link to PBBR2. 
ROUTER2 sends `Link Request` at PKT-336, PBBR sends `Link Accept and Request` at PKT-343. The first DUA.req is at PKT-341, thus being dropped by PBBR. 

I think ROUTER2 should wait for more time to send DUA.req: it should wait for link being established with neighbors. 
The current implementation wait for `mle.GetRouterSelectionJitterTimeout() + 1` if the node is expected to become a Router soon as shown below. But it failed to wait for the link establishment. 
```c++
        else if (mle.IsExpectedToBecomeRouter())
        {
            // Will check again in case the device decides to stay REED when jitter timeout expires.
            UpdateRegistrationDelay(mle.GetRouterSelectionJitterTimeout() + 1);
        }
...
```

I think changing this code to `UpdateRegistrationDelay(mle.GetRouterSelectionJitterTimeout() + kNewRouterRegistrationDelay + 1);` can be a good idea. 

Currently `kNewRouterRegistrationDelay` is 5 seconds. I think can reduce it to 2 seconds because a new Router sends `Link Request` immediately and should expect responses within `kMaxResponseDelay` (1s). 

Thoughts? @jwhui @librasungirl

abtink, 2020-11-16:
@simonlingoogle, let me suggest a possibly different approach: how about updating the test such that we check if the neighbor table has the entries we want (e.g link to ROUTER2 is  established) before proceeding with the next step of test?

simonlingoogle, 2020-11-17:
> @simonlingoogle, let me suggest a possibly different approach: how about updating the test such that we check if the neighbor table has the entries we want (e.g link to ROUTER2 is established) before proceeding with the next step of test?

Note that the DUA registration and DAD process starts **automatically** after `Router2` attached to `Leader2`. The test script does not **"proceed"** with the next step, but only wait for the DAD process to complete and verify the messages afterwards. 

The DUA registration was still successful. But the first DUA.req message was dropped by `Leader2` and only the COAP retransmission of DUA.req was handled.

simonlingoogle, 2020-11-17:
@librasungirl Are you going to propose an enhancement that allows a new Router to establish links to neighbors much faster? I think this test case should be more stable with this enhancement. And I think my suggestions in https://github.com/openthread/openthread/issues/5806#issuecomment-726456989 still works.

librasungirl, 2020-11-17:
@abtink Ideally we may more expect that per msg transmitted over the air could serve its purpose, that is, better to only send the msg when there is big chance that it could be handled properly.

@simonlingoogle for the faster link/route syn thoughts for linear or deeper-hop networks we discussed, I am drafting one proposal and will share for discussion later.

I am fine with the suggestion you proposed so far.

closed 2020-12-01 by jwhui