openthread/openthread #5817: Override radio caps to retries

rodrigobrochado, 2020-11-16:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.  

We use zephyr that has it's own radio HAL and we were having poor throughput results in a network with more than 100 SEDs. After insights from #4508 we increased OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT without success. After digging deeper we found that zephyr radio HAL implementation for our SoC (nrf52840) supports retries, so the [shouldRetx](https://github.com/openthread/openthread/blob/ac86fe52e62e60a66aeeb1c905cb1294709147e9/src/core/mac/sub_mac.cpp#L417) is always false. However, the zephyr HAL makes zero retries. You could check the logic [here](https://github.com/zephyrproject-rtos/hal_nordic/blob/8139a8445900ddbf495f65de21ddeab31ac97951/drivers/nrf_radio_802154/mac_features/nrf_802154_precise_ack_timeout.c#L75).

**Describe the solution you'd like** A clear and concise description of what you want to happen.

Option to either or not check the radio retries caps on [ShouldHandleRetries()](https://github.com/openthread/openthread/blob/ac86fe52e62e60a66aeeb1c905cb1294709147e9/src/core/mac/sub_mac.cpp#L609).

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.

We are using the following on ShouldHandleRetries():
```cpp
#if OPENTHREAD_CONFIG_CHECK_RADIO_RETRIES
    VerifyOrExit(!RadioSupportsRetries(), swRetries = false);
#endif
```

**Additional context** Add any other context or screenshots about the feature request here.

We are on zephyr [ec5823c1](https://github.com/zephyrproject-rtos/zephyr/tree/ec5823c1), that make few modifications on openthreadd ac86fe52e62. It could make more sense to ask for configuration options on HAL maintained by zephyr, but giving the configuration requested above could simplified portability between SoC's and that openhtread support is a lot faster, it would be nice to have it on openthread master.

abtink, 2020-11-16:
Thanks @rodrigobrochado for raising the issue.

Let me give a quick overview of the radio capability model:
- There are a bunch of MAC/radio level functionalities/behaviors that can either be performed by the OT core (part of `SubMac`) or they can be delegated to the radio platform layer (radio driver).
  - These include: ack timeout, CSMA backoff retries, frame retx, energy scan, etc.
- The idea is that radio driver may be able to support/implement these more efficiently (e.g., they may be supported in hw)
- We have a radio platform API, `otPlatRadioGetCaps()` where within OT core we ask platform radio layer about the capabilities it would support.
- If the capability is supported by radio platform layer, then it would be used (implementation in OT would be bypassed), otherwise OT core `SubMac` layer implements the behavior required.

If I understand the issue correctly, on your set-up, the radio platform seems to claim that it would do `OT_RADIO_CAPS_TRANSMIT_RETRIES` but does not do retries. Is this correct? If so, I would suggest changing the radio platform layer to not include this capability in response to `otPlatRadioGetCaps()` (this way the OT core would implemet the frame retries).

jwhui, 2020-11-16:
> After digging deeper we found that zephyr radio HAL implementation for our SoC (nrf52840) supports retries, so the shouldRetx is always false. However, the zephyr HAL makes zero retries.

@LuDuda, thoughts on the above?

rodrigobrochado, 2020-11-16:
Thanks for the responses in so short notice. 

@abtink Yes, your approach would make more sense. I will suppress the OT_RADIO_CAPS_TRANSMIT_RETRIES on the response from zephyr's radio [platform](https://github.com/zephyrproject-rtos/zephyr/blob/ec5823c1ea810f69473346d2a8f845b37ab34515/subsys/net/lib/openthread/platform/radio.c#L615) for now and try to send a message on zephyr's mailing list to check if the zero retry is correct or if I've made something wrong.

One more thing, but now about otbr, that don't use zephyr. I didn't understand yet much of spinel communications between otbr (posix) and a RCP. So, in my case, where I have an otbr container and a nrf5840 RCP build out of this repo "examples", could I say that the flags returned by otPlatRadioGetCaps() would be the following?

https://github.com/openthread/openthread/blob/515b2bb0671d63f4ca4e2abcd028ccc4c6dc61af/examples/platforms/nrf528xx/src/radio.c#L531-L540

So the retries will be specified by OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT as it doesn't return OT_RADIO_CAPS_TRANSMIT_RETRIES, right?

LuDuda, 2020-11-16:
@rodrigobrochado thanks for raising the issue.

This particular bug was fixed few months ago ([here](https://github.com/zephyrproject-rtos/zephyr/pull/27383)).

It seems you use quite old revision. I encourage you to try Nordic Connect SDK instead, which internally includes Zephyr OS and adds additional functionalities (e.g. examples, multiprotocol support) on top of it. The version under v1.4.0 tag has been recently officially certified as well.

rodrigobrochado, 2020-11-16:
Thank you @LuDuda!

We are preparing to upgrade to zephyr 2.4.0 soon, so it's nice knowing this. I've already seen mentions of the SDK, I'll certainly take a deeper look at it asap. Thanks!!

rodrigobrochado, 2020-11-17:
> So the retries will be specified by OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT

I should be getting an `Frame tx attempt %d/%d` for every tx frame retry, right? I'm getting just the last retry printed on otbr log. This behavior is the same when sub_mac doesn't perform the retry.

abtink, 2020-11-17:
@rodrigobrochado, let me try to help provide some inputs/answers:

On `OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT`, this paramter should technically be used as number of retries independent of whether the retx logic is being handled by OT core (`SubMac`) or delegated to radio platform layer:
- The way it is intended to work is that the when a frame is prepared the number of retries is included in the frame itself (and would be set to the configured value).
- The `otRadioFrame` struct (which represents a tx/rx frame), has a field `mMaxFrameRetries` under its `mTxInfo` sub-struct.
- When `SubMac` is taking care of retx, it will use the value in the frame for number of tx attempts.
- The expectation is that when radio platform layer is handling retx, it would also use the value from `otRadioFrame` as number of retx 

----

Related to logging on tx attempt failure and the log line `"Frame tx attempt %d/%d"`:
- It  would be emitted for every tx attempt when OT core (`SubMac`) is taking care of the retx logic. 
- When retx logic is delegated/done by radio platform layer, the logging of failure per tx attempt needs to be done by radio platform as well (up to platform layer to log it or not, and how to log it)
  - But we would still see the same log if all attempts fail (i.e., when we get the `otPlatRadioTxDone()` callback from radio platform layer after all retx attempt by radio platform layer fail).
- When using posix-host and RCP model (which I think would apply to otbr), the OT core is running on posix host side, and it would see RCP as a radio platform abstraction which does support and handle retx logic (so RCP would be the one doing the retx logic).
   - The logs from posix host code would then show `"Frame tx attempt %d/%d"` only when last attempt fails (not per tx)
   - If the RCP logs are enabled (at INFO level or more), we should then see `"Frame tx attempt %d/%d"` in the RCP logs per tx attempt failure.

rodrigobrochado, 2020-11-17:
@abtink thanks for the explanation. In sum, the SubMac "asks" the radio through `otPlatRadioGetCaps()` if he can handle retry. If positive, the `SubMac` skips this logic. If negative, the `SubMac` handle the retries, right?  

By this part  
> and it would see RCP as a radio platform abstraction which does support and handle retx logic   

do you mean that when we have a posix-host (otbr) and a RCP, the retx logic is **always** done by the RCP?   

Because I didn't find any mention to `OT_RADIO_CAPS_TRANSMIT_RETRIES` on `examples` folder related to nrf52840, neither on third_party/NordicSemiconductor.  I did find it on 7 out of 11 `examples` platforms, exactly on the return of theirs `otPlatRadioGetCaps()`, but the nrf528xx one doesn't include the retries caps https://github.com/openthread/openthread/issues/5817#issuecomment-728346188. 

Based on the above, I've previously conclude a nrf RCP would answer to the `SubMac` that it couldn't handle the retries, but through the logs and reassured by your explanation, I could see it's not what's happening. The otbr gets the radio (RCP) caps through spinel on the code bellow, right?

https://github.com/openthread/openthread/blob/e73882255fb551ba890d92735ccc05677ed4ac33/src/posix/platform/radio.cpp#L206-L210

So:
1) Where is the nrf RCP answering the `otPlatRadioGetCaps()` with OT_RADIO_CAPS_TRANSMIT_RETRIES? 
2) If it is really handling the retx, is it using the `mMaxFrameRetries` from `otRadioFrame` set on `Mac`? You mention that it's _expected_ that the platform layer uses it. Is it the case for the RCP?

Thank you!!

abtink, 2020-11-17:
> do you mean that when we have a posix-host (otbr) and a RCP, the retx logic is always done by the RCP?

Yes. 

-----

> The otbr gets the radio (RCP) caps through spinel on the code bellow, right?
> Where is the nrf RCP answering the otPlatRadioGetCaps() with OT_RADIO_CAPS_TRANSMIT_RETRIES?

Yes. Here is flow/chain of interactions:
- Posix host uses `radio.cpp` APIs from `src/posix/platform/radio.cpp`.
- It would call `sRadioSpinel.GetRadioCaps()`
- This in in turn get the caps from RCP using spinel property get command `VALUE_GET(SPINEL_PROP_RADIO_CAPS)` 
- On RCP side, the property getter for `SPINEL_PROP_RADIO_CAPS ` uses `LinkRaw` module API `otLinkRawGetCaps()` 
- The `otLinkRawGetCaps()` maps to `LinkRaw::GetCaps()` which in turn `{ return mSubMac.GetCaps(); }`
- The `SubMac::GetCaps()` then determine the list of cap based on the `OPENTHREAD_CONFIG_MAC_SOFTWARE_{name}` config options (which are in general always enabled in RCP build).

Basically, the RCP firmware uses `LinkRaw` APIs. Note that `LinkRaw` itself includes `SubMac` module. The `LinkRaw` (or in general `SubMac`) themselves looks like a radio platform abstraction (provide same set of APIs) but they can support more capabilities than what radio driver natively supports.

In our RCP (or NCP/CLI) builds the `SubMac` layer always provides all radio capabilities (it may handle them itself or delegate them to the radio driver when supported by radio driver). 

On the host `RadioSpinel` side we have code to verify that the RCP is reporting all required radio capabilities:
https://github.com/openthread/openthread/blob/e73882255fb551ba890d92735ccc05677ed4ac33/src/lib/spinel/radio_spinel_impl.hpp#L318-L333

-----

> If it is really handling the retx, is it using the mMaxFrameRetries from otRadioFrame set on Mac? You mention that it's expected that the platform layer uses it. Is it the case for the RCP?

Yes. RCP would bevae similarly to an NCP build. We have `LinkRaw` and `SubMac` in an RCP build and if radio platform driver does not handle certain capability it is taken care of by `SubMac`.

rodrigobrochado, 2020-11-19:
Thank you @abtink !!!!!!!!!!

I could find the flow after your answer and also checked RCP frame retries with the [sniffer](https://openthread.io/guides/pyspinel/sniffer).

Really appreciate the attention, guys!

closed 2020-11-19 by rodrigobrochado