openthread/openthread #5819: [efr32] radio: should the capabilities include OT_RADIO_CAPS_SLEEP_TO_TX?

dismirlian, 2020-11-16:
Hi, I think the efr32 radio driver is able to transition to TX directly from idle (`OT_RADIO_CAPS_SLEEP_TO_TX`), but this is not declared in `otPlatRadioGetCaps`:

https://github.com/openthread/openthread/blob/515b2bb0671d63f4ca4e2abcd028ccc4c6dc61af/examples/platforms/efr32/src/radio.c#L671-L676

This would help reduce power for SEDs, for instance.

What do you think?

jwhui, 2021-03-10:
@romacdon , thoughts on this issue?

romacdon, 2021-03-10:
Thanks for the ping on this @jwhui.

@dismirlian - I checked with a few people here and you are correct.  We support `OT_RADIO_CAPS_SLEEP_TO_TX`.  Have you already tried this?  If so feel free to open a PR.  If not I will get one of the team members here at Silabs to make the change.

dismirlian, 2021-03-11:
@romacdon I did try it, and I had to make a very small modification (IIRC I removed/changed some `otEXPECT_ACTION` or `assert` to cope with an unexpected sleep->transmit transition). It worked fine, but the radio driver has changed a lot since then; this was some months ago (maybe when I posted the issue?) and I didn't save the modification.

romacdon, 2021-03-11:
Thanks @dismirlian.  I'll get someone to make the official change.  We definitely appreciate all your help.

PR #6329 (2021-03-24): [efr32] bug fixes, UART API changes
- OPENTHREAD_CONFIG_NCP_UART_ENABLE is now OPENTHREAD_CONFIG_NCP_HDLC_ENABLE
- Pull in cli_uart.cpp UART API interface for sleepy-demo apps.
- Fix some erroneous ifdefs that affect 1.2 SoC builds and 1.1 builds.
- Add OT_RADIO_CAPS_SLEEP_TO_TX to radio capabilities (Fixes #5819).
commit 846fffed (2021-03-24): [efr32] bug fixes, UART API changes (#6329)

- OPENTHREAD_CONFIG_NCP_UART_ENABLE is now OPENTHREAD_CONFIG_NCP_HDLC_ENABLE
- Pull in cli_uart.cpp UART API interface for sleepy-demo apps.
- Fix some erroneous ifdefs that affect 1.2 SoC builds and 1.1 builds.
- Add OT_RADIO_CAPS_SLEEP_TO_TX to radio capabilities (fixes #5819).
closed 2021-03-24 by jwhui