openthread/openthread #5826: [nrf52] What could cause `NRF_802154_TX_ERROR_TIMESLOT_DENIED` in `nrf_802154_transmit_raw_at`?

Irving-cl, 2020-11-18:
I'm doing some experiment with CSL on nrf52840 dk. Now I get `OT_ERROR_CHANNEL_ACCESS_FAILURE` in `otPlatRadioTxDone` after using `nrf_802154_transmit_raw_at` to transmit a frame. I used some trick to get the detailed nrf error code. The error is `NRF_802154_RX_ERROR_DELAYED_TIMESLOT_DENIED`. What could be the possible reason for this?

@hubertmis @LuDuda @edmont

edmont, 2020-11-18:
@Irving-cl do you have a procedure to reproduce it?

Irving-cl, 2020-11-18:
@edmont I cannot put those code on github (for policy reason). I can share the firmware and the steps to reproduce it. Would that be enough for you? If you need the code to find out the reason, I think I can share a package, maybe through drive.

edmont, 2020-11-18:
I believe having the code and the instructions on how to reproduce the issue would be the best way to understand it.

Irving-cl, 2020-11-18:
@edmont 

Thanks. Then I'll share a package with you.

hubertmis, 2020-11-18:
@Irving-cl 
I would like to make sure I understood correctly. When you use `nrf_802154_transmit_raw_at` function, you receive `nrf_802154_transmit_failed` error with error code 7 that translates to `NRF_802154_TX_ERROR_TIMESLOT_DENIED`, not `NRF_802154_RX_ERROR_DELAYED_TIMESLOT_DENIED`. Am I correct?

`NRF_802154_TX_ERROR_TIMESLOT_DENIED` can be raised in following conditions:
1. Channel cannot be updated before expected transmission. It may happen only when transmission is scheduled too short before BLE activity.
2. At scheduled transmission time (when timer IRQ is executed), radio driver critical section is taken by lower-priority execution context. It is possible that radio.c calls `nrf_802154_receive()` and during such call timer IRQ fires. Critical section is already taken by `nrf_802154_receive()` and scheduled transmission cannot happen. Such race condition should happen very rarely, unless your code calls 802.15.4 radio driver API in a loop. You can verify if critical section is taken checking `m_nested_critical_section_counter` : https://github.com/openthread/openthread/blob/f7e16fec240439f54998463f6d8ab2945edbe5b3/third_party/NordicSemiconductor/drivers/radio/nrf_802154_critical_section.c#L55 . If its value is greater than 0, critical section is taken.
3. High Frequency clock is not ready when TX should happen. If you transit from RX to TX mode, there should be no problem with hfclk. However, if you transit from sleep to TX (could be through RX, but time from sleep to TX is shorter than crystal ramp up), then crystal might not be ready when the transmission should start. You can tweak crystal ramp up time with this constant: https://github.com/openthread/openthread/blob/f7e16fec240439f54998463f6d8ab2945edbe5b3/third_party/NordicSemiconductor/drivers/radio/rsch/nrf_802154_rsch.c#L25 . You can verify HFclk readiness in `m_approvied_prios` variable captured when the problem occurs (you must stop execution with a debugger on the problem, because content of this variable would change in dozens of microseconds): https://github.com/openthread/openthread/blob/f7e16fec240439f54998463f6d8ab2945edbe5b3/third_party/NordicSemiconductor/drivers/radio/rsch/nrf_802154_rsch.c#L55
4. Radio arbiter prevents radio access for 802.15.4. In dynamic multi-protocol applications access to radio is granted by radio arbiter. It is possible that sometimes arbiter prevents radio access when scheduled transmission should start. This also can be verified in `m_approved_prios`.

I hope these details are good hints to move debugging process forward.

Irving-cl, 2020-11-19:
@hubertmis 

Thanks very much for the hints! 

Let me supplement something about my case. So I found that frames are not sent successfully (from `otPlatRadioTxDone`, and the error is `OT_ERROR_CHANNEL_ACCESS_FAILURE`. From the log I can see that `nrf_802154_transmit_raw_at` is successfully called in `otPlatRadioTransmit`. I found that the only two places to set `kPendingEventChannelAccessFailure` (which can cause `OT_ERROR_CHANNEL_ACCESS_FAILURE`) is in `otPlatRadioTransmit` and `nrf_802154_transmit_failed`. As I said, `otPlatRadioTransmit` was successfully called. So it should be `nrf_802154_transmit_failed` that sets `kPendingEventChannelAccessFailure`. To know the exact error, I added a static variable in `nrf_802154_transmit_failed` and record the `error`. And that error is `7`. So I believe it's `NRF_802154_TX_ERROR_TIMESLOT_DENIED`.

From your hints, I think the most possible reason is that my device transit from sleep to TX. (I haven't verified yet) If this issue only happens for `nrf_802154_transmit_raw_at`? I mean, will `nrf_802154_transmit_raw` experience the same issue? I guess no because we have sleepy end devices and when they do TX the transit from sleep to TX. Am I right about this?

Irving-cl, 2020-11-19:
Follow up:

The reason seems to be transition from sleep to TX. I turned the capability of `OT_RADIO_CAPS_SLEEP_TO_TX` on nrf so that  in `SubMac::BeginTransmit` so that it is enforced to do receiving before transmission, then the transmission was successful:
```
    if ((mRadioCaps & OT_RADIO_CAPS_SLEEP_TO_TX) == 0)                                                                                                                                                                                                                       
    {                                                                                                                                                                                                                                                                                
        error = Get<Radio>().Receive(mTransmitFrame.GetChannel());                                                                                                                                                                                                           
        OT_ASSERT(error == OT_ERROR_NONE);                                                                                                                                                                                                                                   
    }
```

hubertmis, 2020-11-19:
> I mean, will `nrf_802154_transmit_raw` experience the same issue?

This function does not require exact timings. If hfclk is not ready when TX is about to happen, TX is delayed until hfclk is ready. However such delay does not work for nrf_802154_transmit_at, because timings for this function must be strict.

Have you tried to increase the value of `PREC_RAMP_UP_TIME`? We've set this value based on crystal timing data from nRF Product Specification. I guess your crystal start time is longer than examples we took while defining this value.

Irving-cl, 2020-11-19:
@hubertmis 

I would try to increase the ramp up time. How should we choose the proper value for it? Is there any side effect if we set a value that's too big? Or we should just increase the value until it can work successfully?

hubertmis, 2020-11-19:
> How should we choose the proper value for it?

Datasheet of the crystal you use should define its startup time. You can use this value, try to add some margin and verify if it works correctly. 

> Is there any side effect if we set a value that's too big?

The side effect is increased power consumption because the crystal is powered longer than necessary. 

> Or we should just increase the value until it can work successfully?

That's one way to go, but be aware that crystal properties vary between each particular component, it changes with temperature and time. Because of that I would recommend to add some safety margin. The cost would be a little higher power consumption, but the risk that the device would stop working correctly after some time or when the temperature changes would be lowered.

hubertmis, 2020-12-07:
@Irving-cl , could you confirm if the solution I suggested work for you? Can we close this issue now?

Irving-cl, 2020-12-08:
@hubertmis 

Sorry for responding late. I'll do it soon.

Irving-cl, 2020-12-08:
@hubertmis 

Update:

I added the value of `PREC_RAMP_UP_TIME` and tested a couple of times. I tested on nrf52840-dk. The value was originally `480` as defined in `nrf_802154_rsch.c`:
```
#define PREC_HFXO_DEBOUNCE_TIME                256
#define PREC_CRYSTAL_WORST_CASE_POWER_UP_TIME  75
#define PREC_TIMER_GRANULARITY_MARGIN          31
#define PREC_POWER_CLOCK_IRQ_HANDLER_PROC_TIME 50
#define PREC_RTC_IRQ_HANDLER_PROC_TIME         60
#define PREC_RAMP_UP_MARGIN                    8

#define PREC_RAMP_UP_TIME                      (PREC_HFXO_DEBOUNCE_TIME +                \
                                                PREC_CRYSTAL_WORST_CASE_POWER_UP_TIME +  \
                                                PREC_TIMER_GRANULARITY_MARGIN +          \
                                                PREC_POWER_CLOCK_IRQ_HANDLER_PROC_TIME + \
                                                PREC_RTC_IRQ_HANDLER_PROC_TIME +         \
```
I changed the value to `1000`, `2000`, `4800`, `12000`, `48000`. For small values like `1000` and `2000`, The transmission still got `Channel Access Failure`. However for big values like `4800` and larger than that, all the CSL transmissions seem to be done earlier and the receiver cannot receive that frame.

hubertmis, 2020-12-08:
Thanks for the update.

> However for big values like `4800` and larger than that, all the CSL transmissions seem to be done earlier and the receiver cannot receive that frame.

That's interesting. `PREC_RAMP_UP_TIME` should not change timings of the actual transmission. It modifies only the time when HFclk is requested (how long before the transmission). Could you share wireshark captures for greater values like 4800, 12000, 48000? I would like to compare them to look for potential source of timing problems.

Could you also comment what is your LFclk source PPM value and HF crystal PPM value?

Irving-cl, 2020-12-08:
> Could you also comment what is your LFclk source PPM value and HF crystal PPM value?

How should I check the PPM value? @hubertmis

hubertmis, 2020-12-08:
> How should I check the PPM value?

For HF xtal you should find it in datasheet of the crystal.

For LF, if you use crystal, you can check datasheet. If you don't use crystal, just let me know if you select RC oscillator or clock synthesis. I'll get values of PPMs for these sources from nRF5 Product Specification

Irving-cl, 2020-12-08:
I'm just using two NRF52840 DKs (without any other devices). Can we find the PPM value for it? I don't have the datasheet of its crystal. @hubertmis

brianljt, 2020-12-08:
Hi @hubertmis , I think the test was done on Nordic 52840DK using API `nrf_802154_transmit_raw_at` with CCA disabled. Before `nrf_802154_transmit_raw_at` is called, device is in sleep mode. So the default ramp up time in code is not enough for DK when board transit from sleep state to tx directly.

Irving-cl, 2020-12-08:
I can see in the <<nRF52840 Development Kit PCA10056 v1.0.0 Use Guide v1.2>> that the crystal is 32.768 kHz.

hubertmis, 2020-12-08:
Thanks for this information. I assumed you use a custom board with different peripherals what causes the problems.

Taking into account that you test using the same hardware I do, the reason must be purely software. Perhaps some differences between our systems.

Do you know what is average/maximal ISR latency in your system?

hubertmis, 2020-12-08:
Let me explain why I ask about ISR latency.

The delayed transmission in 802.15.4 works in the following way:

1. main context schedules transmission at timestamp X
2. RTC ISR at (X - radio ramp up - precondition ramp up) requests HFclk
3. RTC ISR at (X - radio ramp up) starts transmission

If RTC ISR is delayed, the preconditions are requested too late, or the transmission starts too late. It seems you observe both problems. Because of that I suspect ISR latency being the source of the problem.

Possible workaround for 802.15.4 system is:
RADIO IRQ priority = 0
RTC IRQ priority = 1
Other IRQs in the systems have lower priority.

Possible workaround for multi-protocol with SoftDevice:
RADIO IRQ: don't change it, it is managed by SoftDevice
RTC IRQ priority = 4
Other IRQs in the system have lower priority.

Moreover, for both workarounds masking interrupts (using NVIC or PRIMASK/__irq_disable()) must be only for very short time (like up to 2 microseconds).

Irving-cl, 2020-12-08:
Here is the pcap for the case when setting `PREC_RAMP_UP_TIME` to `4800`: 
[csl.pcapng.zip](https://github.com/openthread/openthread/files/5657867/csl.pcapng.zip)

Note that No.21 (Link Request) and No.29 (Ping, and its re-tx) are sent using CSL transmission. Sending time of No.21 is based on CSL IE of No.20 (seems correct). Sending time of No.29 is based on CSL IE of No.25 (seems incorrect).

 I think there may be some software problems. Just provide the capture for you for more details. BTW, if we disable `OT_RADIO_CAPS_SLEEP_TO_TX` and don't change ramp up time (480), everything goes fine. 

@hubertmis

hubertmis, 2020-12-08:
@Irving-cl , I have one more question about the setup. If I remember correctly, OpenThread MAC keeps a constant value, how long in advance the delayed transmission is requested (is it `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US`?).

I think this value must be greater than `PREC_RAMP_UP_TIME`. If you set `PREC_RAMP_UP_TIME` to `4800`, the OT MAC "in-advancie-csl-time" should be like `5200`

Irving-cl, 2020-12-08:
@hubertmis Ah ok. Currently we set `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US` to `2000`. I also tried setting `PREC_RAMP_UP_TIME` to `2000` or `1000`, but I got `Channel Access Failure`. Does that mean if we want to solve this problem by increasing `PREC_RAMP_UP_TIME`, we have to use greater AHEAD time?

I will try using larger `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US`. But it seems that `5200` blocks our system a little bit longer (and not even sure if the value works). I'm also curious what would happen if the AHEAD time is not greater than `PREC_RAMP_UP_TIME`. Currently as I observed, the transmission is done earlier which is quite weird. Why would that happen? I'm also checking if there's some error in OT logic. BTW, in practice, the actual AHEAD time is not ensured greater than `PREC_RAMP_UP_TIME` even if we set it greater than it because there may be some delay for the software timer.

hubertmis, 2020-12-08:
> Does that mean if we want to solve this problem by increasing `PREC_RAMP_UP_TIME`, we have to use greater AHEAD time?

I'm not sure, I'm trying to order information we have in my mind. I don't have the final conclusion yet.

> I'm also curious what would happen if the AHEAD time is not greater than `PREC_RAMP_UP_TIME`.

By design, calling `nrf_802154_transmit_at()` too late should result in reporting error. I think the function returns `false`. However, I think that such erroneous condition could have lower test coverage and there could be some miscalculation we have not hit in our tests yet.

I'm checking the pcap you posted. I agree that frame 21 is transmitted with correct delay.
The frame with number 29 seems transmitted 1126 microseconds too early, what gives us in 10.272 seconds timespan 109 PPM of imprecision. I think that crystals in nRF52840-DKs are better than 40 PPM. Even if they drifted from each other with maximal value, the result should not exceed 80 PPM. 

Frame 30 is transmitted 1004 us too early (85 PPM)
Frame 31 is transmitted 893 us too early (67 PPM)
Frame 32 is transmitted 796 us too early (54 PPM)

From the numbers above, I suspect that the timestamp of frame 25 was incorrectly captured. I'm not sure yet how to verify. I'll check the code later today to find what variables might be helpful to capture.

Irving-cl, 2020-12-08:
@hubertmis 

Thanks. I'm also working on this to see if there's any other error in our code.

hubertmis, 2020-12-08:
I calculated the values a little more. It seems that RTC of failing device drifts around 75 PPM from sniffers TIMER. But it does not explain why timestamp inaccuracy is around 2 ms. I think we need more data to narrow down the problem.

@Irving-cl , could you repeat your test with more ping requests? I mean after series of 4 retransmission, could you start another sequence or two?
Would it be possible to run the same test starting ping sequence with different delays from synchronization "Link Accept" message? Like 2 seconds after, 5 seconds after, 10 seconds after, 15 seconds after. I'm curious if such delay would affect the results.

I have one more question about the test setup. It seems that both devices include CSL IE in data frames. Does it mean it is CSL to CSL communication? Is Link Req -> Link Acc+Req -> Link Acc sequence sent using CSL? Are any of the devices entering sleep state during this sequence?

hubertmis, 2020-12-08:
@Irving-cl , or alternatively could you share your project, or elf files with debug symbols enabled? I think that running it in trace debugger would rapidly answer what's the source of the timing problem.

Irving-cl, 2020-12-09:
@hubertmis 

> or alternatively could you share your project, or elf files with debug symbols enabled? I think that running it in trace debugger would rapidly answer what's the source of the timing problem.

Sorry I'm not allowed to share it due to our policy. Could you give me some guide about how to use the trace debugger? I'm interested in that and I think it may help a lot in our future work.

> I have one more question about the test setup. It seems that both devices include CSL IE in data frames. Does it mean it is CSL to CSL communication? Is Link Req -> Link Acc+Req -> Link Acc sequence sent using CSL? Are any of the devices entering sleep state during this sequence?

Yes, we're doing some experimental work for communication between two SSEDs. That's why Sleep->TX transition happens. In short, the first Link Req and the ping were sent using CSL transmission. And the Link Acc+Req and Link Acc are sent directly, not CSL transmission. So they would both enter sleep state at most time, and wake up to receive or transmit when there's a need.

hubertmis, 2020-12-09:
> Could you give me some guide about how to use the trace debugger?

I meant this debug probe: https://www.segger.com/products/debug-probes/j-trace/models/j-trace/ . It gets program flow while the application is running and among other things it shows preemptions or other ISR delay causes.

hubertmis, 2020-12-09:
@Irving-cl , could you share a pcap file with `PREC_RAMP_UP_TIME` set to default value? I would like to compare them.

Irving-cl, 2020-12-09:
@hubertmis 

When we use default `PREC_RAMP_UP_TIME` and `SLEEP_TO_TX` enabled, we cannot see the ping because it wasn't sent (because of Channel Access Failure).  Here is the capture: 
[csl-default-ramp-up.pcapng.zip](https://github.com/openthread/openthread/files/5666433/csl-default-ramp-up.pcapng.zip) The first Link Request could be sent because we turned on the device's radio (start to receive). And some logs from the device:
[image]

PR #5960 (2020-12-16): [nrf528xx] fix sleep_to_tx transition problem
This PR fixes #5826.

Thanks @hubertmis for debugging and solving this problem. The cause is that in radio scheduler, after coex precondition is approved, no action is taken to notify the core. As a result, the coex precondition is not met when doing the transmission. Now we move the code before HFCLK handling so that HFCLK handling would notify the core.

As there's no recent plan for a new bug-fix release of nRF 802.15.4 driver, let's submit this change to the repo temporarily.
closed 2020-12-16 by jwhui