openthread/openthread #5887: [github-actions] toranj tests 026 and 028 fail intermittently

Irving-cl, 2020-11-30:
**Describe the bug** A clear and concise description of what the bug is.
`tests/toranj/test-026-router-leader-reset-recovery.py` and `tests/toranj/test-028-router-leader-reset-recovery.py` fail intermittently.

Failed run: https://github.com/openthread/openthread/runs/1472778130?check_suite_focus=true

**Console/log output** If applicable, add console/log output to help explain your problem.
```
Traceback (most recent call last):
------------------------------------------------------------------------------------------------------------------------
  File "test-026-slaac-address-wpantund.py", line 175, in <module>
Starting 'test-026-slaac-address-wpantund'
    wpan.verify_within(check_prefix_and_slaac_address_are_removed, WAIT_INTERVAL)
Took too long to pass the condition (5.03834009171>5 sec)
  File "/home/runner/work/openthread/openthread/tests/toranj/wpan.py", line 1042, in verify_within
verify() failed at line 100 in "test-026-slaac-address-wpantund.py"
    raise e
wpan.VerifyError: verify() failed at line 100 in "test-026-slaac-address-wpantund.py"
Attempt 1 running test-026-slaac-address-wpantund.py failed. Trying again.
------------------------------------------------------------------------------------------------------------------------
Starting 'test-026-slaac-address-wpantund'
'test-026-slaac-address-wpantund' passed.
------------------------------------------------------------------------------------------------------------------------
Starting 'test-027-child-mode-change'
'test-027-child-mode-change' passed.
------------------------------------------------------------------------------------------------------------------------
Starting 'test-028-router-leader-reset-recovery'
Took too long to pass the condition (6.10256814957>6 sec)
Failed to find a neighbor entry for extended address E22DF6AE92E29C30 in table
Traceback (most recent call last):
  File "test-028-router-leader-reset-recovery.py", line 129, in <module>
    wpan.verify_within(check_r1_neighbor_table, WAIT_INTERVAL)
  File "/home/runner/work/openthread/openthread/tests/toranj/wpan.py", line 1042, in verify_within
    raise e
wpan.VerifyError: Failed to find a neighbor entry for extended address E22DF6AE92E29C30 in table
Attempt 1 running test-028-router-leader-reset-recovery.py failed. Trying again.
------------------------------------------------------------------------------------------------------------------------
Starting 'test-028-router-leader-reset-recovery'
Took too long to pass the condition (6.0733101368>6 sec)
Failed to find a neighbor entry for extended address 0AE3C5F56BE0B197 in table
Traceback (most recent call last):
  File "test-028-router-leader-reset-recovery.py", line 129, in <module>
    wpan.verify_within(check_r1_neighbor_table, WAIT_INTERVAL)
  File "/home/runner/work/openthread/openthread/tests/toranj/wpan.py", line 1042, in verify_within
    raise e
wpan.VerifyError: Failed to find a neighbor entry for extended address 0AE3C5F56BE0B197 in table
Attempt 2 running test-028-router-leader-reset-recovery.py failed. Trying again.
------------------------------------------------------------------------------------------------------------------------
Starting 'test-028-router-leader-reset-recovery'
Took too long to pass the condition (6.05735206604>6 sec)
Failed to find a neighbor entry for extended address 0A7F82CBA4FCA250 in table
Traceback (most recent call last):
  File "test-028-router-leader-reset-recovery.py", line 129, in <module>
    wpan.verify_within(check_r1_neighbor_table, WAIT_INTERVAL)
  File "/home/runner/work/openthread/openthread/tests/toranj/wpan.py", line 1042, in verify_within
    raise e
wpan.VerifyError: Failed to find a neighbor entry for extended address 0A7F82CBA4FCA250 in table
 *** TEST FAILED
==> wpantund-logs1.log <==
```

abtink, 2020-11-30:
I have noticed `test28` failing often as well recently.  Don't recall it failing as frequently before.

I think something may have changed that is causing this happening more often (maybe a change/bug in core or in the test itself).

When/if I get some time, will try to run a stress test on this to see if we can reproduce it with logs and see what's happening.

jwhui, 2024-06-21:
Closing stale issue.

closed 2024-06-21 by jwhui