openthread/openthread #5897: OpenThread support for 868MHz band on EFR32MG12

SwathiRamachandra, 2020-12-01:
Do we have any plans to support OpenThread for 868MHz on EFR32MG12?
Could you please let me know the configuration/changes that are needed in OT to enable this feature?

Thanks,
Swathi

jwhui, 2020-12-05:
OpenThread's support for sub-GHz is currently experimental. The [Thread Specification](https://www.threadgroup.org/support#specifications) does not currently include operation over sub-GHz bands.

We do not currently have any plans to support 868 MHz, but it should be pretty easy to add. The current prototype:
- Supports the [EFR32MG12 2400 / 868-915 MHz](https://www.silabs.com/documents/public/user-guides/ug342-brd4170a-user-guide.pdf) radio.
- Leverages IEEE 802.15.4-2006 Channel Page 0, which assigns channel number 0 to the 868 MHz band.

For a quick test, it may be as simple as:
1. Changing OT_RADIO_915MHZ_OQPSK_CHANNEL_MIN to 0.
https://github.com/openthread/openthread/blob/82eee1f8d32980279ba5b9b63f53609325d910fb/include/openthread/platform/radio.h#L100

2. Changing OT_RADIO_915MHZ_OQPSK_CHANNEL_MASK to 0x7ff
https://github.com/openthread/openthread/blob/82eee1f8d32980279ba5b9b63f53609325d910fb/include/openthread/platform/radio.h#L102

Then configuring the channel to 0 for operation in the 868 MHz band.

Hope that helps.

SwathiRamachandra, 2020-12-07:
I am using BRD4163A Radio board (https://www.silabs.com/documents/public/user-guides/ug262-brd4163a.pdf)
I have made the changes as per above, mentioned by you and configured the channel to 0 to operate on 868MHz band.
Addition to the above change I have also generated rail_config.c file for 868MHz board from Radio configurator in Simplicity studio IDE and copied it to third_party/silabs/rail_config.
It is failing in joining process (error message: Join failed [NotFound]).
In joiner device, scan command does not discover Thread network.

Background: At first I tried SubGHz on board BRD4164A (EFR32MG12 2.4GHz/915MHz Radio board). I added board_config.h and hal-config.h in BRD4164A folder for EFR32MG12P433F1024GL125 MCU and changed Makefile-efr32mg12 to support this board. Everything works fine with 915MHz board. Verified the frequency using hackrf one hardware, the frequency was around 904.2-906.4. We need to work on 868MHz band that is supported for Europe.

Am I missing out anything here?

Thanks,
Swathi

jwhui, 2020-12-08:
@SwathiRamachandra , unfortunately, I do not have a BRD4163A board to help with this.

Maybe someone from Silicon Labs can chime in? /cc/ @mhallam91 @cbabcock2

SwathiRamachandra, 2020-12-14:
Thanks @jwhui 
@mhallam91 @cbabcock2 Any updates on this?

silabs-YupingX, 2020-12-15:
@SwathiRamachandra - I think 868 MHz support involves more than the changes in radio.h that Jonathan suggested. One key thing might be the symbol rate. If you look up “Symbol rate” in IEEE 802.15.4-2006, you will see that the O-QPSK PHY symbol rate is quite different when operating in the 868 MHz band compared to when operating in the 915 MHz band. This will require some changes in our [RAIL configuration](https://github.com/openthread/openthread/blob/master/third_party/silabs/rail_config/rail_config.c) that we don’t have today. 
 
As Jonathan said earlier, "OpenThread's support for sub-GHz is currently experimental. The [Thread Specification](https://www.threadgroup.org/support#specifications) does not currently include operation over sub-GHz bands." You can probably experiment with our [Flex SDK](https://www.silabs.com/developers/flex-sdk-radio-abstraction-interface-layer) and see if you can generate a RAIL configuration for the 868 MHz PHY, however, I wonder if that means your solution would potentially become “proprietary” rather than “standard” depending on where the Thread Specification goes?

Thanks,
Yuping

SwathiRamachandra, 2020-12-22:
Thanks @silabs-YupingX for your reply.
1. I used RAIL Test example to create RAIL configuration file, and changed bitrate to 100kbps. as attached in   
    radio_settings.zip. I still getting same result. Join failed[Not found]. Also using scan command not able to discover   
    thread network.
2. I used Mac Mode device example Radio configuration file, I get Join failed[Security]. i have given correct PSKId. Using scan 
   command able to discover network.
3. I tried to edit rail_config.c file of 915MHz frequency band. Changed frequency to 868MHz, channel spacing to 400KHz and   
   channel number min to 0 and max to 10. I am able to join successfully and have DTLS communication and get the data. But i 
   am getting error 0x6900 (MBEDTLS_ERR_SSL_WANT_READ) during mbedtls_ssl_handshake and mbedtls_ssl_read. But yet able to  
   pass all the commands successfully.

Could you please share your thoughts on above? If possible would you able to share the radioconf file of 915MHz?

Thanks,
Swathi

[radio_settings.zip](https://github.com/openthread/openthread/files/5731115/radio_settings.zip)

SwathiRamachandra, 2021-01-13:
@silabs-YupingX A friendly reminder.

Thanks,
Swathi

silabs-YupingX, 2021-01-22:
@SwathiRamachandra From the radioconf you attached, it looks like the frame is not set up for 15.4 (e.g. length decoding and crc). We don’t currently have a tested radio_settings.radioconf file for 915MHz.
 
Have you seen this KBA? Wondering if it might be of any help at all: https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2017/02/24/ieee_802_15_4_andra-vGci
 
Also, have you checked if the config is working reliably in railtest (at least a tx/rx test with 2 devices)?

SwathiRamachandra, 2021-01-22:
@silabs-YupingX, Yes the default config file which is in railtest example works fine for tx/rx test with two devices. 
I will try with the link provided.

Thanks,
Swathi

jwhui, 2021-03-10:
Closing stale issue.

closed 2021-03-10 by jwhui