openthread/openthread #5914: toranj-rcp test-001-get-set.py fail intermittently

simonlingoogle, 2020-12-04:
**Describe the bug**
toranj-rcp test test-001-get-set.py fail intermittently

**Console/log output**

[logs_43322.zip](https://github.com/openthread/openthread/files/5640845/logs_43322.zip)

abtink, 2020-12-05:
We have seen this patten in the past (when the very first test fails). It seems to be dbus/system related (not a test issue). We have seen similar issue on actual devices as well. 

I have investigated this in past (when we are using travis). I know re-runs of the test itself and/or re-building (re-installing) of `wpantund` won't help with it. Re-run/re-build of whole test would address it.

I don't expect it to happen often. If we see it happen a lot then we may need to figure a way around it (e.g. an automated way to re-kick the whole setup and test so everything is re-built fresh).

simonlingoogle, 2020-12-08:
@abtink Per my experience, it happens quite often recently. 

e.g. https://github.com/openthread/openthread/pull/5910/checks?check_run_id=1515464577

It's interesting that this fail can only be resolved after a fresh rebuild.

jwhui, 2024-06-21:
Closing stale issue.

closed 2024-06-21 by jwhui