openthread/openthread #5918: Configuration for Tread Devices Roles missing!

uttaravadina, 2020-12-04:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
1. The build time configuration for various roles of Thread Device (REED, SED, FED, MED) is missing. --enable-ftd and --enable-med is not sufficient. For eg: It doesn't translate to how a REED can be built or just a SED.
2. No available documentation on OPENTHREAD_CONFIG_* configuration available which makes it difficult to understand how the 1. can be configured for different roles, what are the mandatory ones and what are to be really careful with.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
In regards to **1.**, it will be much effective if each of those roles has separate configuration files.
In regards to **2.**, brief documentation on each of the OPENTHREAD_CONFIG_* config MACROS.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
I tried reading the code, trying to understand and connect the dots but this brute-force approach is taxing and costs a lot of time.

**Additional context** Add any other context or screenshots about the feature request here.
null

jwhui, 2020-12-04:
Device modes are configurable at run-time via [otThreadSetLinkMode()](https://openthread.io/reference/group/api-thread-general#otthreadsetlinkmode). The Thread Specification allows devices to change their device mode dynamically.

You can find more information about the `OPENTHREAD_CONFIG_*` build options in [src/core/config](https://github.com/openthread/openthread/tree/master/src/core/config).

uttaravadina, 2020-12-05:
So most of the roles and device modes are covered by that then, except for 2.

1. Although a Router can be manually turned to a child and it can again elect itself as a Router/Leader. FED falls under FTD but **how does FED limit itself to not be elected as Router/Leader no matter what**? And if that's possible then why not put FED in the MTD category as it won't require to subscribe to all-routers and preserve address mappings. It just needs to know about one Router (parent). 

2. `A Minimal Thread Device does not subscribe to the all-routers multicast address and **forwards** all messages to its Parent.`
By forwards does it literally forwards messages or was it meant to send the Minimal Thread Device's own message to its Parent.
If yes, then whose messages is it going to forward? 
If not, then can you please update **forwards** to **sends** because of the Networking vocabulary it creates confusion.

jwhui, 2020-12-05:
> 1. Although a Router can be manually turned to a child and it can again elect itself as a Router/Leader. FED falls under FTD but **how does FED limit itself to not be elected as Router/Leader no [...]

See [otThreadSetRouterEligible()](https://openthread.io/reference/group/api-thread-router#otthreadsetroutereligible) for enabling/disabling router eligibility.

FEDs actually benefit from maintaining links to multiple neighboring routers when receiving IPv6 multicast communication. It offers greater diversity in the set of devices it can receive an IPv6 multicast message from.

FEDs also perform FTD functionality such as EID-to-RLOC address resolution, forwarding using the 6LoWPAN Mesh Addressing header, etc.

As the name suggests, a Full End Device (FED) is not a Minimal Thread Device (MTD). On the other hand, a Minimal End Device (MED) is a Minimal Thread Device.

> 2. `A Minimal Thread Device does not subscribe to the all-routers multicast address and **forwards** all messages to its Parent.`
>    By forwards does it literally forwards messages or was it meant [...]

There is nothing in the Thread Specification that explicitly forbids an MTD from operating as a Thread Border Router.

uttaravadina, 2020-12-05:
I totally missed that in the API Docs.
Have you gotten any requests to allow these configs to be set during the build times itself? Like maybe creating OPENTHREAD_CONFIG_ROUTER_ELIGIBLE=1 instead of programmatically setting it?

That makes total sense. There might be use-cases where that will be good to have. Got it.

Regarding MTD, my concern is with the **forwards** and **sends** used in the Networking terminology, which mean different. I picked that quote from OpenThread's Node Roles and Types in which it was mentioned that `MTD **forwards**` which I think must have been **sends**. CM if I am wrong.

jwhui, 2020-12-05:
> I totally missed that in the API Docs.
> Have you gotten any requests to allow these configs to be set during the build times itself? Like maybe creating OPENTHREAD_CONFIG_ROUTER_ELIGIBLE=1 instead of programmatically setting it?

You are the first :)

> Regarding MTD, my concern is with the **forwards** and **sends** used in the Networking terminology, which mean different. I picked that quote from OpenThread's Node Roles and Types in which it was mentioned that `MTD **forwards**` which I think must have been **sends**. CM if I am wrong.

As previously mentioned, it is possible for an MTD to serve as a border router, which means that the MTD may forward messages originated by other devices. Admittedly, this is not likely to be a common use case, but it is possible.

uttaravadina, 2020-12-05:
@jwhui would you consider having this option during the build time? It sure will save a lot of time for who are tinkering and prototyping.

But would it forward messages from any other device in the thread network if it's not a Border Router? That is all.

uttaravadina, 2020-12-05:
Would it be fine if I added a PR for this build-time configuration?

jwhui, 2020-12-05:
> Would it be fine if I added a PR for this build-time configuration?

Yes, feel free to submit a PR that allows changing the default router eligible value at build time.

> But would it forward messages from any other device in the thread network if it's not a Border Router?

A SED could receive an IPv6-in-IPv6 datagram, where the inner IPv6 datagram is not originated by the SED.

uttaravadina, 2020-12-05:
Thanks and thanks again!
That will be it.

closed 2020-12-05 by uttaravadina